ros2/ros2 #1048: i can't get colored log in terminal

mantangL, 2020-10-14:
my src:
#include <memory>

#include "rclcpp/rclcpp.hpp"

int main(int argc, char * argv[])
{
  // Force flush of the stdout buffer.
  setvbuf(stdout, NULL, _IONBF, BUFSIZ);

  rclcpp::init(argc, argv);
  
  rclcpp::Node::SharedPtr node = std::make_shared<rclcpp::Node>("logging_test");
  RCLCPP_INFO(node->get_logger(), "dsad");
  RCLCPP_WARN(node->get_logger(), "dsad");
  RCLCPP_ERROR(node->get_logger(), "dsad");

  rclcpp::spin(node);

  rclcpp::shutdown();
  return 0;
}

my problem:
when ros2 run this node, the log is colored.  but ros2 launch a .py file include this node, it is no color.
my logging_test.py:
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='logging_test',
            node_namespace='logging_test',
            node_executable='logging_test'
        )
    ])

how can i get a color log launched by this .py file

iuhilnehc-ynos, 2020-10-22:
@mantangL 

You can set `emulate_tty` with `True` for this Node,
```
def generate_launch_description():
return LaunchDescription([
Node(
package='logging_test',
node_namespace='logging_test',
node_executable='logging_test',
emulate_tty=True
)
])
```

or use an environment variable, such as `export RCUTILS_COLORIZED_OUTPUT=1`

mantangL, 2020-10-22:
> export RCUTILS_COLORIZED_OUTPUT=1

thanks very much, i solve this problem using export RCUTILS_COLORIZED_OUTPUT=1.
and i use 

> emulate_tty=True
a problem arises:

> [ERROR] [launch]: Caught exception in launch (see debug for traceback): __init__() got an unexpected keyword argument 'emulate_tty'

how can i solve it?

mantangL, 2020-10-22:
> @mantangL
>
> You can set `emulate_tty` with `True` for this Node,
>
> ```
> def generate_launch_description():
> return LaunchDescription([
> Node(
> package='logging_test',
> node_namespace='logging_test', [...]

thanks very much, i solve this problem using export RCUTILS_COLORIZED_OUTPUT=1.
and i use

emulate_tty=True

a problem arises:

[ERROR] [launch]: Caught exception in launch (see debug for traceback): init() got an unexpected keyword argument 'emulate_tty'

how can i solve it?

iuhilnehc-ynos, 2020-10-22:
> how can i solve it?

I think you need to use ros2 with distro `foxy`(or `eloquent`, which I didn't check) not `dashing` if you want to use `emulate_tty`. It seems this feature doesn't backport for `dashing`.

mantangL, 2020-10-22:
> > how can i solve it?
> 
> I think you need to use ros2 with distro `foxy`(or `eloquent`, which I didn't check) not `dashing` if you want to use `emulate_tty`. It seems this feature doesn't backport for `dashing`.
> 
> BTW: I suggest that you could create an issue by using the default template next time.

thanks

clalancette, 2020-11-05:
It looks like this is solved, so I'm going to close this out.  Feel free to reopen if there is still something to be done here.

closed 2020-11-05 by clalancette

nlgilbert, 2024-02-12:
# Functionality Review

1. Source address is taken from the settings file. Good work!
2. When a Request for Address Claim message is sent out, the camera doesn't respond right away. After a few seconds it gives the expected response, but with the following terminal output. This happens both when PeP is enabled and disabled.
[image]
3. There's also still the issue where sometimes when starting up we get the following error.
[image]