ros2/ros2 #1049: macOS builds fail with "Alarm clock: 14"

jacobperron, 2020-10-15:
It's rare, but sometimes we see macOS jobs fail with a cryptic error message in the middle of compilation, e.g.

```
01:45:41 [ 35%] Linking CXX executable tf2_ros_test_transform_broadcaster
01:45:41 make: *** [all] Alarm clock: 14
01:45:41 make: INTERNAL: Exiting with 1 jobserver tokens available; should be 4!
```

It appears to be agnostic to the package being built.

Instances of this kind of failure:

- https://ci.ros2.org/view/nightly/job/nightly_osx_debug/1778
- https://ci.ros2.org/view/nightly/job/nightly_osx_debug/1757
- https://ci.ros2.org/view/nightly/job/nightly_osx_release/1813
- https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2081

dirk-thomas, 2020-10-15:
The error message says

> `make: INTERNAL: Exiting with 1 jobserver tokens available; should be 4!`

which indicates that a make process failed. If that happens while holding jobserver tokens those aren't returned to the pool which results in the reported mismatch.

The actual question is why was the `make` process failing? It could be something like running out of memory.

mjcarroll, 2020-10-16:
The `Alarm Clock: 14` would indicate that something is throwing up `SIGALRM` based on this [stackoverflow answer](https://unix.stackexchange.com/a/250300)

clalancette, 2020-10-16:
> The `Alarm Clock: 14` would indicate that something is throwing up `SIGALRM` based on this [stackoverflow answer](https://unix.stackexchange.com/a/250300)

Yeah, that's probably what it means.  I dug a bit into the Make source code a while back, and it uses SIGALRM (on non-Linux Unix) to wake up every X milliseconds to check on the status of processes it is monitoring.  But its unclear to me why that is failing (nor what the error number 14 means).

gavanderhoorn, 2020-10-16:
> nor what the error number 14 means)

`14` is the value of `SIGALARM` (see [signal(7) — Linux manual page](https://man7.org/linux/man-pages/man7/signal.7.html)).

jacobperron, 2020-12-03:
Another instance: https://ci.ros2.org/job/nightly_osx_release/1889/

chapulina, 2021-01-21:
It happened again: https://ci.ros2.org/view/nightly/job/nightly_osx_debug/1899

ivanpauno, 2021-01-21:
The only place where they seem to be setting an alarm is [here](https://github.com/mirror/make/blob/a80b0096f501b0e9a3349858396841d493cc45a4/src/posixos.c#L399-L401).

I'm not sure what's the problem, but I have some speculations:

- Is the compiler allowed to reorder the call to `sigaction()` and the call to `alarm()`?
  My understanding is that when there are no dependencies between the functions, it can.
  I would be surprised if it's allowed to reorder system calls, that would be completely crazy.
  But I'm not a C standard lawyer :joy: (not a reliable discussion, but see https://www.reddit.com/r/cpp/comments/dh3hle/why_is_compiler_allowed_to_reorder_instructions/).
  (the only way to check this is to see the dissembled code in macOS).
- Maybe the OS has a race between disabling the `alarm` and reinstalling the default alarm signal handler.
  Again, that's crazy, but I'm not sure what exactly posix specifies, neither if macOS is compliant.

I don't see what else could be wrong ....

ivanpauno, 2021-01-21:
disassembled code (on Linux) `alarm(0)` after sigaction resintalling `SIG_DFL`:

```
   1917b:       48 89 85 30 ff ff ff    mov    %rax,-0xd0(%rbp)
   19182:       e8 89 f5 fe ff          callq  8710 <sigaction@plt>
   19187:       bf 01 00 00 00          mov    $0x1,%edi
   1918c:       e8 8f f7 fe ff          callq  8920 <alarm@plt>
   19191:       8b 3d 89 cb 21 00       mov    0x21cb89(%rip),%edi        # 235d20 <job_rfd@@Base>
   19197:       48 8d b5 2f ff ff ff    lea    -0xd1(%rbp),%rsi
   1919e:       ba 01 00 00 00          mov    $0x1,%edx
   191a3:       e8 d8 f7 fe ff          callq  8980 <read@plt>
   191a8:       48 89 85 18 ff ff ff    mov    %rax,-0xe8(%rbp)
   191af:       e8 ac f4 fe ff          callq  8660 <__errno_location@plt>
   191b4:       b9 12 00 00 00          mov    $0x12,%ecx
   191b9:       44 8b 38                mov    (%rax),%r15d
   191bc:       49 89 c6                mov    %rax,%r14
   191bf:       4c 89 ef                mov    %r13,%rdi
   191c2:       31 c0                   xor    %eax,%eax
   191c4:       4c 8b 0d a5 cf 21 00    mov    0x21cfa5(%rip),%r9        # 236170 <batch_mode_shell@@Base+0x90>
   191cb:       f3 48 ab                rep stos %rax,%es:(%rdi)
   191ce:       31 d2                   xor    %edx,%edx
   191d0:       4c 89 e6                mov    %r12,%rsi
   191d3:       bf 11 00 00 00          mov    $0x11,%edi
   191d8:       4c 89 8d 10 ff ff ff    mov    %r9,-0xf0(%rbp)
   191df:       48 89 9d 30 ff ff ff    mov    %rbx,-0xd0(%rbp)
   191e6:       c7 45 b8 00 00 00 10    movl   $0x10000000,-0x48(%rbp)
   191ed:       e8 1e f5 fe ff          callq  8710 <sigaction@plt>
   191f2:       4c 8b 8d 10 ff ff ff    mov    -0xf0(%rbp),%r9
   191f9:       4d 85 c9                test   %r9,%r9
   191fc:       74 28                   je     19226 <new_job@@Base+0x5e6>
   191fe:       31 ff                   xor    %edi,%edi
   19200:       e8 1b f7 fe ff          callq  8920 <alarm@plt>
   19205:       31 d2                   xor    %edx,%edx
   19207:       4c 89 e6                mov    %r12,%rsi
   1920a:       bf 0e 00 00 00          mov    $0xe,%edi
   1920f:       48 c7 85 30 ff ff ff    movq   $0x0,-0xd0(%rbp)
   19216:       00 00 00 00
```

bingo!

I checked `make` git history that code wasn't modified between my version of `make` and now, so it's definitely a compiler reordering issue.

jacobperron, 2021-04-01:
Another instance: https://ci.ros2.org/view/nightly/job/nightly_osx_release/2015/

ivanpauno, 2021-04-05:
There is a super old gnu make issue about this: http://savannah.gnu.org/bugs/?func=detailitem&item_id=41341.
Though it has been fixed ages ago, it would be great to run `make --version` on `mini1`, `mini3` or `lore` to see what version we're using. `4.1` or above is fine, `4.0` or older is not good.

clalancette, 2021-04-05:
Its:

```
GNU Make 3.81
```

And that's because macOS stopped updating anything GNU at around 2006.

There is a brew `make` package available; we could consider trying to use that instead, though I don't know what the fallout of that will be.

ivanpauno, 2021-04-05:
> GNU Make 3.81

Nice, now this make more sense!

> There is a brew make package available; we could consider trying to use that instead, though I don't know what the fallout of that will be.

:+1: 
I'm not completely sure how to ensure that `cmake` is using the brew installed `make` though.
Based on the [homebrew notes](https://formulae.brew.sh/formula/make):

> GNU "make" has been installed as "gmake".If you need to use it as "make", you can add a "gnubin" directory to your PATH from your bashrc like:     PATH="$(brew --prefix)/opt/make/libexec/gnubin:$PATH"

I guess that that will be enough.
See also [this](https://stackoverflow.com/questions/43175529/updating-make-version-on-mac).

clalancette, 2021-04-05:
> I guess that that will be enough.

Yeah, the thing is that will also use other newer GNU utilities that are installed.  That may work OK, but I'm not sure.  I think we should wait on this until after the Galactic freeze, and we can do some playing around in the next few weeks.

sloretz, 2021-09-23:
Happened again: https://ci.ros2.org/view/nightly/job/nightly_osx_debug/2157/

```
make: *** [all] Alarm clock: 14
make: INTERNAL: Exiting with 1 jobserver tokens available; should be 4!
```

clalancette, 2022-02-03:
Since macOS is no longer a Tier 1 platform, this isn't really a problem anymore.  So I'm going to close this out.

closed 2022-02-03 by clalancette