ros2/ros2 #1051: Unable to run SROS on ROS 2 foxy

evshary, 2020-10-21:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04
- Installation type:
  binaries
- Version or commit hash:
  ROS 2 foxy
- DDS implementation:
  Cyclone DDS
- Client library (if applicable):
  N/A

#### Steps to reproduce issue

```
# Follow the steps in SROS: https://github.com/ros2/sros2/blob/foxy/SROS2_Linux.md
mkdir ~/sros2_demo
cd ~/sros2_demo
ros2 security create_keystore demo_keys
ros2 security create_key demo_keys /talker_listener/talker
ros2 security create_key demo_keys /talker_listener/listener
export ROS_SECURITY_KEYSTORE=~/sros2_demo/demo_keys
export ROS_SECURITY_ENABLE=true
export ROS_SECURITY_STRATEGY=Enforce
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
ros2 run demo_nodes_cpp talker --ros-args --enclave /talker_listener/talker
```

#### Expected behavior
Able to run talker.

#### Actual behavior
Get the following errors.
```
1603240969.859947 [30]     talker: Could not load Authentication library: dds_security_auth: cannot open shared object file: No such file or directory
1603240969.860362 [30]     talker: Could not load Authentication plugin.
1603240969.860652 [30]     talker: Could not load security
[ERROR] [1603240969.862000026] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create DDS participant

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /tmp/binarydeb/ros-foxy-rcl-1.1.8/src/rcl/node.c:276'

with this new error message:

  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.8/src/rcl/node.c:428'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1603240969.864867262] [rcl]: Failed to fini publisher for node: 1
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-foxy-rcl-1.1.8/src/rcl/node.c:428
```

#### Additional information

If I build Cyclone DDS from source code, I can run SROS successfully. I guess this is because the security library isn't included into the Debian Package of CycloneDDS, that is, ros-foxy-cyclonedds.

clalancette, 2020-11-05:
@ros-security-wg Can you take a look?

mikaelarguedas, 2020-11-05:
This looks like a debian packaging issue.
https://github.com/eclipse-cyclonedds/cyclonedds/pull/631 should fix it

jacobperron, 2021-01-13:
This looks fixed to me, though we probably have to wait for another release of CycloneDDS so see it propagated to a binary installation.

closed 2021-01-13 by jacobperron