ros2/ros2 #1052: In the case of no network connection, there can be no more than 3 nodes

JoveH-H, 2020-10-25:
In the case of no network connection, there can be no more than 3 nodes

- Operating System:
  Ubuntu 20.04
- Installation type:
 binaries 

#### Expected behavior
Start 5 nodes
#### Actual behavior
Only 3 nodes can be started

fujitatomoya, 2020-10-26:
I would request more information here, if possible. did you meet any error or warning messages? and what is the use case of this requirement? probably just using `intra-process communication` in the single process space but network?

JoveH-H, 2020-10-26:
The terminal does not display any error or warning messages.
For example, Without Internet connection:
ros2 run demo_nodes_py add_two_ints_client
ros2 run demo_nodes_py add_two_ints_client
ros2 run demo_nodes_py talker
ros2 run demo_nodes_py listener

There should be four viewing nodes, but only three.
 ros2 node list
  /add_two_ints_client
  /add_two_ints_client
  /talker

Although listener node can be started normally, it cannot be found.

fujitatomoya, 2020-10-26:
> For example, Without Internet connection:

Does this mean that it uses localhost(local loopback) only?

> Although listener node can be started normally, it cannot be found.

listener is in the same localhost with talker, right?

besides what ros version do you use? foxy with default rmw implementation(fastdds)?

JoveH-H, 2020-10-26:
> > For example, Without Internet connection:
> 
> Does this mean that it uses localhost(local loopback) only?
> 
Yes.

> > Although listener node can be started normally, it cannot be found.
> 
> listener is in the same localhost with talker, right?
> 
> besides what ros version do you use? foxy with default rmw implementation(fastdds)?
I think they are. They are all default.

iuhilnehc-ynos, 2020-10-26:
@JoveH-H @fujitatomoya 

I can reproduce this issue.
If using `rmw_cyclonedds_cpp` or `rmw_connext_cpp`, this test case works well.

FYI: We can use the simplest example `HelloWorldExample` of `fastdds` to reproduce this issue.
```
# terminal 1
./HelloWorldExample publisher 1000

# terminal 2
./HelloWorldExample publisher 1000

# terminal 3
./HelloWorldExample publisher 1000

# terminal 4
./HelloWorldExample publisher 1000

# terminal 5
./HelloWorldExample subscriber
```
The last publisher can't be established to publish a message
[image]
It seems this issue belongs to `fastdds`.

clalancette, 2020-10-26:
@MiguelCompany @richiware Any thoughts here on what might be causing this?

MiguelCompany, 2020-10-26:
I've been able to reproduce it both on Linux and Windows. I've found the place in the code to look at, but the solutions I've been trying differ on Windows and Linux, and I would like to find something that suits them both. Will keep you informed.

fujitatomoya, 2020-10-26:
Just FYI, it can be reproducible with the following,

```
# start docker container w/o network interfaces except loopback.
> docker run -it --network none ros:foxy /bin/bash
> apt install ros-foxy-demo-nodes-cpp

# start talker and  listenr
> ros2 run demo_nodes_cpp talker &
> ros2 run demo_nodes_cpp talker &
> ros2 run demo_nodes_cpp talker &
> ros2 run demo_nodes_cpp listener &
> ros2 node list
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
/talker
/talker
/talker
```

listener cannot be printed, even though subscription works okay. (receiving the messages.) restarting ros2 daemon does not fix the problem either.

fujitatomoya, 2020-10-26:
with [ROS_LOCALHOST_ONLY](https://github.com/ros2/ros2/issues/798), it does not work either.

MiguelCompany, 2020-10-27:
@JoveH-H @fujitatomoya @iuhilnehc-ynos @clalancette 

eProsima/Fast-DDS#1520 should fix the original problem

The issue with `ROS_LOCALHOST_ONLY` will persist, as the problem lies on `rmw_fastrtps`, not Fast-DDS. I will try to fix that part during the afternoon.

JoveH-H, 2020-10-27:
OK, thank you for your guidance.

fujitatomoya, 2020-10-28:
@MiguelCompany 
CC: @JoveH-H 

i confirmed https://github.com/eProsima/Fast-DDS/pull/1520 with https://github.com/ros2/ros2/commit/4aeefc197b1e7a4c267ae0e36147db15e565e849 can fix the original problem. (which is not `ROS_LOCALHOST_ONLY`)

JoveH-H, 2020-10-28:
> @MiguelCompany @fujitatomoya 
Thanks

clalancette, 2020-11-18:
Since the original bug here is fixed in upstream Fast-DDS, closing this out.  Feel free to reopen if this is in error.

closed 2020-11-18 by clalancette

BernardinD, 2021-03-08:
I seem to be having the same issue with Eloquent. But I install it after this bug was supposedly handled. Has this fix been applied to Eloquent??

fujitatomoya, 2021-03-08:
@BernardinD 

no i don't think so. according to https://docs.ros.org/en/foxy/Releases.html, Eloquent has been E.O.L.

BernardinD, 2021-03-08:
@fujitatomoya Could you advise on how to handle it manually myself? I'm assuming building Fast-DDS from source would do the trick

fujitatomoya, 2021-03-09:
@BernardinD 

i strongly recommend that rebasing on foxy... 😦 

i would not do and i do not recommend. but i guess the following would be a try. you would probably see the unexpected problems. (i never tried before)

1. Based on ros:eloquent colcon build environment. (refer to https://docs.ros.org/en/eloquent/Installation/Linux-Development-Setup.html)
2. Merge https://github.com/eProsima/Fast-DDS/pull/1520 to https://github.com/eProsima/Fast-DDS/tree/ros2-eloquent.
3. Merge https://github.com/ros2/rmw_fastrtps/pull/476 to https://github.com/ros2/rmw_fastrtps/tree/eloquent.
4. colcon build & run