ros2/ros2 #1057: get_topic_names_and_types often fails to get active topics

smorad, 2020-11-10:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04 x64
- Installation type:
  Binary
- Version or commit hash:
  `ros-foxy-ros2node amd64 0.9.7-1focal.20201103.152358`
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclpy

#### Steps to reproduce issue

term1:
```
ros2 topic pub /bork std_msgs/msg/Float32 '{data: 1.0}'
```

term2:
```
~/code/prorok/ros2$ python3 test.py 
~/code/prorok/ros2$ python3 test.py 
~/code/prorok/ros2$ python3 test.py 
...
~/code/prorok/ros2$ python3 test.py 
```

`test.py`:
```
#! /bin/env python3
import rclpy
from rclpy.node import Node
rclpy.init()
n = Node('abc')
print(n.get_topic_names_and_types())
```

#### Expected behavior
```
[('/bork', ['std_msgs/msg/Float32']), ('/parameter_events', ['rcl_interfaces/msg/ParameterEvent']), ('/rosout', ['rcl_interfaces/msg/Log'])]
...
[('/bork', ['std_msgs/msg/Float32']), ('/parameter_events', ['rcl_interfaces/msg/ParameterEvent']), ('/rosout', ['rcl_interfaces/msg/Log'])]
```
#### Actual behavior
```
[('/bork', ['std_msgs/msg/Float32']), ('/parameter_events', ['rcl_interfaces/msg/ParameterEvent']), ('/rosout', ['rcl_interfaces/msg/Log'])]
[('/parameter_events', ['rcl_interfaces/msg/ParameterEvent']), ('/rosout', ['rcl_interfaces/msg/Log'])]
```
#### Additional information
Perhaps it makes sense to add an optional `search_time` parameter to `get_topic_names_and_types`?

smorad, 2020-11-16:
The cli tool does not have this issue
```
smorad@smorad-ThinkPad-T490:~/code/prorok/ros2$ while true; do ros2 topic list; done
/bork                                             
/parameter_events                                 
/rosout                                                                                              
/bork                                                                                                
/parameter_events                                                                                    
/rosout                                           
/bork                                                                                                                                                                                                      
/parameter_events                                 
/rosout                                                                                                                                                                                                    
/bork                                             
/parameter_events                                                                                                                                                                                          
/rosout                                           
/bork                                                                                                                                                                                                      
/parameter_events                                 
/rosout                                                                                                                                                                                                    
/bork                                             
/parameter_events                                                                                                                                                                                          
/rosout                                           
/bork                                                                                                
/parameter_events                                 
/rosout                                                                                              
/bork                                                                                                
/parameter_events                                                                                    
/rosout                                                                                              
/bork                                                                                                
/parameter_events                                                                                    
/rosout                                                                                              
/bork                                             
/parameter_events                                                                                    
/rosout                                           
/bork                                                                                                
/parameter_events                                                                                    
/rosout                                                                                              
/bork                                             
/parameter_events                                                                                                                                                                                          
/rosout                                                                                              
/bork                                                                                                                                                                                                      
/parameter_events                                 
/rosout                                                                                              
/bork                                             
/parameter_events                                                                                    
/rosout                                           
/bork                                                                                                
/parameter_events                                 
/rosout                                                                                              
/bork                                             
/parameter_events                                 
/rosout                                           
/bork                                             
```

clalancette, 2020-11-16:
In general, it can take some time for discovery to happen.  That's because we are currently doing "P2P" discovery on the backend, and if you create a Node and call `get_topic_names_and_types` on it right away, discovery may not have happened yet.  It may make sense to add a timeout to `get_topic_names_and_types`; I'm not sure.

The command-line tools get around this a different way.  When you launch any `ros2` command-line tool the first time, a daemon is spun up in the background.  The daemon then does discovery, and hangs around.  The command-line tool then queries the daemon for information.  You may want to think about architecting something more like that; currently rapidly creating Nodes doesn't always do the discovery you want (as you discovered).

smorad, 2020-11-16:
I suspected as much, and had initially tried adding a sleep period after `rclpy.init` with no results. Adding a `sleep(0.5)` after creating the node but before calling `get_topic_names_and_types` seems to solve the problem.

closed 2020-11-16 by smorad