ros2/ros2 #1062: Dashing Patch Release 9

nuclearsandwich, 2020-11-18:
This issue will track the broad status of the next Dashing patch release. The date for this patch release is not currently set and will be determined by a critical mass of desired changes ready to be released.

We're going to use a GitHub Project board for this patch release again: [Dashing Patch Release 9](https://github.com/orgs/ros2/projects/29)

As with prior releases, if you want to contribute fixes please open pull requests targeting the latest development branch and mention to the reviewer that you'd like the pull request to be considered for backporting into Dashing.

nuclearsandwich, 2021-06-10:
Patch release packaging jobs

Ubuntu AMD64 [image]
Ubuntu ARM64 [image]
Ubuntu ARMHF [image]
Windows AMD64 [image]
Windows Debug AMD64 [image]

clalancette, 2021-06-24:
@nuclearsandwich Can we close this now?

closed 2021-07-01 by nuclearsandwich