ros2/ros2 #1063: ROS2 (Foxy): "launch" is not working on windows 10.

naeempnu, 2020-11-19:
I get this error when used: usage: ros2 [-h] Call ros2 <command> -h for more detailed usage. ... ros2: error: argument Call ros2 <command> -h for more detailed usage.: invalid choice: 'launch' (choose from 'action', 'bag', 'component', 'daemon', 'doctor', 'extension_points', 'extensions', 'interface', 'lifecycle', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'test', 'topic', 'trace', 'wtf')

Is there any alternative to launch for windows? Thanks.

clalancette, 2020-11-19:
Launch should work in Windows.  It looks like it wasn't installed properly somehow.  What version of ROS 2 are you using, and how did you install it?

naeempnu, 2020-11-19:
I have installed ROS2-Foxy following this: https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Development-Setup/

I also checked the ros2 commands and I get this:

usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage.

ros2 is an extensible command-line tool for ROS 2.

optional arguments:
  -h, --help            show this help message and exit

Commands:
  action     Various action related sub-commands
  bag        Various rosbag related sub-commands
  component  Various component related sub-commands
  daemon     Various daemon related sub-commands
  doctor     Check ROS setup and other potential issues
  interface  Show information about ROS interfaces
  launch     Run a launch file
  lifecycle  Various lifecycle related sub-commands
  multicast  Various multicast related sub-commands
  node       Various node related sub-commands
  param      Various param related sub-commands
  pkg        Various package related sub-commands
  run        Run a package specific executable
  security   Various security related sub-commands
  service    Various service related sub-commands
  test       Run a ROS2 launch test
  topic      Various topic related sub-commands
  trace      Trace ROS nodes to get information on their execution
  wtf        Use `wtf` as alias to `doctor`

These commands have no "Launch" command.
Please let me know if you need any further information.
Thanks.
  Call `ros2 <command> -h` for more detailed usage.

MiguelCompany, 2020-11-20:
> I also checked the ros2 commands and I get this:
> ...
> Commands:
> ...
> launch Run a launch file
> ...

I see a `launch` command listed. 

> These commands have no "Launch" command.

Is it possible that you are trying `Launch` instead of `launch`? I guess ros2 CLI is case sensitive.

naeempnu, 2020-11-20:
Have you checked on windows?
No, I am using launch and it is working fine on Linux.

naeempnu, 2020-11-24:
So I reinstalled ROS2 and the launch is there. Problem solved. Thanks.

closed 2020-11-24 by naeempnu