ros2/ros2 #1067: publish topic with network, then get network disconnected, subscriber won't get the data

erinliu0428, 2020-12-07:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu18.04
- Installation type:
  binaries 
- Version or commit hash:
 ros2 dashing
- DDS implementation:
  Fast - RTPS,1.8.2
- Client library (if applicable):
   rclcpp, rclpy

#### Steps to reproduce issue
- connect computer to Internet
- one terminal pub a topic and another  echo 
```
$ ros2 topic pub /int1 std_msgs/msg/Int32 '{data: 1}'
$ ros2 topic echo  /int1 std_msgs/msg/Int32 
```
- get the network disconnected,
- interrupt the echo command use "ctrl + c", and execute the command again
```
$ ros2 topic echo /int1 std_msgs/msg/Int32 
```

#### Expected behavior
after executing the echo command again,  we should see the data published on the topic at the echo terminal.

#### Actual behavior
But nothing was recieved.
#### Additional information
Actually, it's not just topic, service has this problem too.  As long as I turn the publisher and server on with network, then get network disconnected,  there will be problems with communication, subscriber and client will nerver get data, unless reopen the network or publisher and server node.

fujitatomoya, 2020-12-07:
> after executing the echo command again, we should see the data published on the topic at the echo terminal.

this is because it is expected to use localhost network?

> Actually, it's not just topic, service has this problem too.

service is based on topic, so it will have the same network issue.

i think this is similar with https://answers.ros.org/question/299940/explicitly-constraining-which-network-interfaces-are-used-in-ros2/

with eloquent or later, there is an environment variable ROS_LOCALHOST_ONLY to limit communication to localhost. i think this helps your requirement. (https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/#new-features-in-this-ros-2-release)

erinliu0428, 2020-12-14:
@fujitatomoya  thankyou

closed 2020-12-14 by erinliu0428