ros2/ros2 #1068: Unable to find rclcpp in windows using nightly build

koonpeng, 2020-12-08:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binary
- Version or commit hash:
  - nightly
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
in CMakeLists.txt
```
find_package(rclcpp)
```

#### Expected behavior
rclcpp is found

#### Actual behavior
got error
```
  Imported target "rclcpp::rclcpp" includes non-existent path
    "C:/ci/ws/install/include"
  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:
  * The path was deleted, renamed, or moved to another location.
  * An install or uninstall procedure did not complete successfully.
  * The installation package was faulty and references files it does not
  provide.
```

#### Additional information

Tested using https://ci.ros2.org/view/packaging/job/packaging_windows/lastSuccessfulBuild/artifact/ws/ros2-package-windows-AMD64.zip, was working with foxy release patch 3

hidmic, 2020-12-14:
I took a brief look, and I might be completely off tracks, but I suspect an interaction between `tracetools` and (non)modern CMake support in `ament_target_dependencies()`. Since `tracetools` does not export targets on `Win32` by default, the exported `_INCLUDE_DIRS` variable containing a `${tracetools_DIR}/../../../include` path gets picked up.

If my assessment is correct, `tracetools` should be exporting an `INTERFACE` target instead and we have a problem with `ament_target_dependencies()` and relocatable packages.

clalancette, 2022-02-03:
Since this very much looks like a duplicate of https://github.com/ros2/rclcpp/issues/1688, I'm going to close this one out.

closed 2022-02-03 by clalancette