ros2/ros2 #1072: rqt cannot find rqt_gui_cpp/plugin.xml and fails to display  approariately in windows10 for foxy release4

dzywater, 2020-12-16:
D:\>rqt
RosPluginProvider._parse_plugin_xml() plugin file "D:\app\ros2\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "D:\app\ros2\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found

[image]

clalancette, 2021-01-07:
@jacobperron will look up related information about this.

jacobperron, 2021-01-07:
@dzywater Do you know if `rqt` worked as expected on Windows in previous releases? Based on this ticket, I don't think rqt_gui_cpp is expected to be working on Windows: https://github.com/ros-visualization/rqt/issues/178. Currently, I would expect to see an empty shell package on Windows and hence no `plugin.xml` file.

Yadunund, 2023-05-08:
Issue persists on `iron`
https://github.com/osrf/ros2_test_cases/issues/846#issuecomment-1534923131

felixf4xu, 2023-05-26:
in my case, `rqt` works, but none of the plugins work.
[image]

Aayushyamaan-Shah, 2023-11-26:
> in my case, `rqt` works, but none of the plugins work. [...]

I am having the same issue, the "rqt" command opens the dialog but none of the plugins work