ros2/ros2 #1073: rviz2 cannot run in windows10 for foxy release4

dzywater, 2020-12-16:
D:\>rviz2
[INFO] [1608098530.478500200] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1608098530.478972100] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1608098530.515230500] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1608098530.516320200] [pluginlib.ClassLoader]: Could not find package manifest (neither package.xml or deprecated manifest.xml) at same directory level as the plugin XML file D:\app\ros2/share/rviz_default_plugins/plugins_description.xml. Plugins will likely not be exported properly.
[image]

mjeronimo, 2021-02-18:
@dzywater I've downloaded Release 4 from here: https://github.com/ros2/ros2/releases/tag/release-foxy-20201211
and tried to reproduce, but wasn't able to. Is see that there is a package.xml file in share\rviz_default_plugins. If I rename this file and re-run rviz2, I do get the output mentioned. Can you please confirm the contents of the BUILD_URL file in your distribution.

dzywater, 2021-02-20:
> @dzywater I've downloaded Release 4 from here: https://github.com/ros2/ros2/releases/tag/release-foxy-20201211
> and tried to reproduce, but wasn't able to. Is see that there is a package.xml file in [...]

[image]

dzywater, 2021-08-29:
> @dzywater I've downloaded Release 4 from here: https://github.com/ros2/ros2/releases/tag/release-foxy-20201211
> and tried to reproduce, but wasn't able to. Is see that there is a package.xml file in [...]

@mjeronimo, could it be a network problem? I ran successfully on another computer.

dzywater, 2021-08-29:
> @dzywater I've downloaded Release 4 from here: https://github.com/ros2/ros2/releases/tag/release-foxy-20201211
> and tried to reproduce, but wasn't able to. Is see that there is a package.xml file in [...]

@mjeronimo could it be the network problem?  Does rviz2 download anything when startup? I ran it successfully on another computer.

mjcarroll, 2025-03-29:
Foxy is end-of-life

closed 2025-03-29 by mjcarroll