ros2/ros2 #1074: Stability regression since the recent foxy sync (Patch Release 4)

tal-grossman, 2020-12-16:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04 (amd64 & arm64)
- Installation type:
  binaries
- Version or commit hash:
 
- DDS implementation:
 Fast-RTPS & Cyclon
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
The degradation is reproduced mostly but not only when running the nav2 both on physical robot and with the tb3 simulation.

- The fastest way is to run the nav2 and manually or with Behavior tree to frequently send_goal to a action server

- CTRL + C get always this msg:
`[waypoint_follower-5] [1608026728.932168403][rclcpp_action][ERROR]: Error in destruction of rcl action client handle: the Node Handle was destructed too early. You will leak memory`

#### Expected behavior
- send_goal will be successful when asked repeatedly in medium-high frequencies  
- no memory leak from nodes 
- general stability improvement
#### Actual behavior
It seems actions and topics sometimes highly unstable, for example:

- frequently action server get this error msg:
`[bt_navigator-4] [1608026552.800316163][bt_navigator][ERROR]: Action server failed while executing action callback: "send_goal failed"`

- after that happens sometimes nodes (even Rviz) stop getting msgs and stop receiving  TFs  - it requires restart the of entire ros system when it happens
- stability in general feels degraded and I require to restart the processes many times
#### Additional information
Talking to @SteveMacenski - Other users experienced similar behaviors on the ros2 master as well.

yonhdee, 2020-12-16:
me and my colleagues suffered a lot from this, since apt update/upgrade few days ago

SteveMacenski, 2020-12-16:
I see this on ROS2 master as well while doing nav2 testing. The errors don't come from Nav2 but from `rclcpp_actions` and I believe `rcl` on occasion. I think this ticket should be moved to https://github.com/ros2/rclcpp/ though.

fujitatomoya, 2020-12-17:
> `Error in destruction of rcl action client handle: the Node Handle was destructed too early. You will leak memory`

only explanation is, object pointed with `node_handle` is gone since nobody refers to it anymore.

https://github.com/ros2/rclcpp/blob/504b082d8be8c14076db631807b98a4dc2b3234a/rclcpp_action/src/client.cpp#L49

i think that using `std::weak_ptr` cannot guarantee `node_handle` object exists when deleter for `client_handle` is called. i think original author was intended to use `std::weak_ptr` to avoid memory leak from cross-reference. i don't think there would be a problem to use `shared_ptr` in this case.

but...i cannot explain why patch release 4 makes it happen more frequently, because above implementations has not been changed with patch release 4. i might be missing on somethin, i'd really like to hear from others.

> Action server failed while executing action callback: "send_goal failed"

this seems to be a message from application callback, is there any message from ros2 system library via rclcpp / rcl?

SteveMacenski, 2020-12-17:
Its not just in the patch, I see this behavior in the state of ROS2 master as of pulling about 10-14 days ago. I think its just appearing in Foxy due to the backport binary release (but it could have been going on for awhile for all I know). 

`send_goal failed` is a log from https://github.com/ros-planning/navigation2/blob/850295c0ec7b73d62efb136fc36b8a6e45a8b97c/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L233 which is basically just saying that the `spin_until_future_complete` failed due to this `rclcpp_action` issue

wjwwood, 2020-12-17:
cc @jacobperron who is the Foxy ROS Boss as well

wjwwood, 2020-12-17:
> which is basically just saying that the spin_until_future_complete failed due to this rclcpp_action issue. That's just another log expressing the symptom on the client side, I don't think it needs particular focus.

Is it not just timing out or getting interrupted? See:

https://docs.ros2.org/latest/api/rclcpp/namespacerclcpp.html#a7b4ff5f1e516740d7e11ea97fe6f5532

That check may not be sufficient.

SteveMacenski, 2020-12-17:
Its possible, but for the purposes of that particular node, it doesn't matter. That check doesn't care why it failed, just that it failed.

 I think what that message is providing is that all the sudden `rclcpp_action` went from 100% functioning to receiving these errors on a regular-ish basis as a result of `rclcpp_action` timing out or interrupted. This was not happening before the sync / recent updates, actions have been really reliable for the last year.

wjwwood, 2020-12-17:
Yeah if you guys could modify the code so we could know if it is timing out or being interrupted? That might give us more to look into. The former might indicate that performance has regressed or there are messages being lost now, but the latter might indicate a change in the shutdown of the system being a problem.

SteveMacenski, 2020-12-17:
Yes, we could definitely do that. I'm trying to trigger it now, but its been really inconsistent and I'm having a hard time reproducing regularly. 

@yonhdee @tal-grossman if you're able to reproduce more regularly (which it sounds like from your comments) please modify the code here (https://github.com/ros-planning/navigation2/blob/850295c0ec7b73d62efb136fc36b8a6e45a8b97c/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L233) to print the return code / a log.

jacobperron, 2020-12-18:
IIUC, it looks like the referenced timeout logic in bt_action_node.hpp was added since Foxy Patch 4 (https://github.com/ros-planning/navigation2/commit/96b47139d3c10f3847c66fe6bf130152dfd1a038). This might explain why we see this stability issue as of patch 4; there was no timeout logic prior.

SteveMacenski, 2020-12-18:
I see this on Nav2 master branch

SteveMacenski, 2020-12-18:
Mhm, you think there's a timing issue? That would line up, though it doesn't answer the "why now?" problem. I've been working on Nav2 master with ROS2 master and this only appeared for me very recently - well after that was implemented.

fujitatomoya, 2020-12-18:
> have you experienced this as well elsewhere?

No, actually i was trying to reproduce the issue with minimal action client/server with some change that stresses server like million times. but so far, i was unable to see any issue.

- SingleThreadedExecutor (i see Nav2 uses this one)
- action server:client = 1:N (N client processes)

<details><summary>minimal action client/server stress patch</summary><div>

```
diff --git a/rclcpp/actions/minimal_action_client/not_composable.cpp b/rclcpp/actions/minimal_action_client/not_composable.cpp
index 406a7d9..e5e72e5 100644
--- a/rclcpp/actions/minimal_action_client/not_composable.cpp
+++ b/rclcpp/actions/minimal_action_client/not_composable.cpp
@@ -33,7 +33,7 @@ int main(int argc, char ** argv)
     RCLCPP_ERROR(node->get_logger(), "Action server not available after waiting");
     return 1;
   }
-
+  for (int i = 0; i< 1000000; i++) {
   // Populate a goal
   auto goal_msg = Fibonacci::Goal();
   goal_msg.order = 10;
@@ -82,10 +82,10 @@ int main(int argc, char ** argv)
   }

   RCLCPP_INFO(node->get_logger(), "result received");
-  for (auto number : wrapped_result.result->sequence) {
-    RCLCPP_INFO(node->get_logger(), "%" PRId32, number);
+  //for (auto number : wrapped_result.result->sequence) {
+    //RCLCPP_INFO(node->get_logger(), "%" PRId32, number);
+  //}
   }
-
   rclcpp::shutdown();
   return 0;
 }
diff --git a/rclcpp/actions/minimal_action_server/not_composable.cpp b/rclcpp/actions/minimal_action_server/not_composable.cpp
index b82cbf8..81d99ab 100644
--- a/rclcpp/actions/minimal_action_server/not_composable.cpp
+++ b/rclcpp/actions/minimal_action_server/not_composable.cpp
@@ -66,10 +66,10 @@ void execute(
     // Update sequence
     sequence.push_back(sequence[i] + sequence[i - 1]);
     // Publish feedback
-    goal_handle->publish_feedback(feedback);
-    RCLCPP_INFO(rclcpp::get_logger("server"), "Publish Feedback");
+    //goal_handle->publish_feedback(feedback);
+    //RCLCPP_INFO(rclcpp::get_logger("server"), "Publish Feedback");

-    loop_rate.sleep();
+    //loop_rate.sleep();
   }

   // Check if goal is done
```　
</div></details>

tgreier, 2020-12-18:
I was able to re-create the problem on foxy branch. It happens for me almost every time I run a navigation. Full disclosure, I am using Fast-RTPS.

```
[controller_server-16] [INFO 1608263558.512489663] [controller_server]: Passing new path to controller. (/home/tom/ros2-moog-foxy-ws/src/navigation2/nav2_controller/src/nav2_controller.cpp:376)
[bt_navigator-19] rtn TIMEOUT (2)
[bt_navigator-19] Is timeout? 1
[bt_navigator-19] Is interrupted? 0
[bt_navigator-19] [ERROR 1608263559.561559769] [bt_navigator]: Action server failed while executing action callback: "send_goal failed" (/home/tom/ros2-moog-foxy-ws/install/nav2_util/include/nav2_util/simple_action_server.hpp:148)
[bt_navigator-19] [WARN 1608263559.567419569] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle. (/home/tom/ros2-moog-foxy-ws/install/nav2_util/include/nav2_util/simple_action_server.hpp:408)
```

```
    auto rtn = rclcpp::spin_until_future_complete(node_, future_goal_handle, server_timeout_);
    if (rtn != rclcpp::FutureReturnCode::SUCCESS)
    {
      std::cout << "rtn " << rtn << std::endl;      
      std::cout << "Is timeout? " << (rtn ==  rclcpp::FutureReturnCode::TIMEOUT) << std::endl;
      std::cout << "Is interrupted? " << (rtn ==  rclcpp::FutureReturnCode::INTERRUPTED) << std::endl;
      throw std::runtime_error("send_goal failed");
    }
```

fujitatomoya, 2020-12-18:
@tgreier 

thanks for checking on this. how about the following error? are you still facing it often?

> `Error in destruction of rcl action client handle: the Node Handle was destructed too early. You will leak memory`

tgreier, 2020-12-18:
@fujitatomoya I myself do not experience the 'Error in destruction of rcl action client handle' error.

SteveMacenski, 2020-12-18:
That makes sense then - so perhaps a performance regression somewhere in the `action` stack or `rmw`s? I'm curious if you switch to cyclone if you can make it happen (might give us some rough insight if its an `rmw` / DDS issue). In the short term, we can push up our action timeout but that's really a bad solution for us. The BT ticks at 100hz so that `10ms` number is actually very important to make sure that we can return a status code. 

That other error around the destruction of RCL action client is also a little scary. I don't have any insight though on what might cause that. 

@tgreier the negative effect is leaking memory, which will build up over time. This is a big problem.

wjwwood, 2020-12-18:
> In the short term, we can push up our action timeout but that's really a bad solution for us. The BT ticks at 100hz so that 10ms number is actually very important to make sure that we can return a status code.

It's possible on non-real-time systems that this may occur occasionally. It shouldn't be all the time, but requiring a 10ms deadline may occasionally fail. So you might want to relax the condition a bit, but that's up to you guys.

> That other error around the destruction of RCL action client is also a little scary. I don't have any insight though on what might cause that.

Typically that occurs when a `rclcpp::Client` is left around after a `rclcpp::Node` is already destroyed, but without more digging it's hard to say. I don't think the memory leak is that big of a deal if it is on shutdown, but the error itself is a problem.

SteveMacenski, 2020-12-18:
> It's possible on non-real-time systems that this may occur occasionally. It shouldn't be all the time, but requiring a 10ms deadline may occasionally fail. So you might want to relax the condition a bit, but that's up to you guys.

Agreed if we were running into this issue ordinarily (e.g. 10ms is really not enough time to reliably send a goal), but for the last 3 months since this was added, I haven't seen it and no one's reported it. I think 10ms isn't enough right now because there was a performance regression somewhere. That seems like the more root issue to resolve. 

> Typically that occurs when a rclcpp::Client is left around after a rclcpp::Node is already destroyed, but without more digging it's hard to say.

Well that's something we can easily test, thanks! @tgreier please add a `future_goal_handle_.reset()` here https://github.com/ros-planning/navigation2/blob/main/nav2_waypoint_follower/src/waypoint_follower.cpp#L123 before the action server reset. Does the resolve it? Also just a little odd to me that this suddenly appeared unless that error is a new addition and its been happening silently up until now.

tgreier, 2020-12-21:
In testing with CycloneDDS, not much of a different in duration of spin_until_future_complete. For both FastDDS and CycloneDDS, I'm seeing about 1-3ms average for both compute_path_to_pose and follow_path, with both occasionally reacong 5-8ms, and then less frequently above 10ms.

For resetting the future_goal_handle_, I'm still looking at how to do that since there is no reset method for std::shared_future.

SteveMacenski, 2020-12-21:
It might not be the best convention (I haven't read to much on best practices with futures), but you could `get` the `rclcpp_action::ClientGoalHandle<ClientT>::SharedPtr` and then reset the ClientGoalHandle shared pointer?

jacobperron, 2020-12-21:
Maybe I'm missing something, but I don't see what `future_goal_handle_` has to do with [this error message](https://github.com/ros2/rclcpp/blob/504b082d8be8c14076db631807b98a4dc2b3234a/rclcpp_action/src/client.cpp#L58). In the case of WaypointFollower, I would expect it to be caused by failing to destroy `nav_to_pose_client_` before the node used to create it, `client_node_`. Indeed, looking at the code, since [`nav_to_pose_client_` is declared *before* `client_node_`](https://github.com/ros-planning/navigation2/blob/e7432f17d72210b2aed71bae084c2fcda30ed888/nav2_waypoint_follower/include/nav2_waypoint_follower/waypoint_follower.hpp#L120-L121), I would expect to see the referenced error. I think swapping those two member variables should fix the error.

SteveMacenski, 2020-12-22:
You are correct, I misread `Error in destruction of rcl action client handle` as action client *goal* handle. Swapping them could help, though we do [`reset` some things](https://github.com/ros-planning/navigation2/blob/e7432f17d72210b2aed71bae084c2fcda30ed888/nav2_waypoint_follower/src/waypoint_follower.cpp#L128) in the lifecycle cleanup states, so making sure that `resets` are called in order since some of them will be destroyed before the destructor cleans up members.

jacobperron, 2020-12-22:
> Swapping them could help, though we do reset some things in the lifecycle cleanup states, so making sure that resets are called in order since some of them will be destroyed before the destructor cleans up members.

The current order in `on_cleanup` looks okay to me. My guess is that `on_cleanup` is not always being called before shutdown.

tgreier, 2021-01-20:
@tal-grossman @SteveMacenski Are you still experiencing this problem:

- frequently action server get this error msg:

`[bt_navigator-4] [1608026552.800316163][bt_navigator][ERROR]: Action server failed while executing action callback: "send_goal failed"`

tal-grossman, 2021-01-21:
> @tal-grossman @SteveMacenski Are you still experiencing this problem:
> 
> * frequently action server get this error msg:
> 
> `[bt_navigator-4] [1608026552.800316163][bt_navigator][ERROR]: Action server failed while executing action callback: "send_goal failed"`

yes. It happens more frequently on my robot with low cost on aarch64 architecture due to the action_server_timeout exceeded - according to tests I did sometimes it gets over 50 ms and sometimes 4 ms probably depending on system load , but it also happens on my i7, just less frequently.  
after it happens I can't send another goal and most of the time the process crashes. when I restart only the nav2 stack, I experience poor subscribers connections and can't get any tf data.

The patch I did for now is to set high `action_server_timeout`

SteveMacenski, 2021-01-21:
I'm still seeing it on my normal developer machine as a new regression

farbod1277, 2021-01-22:
I was getting the "send_goal failed" error also intermittently and as far as I can tell stochastically. I am running nav2 on a Nvidia Xavier ARM computer so I expect its CPU to be quite slow, yet this issue only happens seemingly at random and not too often. But when it does occur, it keeps occurring for a good while very consistently (sometimes even after a power cycle). So what I did was try the ad hoc solution suggested here which is to increase the timeout of the action_server. I initially set it to 500ms and ran a series of tests to see if I can get the failure reproduced. I tried power cycles, node resets etc. but it didn't fail at all. Next I systematically did the same thing for a timeout of 10ms but again could not get it to fail to my surprise. I did 27 trials of goal planning, none failed! Then I tried setting the timer to 1ms, and was expecting it to timeout every time but again nothing, it ran fine without failing even once (ran 27 tests again). As far as I can tell it could fail tomorrow again randomly! 

> after it happens I can't send another goal and most of the time the process crashes. when I restart only the nav2 stack, I experience poor subscribers connections and can't get any tf data.

I did however experience the issue mentioned above, even though I didn't get a "send_goal failed" error. If I run my system, then run nav2 it works fine. If I then stop nav2, then wait a few seconds and run nav2 again, I get the tf issues, which means the whole system would need to be reset. Again this was without the "send_goal" failed error.

Hope this new information helps in solving the issue.

SteveMacenski, 2021-01-22:
> this issue only happens seemingly at random and not too often

I see this on my 8th gen i7 Intel CPU, one of the reasons its hard for me to "know" if it went away. Its not every day, but certainly at least once a week (so probably once every 2-3 hours of simulation time). The issue I think is a performance regression in the action/service such that things are taking longer than they used to and triggering the timeout that previously we never saw hit.

SteveMacenski, 2021-02-11:
I have a new report from another party (@shrijit99) working with ROS2 rolling and master builds of everything else still experiencing this. 

```
[controller_server-7] [INFO] [1613072627.935646344] [controller_server]: Received a goal, begin computing control effort.
[bt_navigator-10] [ERROR] [1613072628.952695922] [bt_navigator]: Action server failed while executing action callback: "send_goal failed"
[bt_navigator-10] [WARN] [1613072628.952760545] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
```

SteveMacenski, 2021-03-05:
https://app.circleci.com/pipelines/github/ros-planning/navigation2/4783/workflows/290b6e0c-2ba1-497b-9b1a-b9f887340305/jobs/17752

I can also reproduce this in our CI system as the cause of our recent flakiness 

```
[tester_node-16] [INFO] [1614915073.782328241] [nav2_tester]: ?[1;37;44mReceived plan?[0m
[bt_navigator-13] [ERROR] [1614915073.802908955] [bt_navigator]: Action server failed while executing action callback: "send_goal failed"
[bt_navigator-13] [WARN] [1614915073.803030510] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[tester_node-16] [INFO] [1614915073.804063332] [nav2_tester]: ?[1;37;44mGoal failed with status code: 6?[0m
[tester_node-16] [ERROR] [1614915073.804435140] [nav2_tester]: ?[1;37;41mTest FAILED?[0m
```

jacobperron, 2021-03-08:
Unfortunately, it doesn't seem I have access to that CI link without signing up.

A reproduction of the issue without nav2 in the loop would be very helpful.

SteveMacenski, 2021-03-08:
I can't get it to happen deterministically, I'm usually only seeing this when there's CPU load on the system (like when running something like Nav2 or a robot stack with the full system going)

Sunshengjin, 2021-03-12:
Is there any progress about this issue? I'm currently suffering on this problem

fujitatomoya, 2021-03-12:
> Error in destruction of rcl action client handle: the Node Handle was destructed too early. You will leak memory

i think this problem should be fixed by https://github.com/ros2/rclcpp/pull/1562 (foxy backport is done too: https://github.com/ros2/rclcpp/pull/1565) they should be available for next release.

> Action server failed while executing action callback: "send_goal failed"

but i am not sure on this one, sorry.

JetBelson, 2021-03-28:
Machine: macOS BigSur 11.2.3 
CPU: 2.6 GHz 6-Core Intel Core i7
GPU: AMD Radeon Pro 5500M 8 GB
Memory: 32 GB
System: VMware + Ubuntu 20.04 + ROS2 Foxy

When I tried the Turtlebot3 simulation examples, everything went well until the nav2 navigation example. 
I can only move the turtlebot3 on the first try. Then errors popped up. I think it may be also related to this problem. 
Below is the terminal message for your reference.

> 
> [rviz2-10] [INFO] [1616906292.733727171] [rviz2]: Waiting for the lifecycle_manager_localization/is_active service...
> [rviz2-10] [INFO] [1616906292.733865665] [rviz2]: Sending lifecycle_manager_localization/is_active request
> [rviz2-10] Start navigation
> [bt_navigator-7] [INFO] [1616906299.096649212] [bt_navigator]: Begin navigating from current location to (-0.05, -0.01)
> [bt_navigator-7] [ERROR] [1616906299.159924996] [bt_navigator]: Action server failed while executing action callback: "send_goal failed"
> [bt_navigator-7] [WARN] [1616906299.161662188] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
> [controller_server-4] [INFO] [1616906299.158912485] [controller_server]: Received a goal, begin computing control effort.
> [controller_server-4] [WARN] [1616906299.763552626] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [controller_server-4] [WARN] [1616906303.467358350] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [controller_server-4] [WARN] [1616906309.187181269] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [rviz2-10] [INFO] [1616906309.447838318] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 14.238 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906309.595741117] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.252 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906309.879611996] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 14.438 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.065911042] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.352 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.066023616] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.465 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.350089643] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 14.638 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.431413933] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.673 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.431464544] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.545 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.431495815] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.622 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.819238558] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 14.839 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906310.899310722] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.847 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906311.311750674] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 15.040 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906311.370901623] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.993 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906311.370963666] [rviz2]: Message Filter dropping message: frame 'odom' at time 15.902 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906311.698626497] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 15.239 for reason 'Unknown'
> [controller_server-4] [WARN] [1616906311.763822124] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [rviz2-10] [INFO] [1616906312.150797764] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 15.438 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906312.596757979] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 15.638 for reason 'Unknown'
> [rviz2-10] [INFO] [1616906313.029920250] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 15.847 for reason 'Unknown'
> [controller_server-4] [WARN] [1616906316.641791944] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [controller_server-4] [WARN] [1616906316.663212186] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [controller_server-4] [WARN] [1616906316.743138179] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [controller_server-4] [WARN] [1616906318.472080475] [controller_server]: Control loop missed its desired rate of 20.0000Hz
> [controller_server-4] [ERROR] [1616906320.013284992] [controller_server]: Failed to make progress
> [controller_server-4] [WARN] [1616906320.013459671] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
> [rviz2-10] [INFO] [1616906459.058002708] [rviz2]: Setting estimate pose: Frame:map, Position(1.29329, -1.27819, 0), Orientation(0, 0, 0.00693619, 0.999976) = Angle: 0.0138725
> [amcl-2] [INFO] [1616906459.062779084] [amcl]: initialPoseReceived
> [amcl-2] [INFO] [1616906459.062839172] [amcl]: Setting pose (93.139000): 1.293 -1.278 0.014
> [rviz2-10] Start navigation
> [bt_navigator-7] [INFO] [1616906464.166274111] [bt_navigator]: Begin navigating from current location to (2.25, -1.56)
[... cut ...]
> [rviz2-10] [INFO] [1616906567.881417064] [rviz2]: Setting estimate pose: Frame:map, Position(0.88138, 1.95489, 0), Orientation(0, 0, -0.00670501, 0.999978) = Angle: -0.0134101
> [amcl-2] [INFO] [1616906567.882148644] [amcl]: initialPoseReceived
> [amcl-2] [INFO] [1616906567.882193823] [amcl]: Setting pose (148.264000): 0.881 1.955 -0.013
> [rviz2-10] Start navigation
> [rviz2-10] [ERROR] [1616906579.131547087] [rviz2]: FollowWaypoints action server is not available. Is the initial pose set?
>

SteveMacenski, 2021-03-30:
I'll try to look into this again - but it looks to me like the same issue last time I looked into it (some performance regression in ROS2 actions / services) https://github.com/ros-planning/navigation2/blob/3bf1f7a93d740c54b4c7ef7fc6dd914175bf46be/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L280 this is timing out due to a `10ms` timeout that wasn't an issue prior to a sync of Foxy / master branches.

charlielito, 2021-05-06:
Machine: NVIDIA Jetson Xavier AGX
Memory: 32 GB
Power mode: 30W 8 ARM cores
System: [Balena](https://www.balena.io/) + Docker + Ubuntu 20.04 + ROS2 Foxy + Navigation2 foxy-devel branch

I am facing the same issue when running navigation2 stack in the described hardware. I tried both Cyclone and Fast-RTPS. I had to make the workaround suggested in this thread to make it work (removed the `server_timeout_` argument to `spin_until_future_complete`). On the first navigation goal, on average the time taken is more than 10 ms, which would have trigger the error. After that the time is on average < 10 ms with some spikes occasionally above 10 ms.

For example, the output:

```
[lifecycle_manager-6] [INFO][1620160597.176711075]: Managed nodes are active
[planner_server-2] [WARN][1620160598.138558169]: Sensor origin at (-0.12, 0.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN][1620160599.138179109]: Sensor origin at (-0.12, 0.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [INFO][1620160599.650320121]: StaticLayer: Resizing costmap to 224 X 106 at 0.050000 m/pix
[bt_navigator-4] [INFO][1620160639.933032534]: Begin navigating from current location to (0.64, 0.11)
[bt_navigator-4] Elapsed: 18.4945 ms
[bt_navigator-4] Elapsed: 2.37796 ms
[controller_server-1] [INFO][1620160639.989455002]: Received a goal, begin computing control effort.
[bt_navigator-4] Elapsed: 5.16832 ms
[bt_navigator-4] Elapsed: 2.97559 ms
[controller_server-1] [INFO][1620160641.041291392]: Passing new path to controller.
```

SteveMacenski, 2021-07-15:
As a general update because I know this is still coming up. The node pointer error is resolved in https://github.com/ros2/rclcpp/pull/1562. The `Action server failed while executing action callback: "send_goal failed"` error has not been resolved, but in Nav2 we did add larger timeouts to essentially try to work around the underlying issue. This is also exposed to the user now to increase as they need it, but there is still no substitute for actions actually being submitted in a timely manner. That is still an issue because that adds a maximum limit in how fast you can possible call actions of ~10hz semi-reliably, and probably 5hz absolutely reliably.

ahopsu, 2021-08-11:
> yes. It happens more frequently on my robot with low cost on aarch64 architecture due to the action_server_timeout exceeded - according to tests I did sometimes it gets over 50 ms and sometimes 4 ms [...]

Exactly the same problem here. I tried to dig out, but as a beginner in ROS world, I did not understand, where exactly I need to change the timeout value / make other necessary modifications? As it seems to solve the issue temporarily? Is this problem by the way solved in Galactic?

xaru8145, 2023-08-15:
Is this issue solved in Humble? Thanks

clalancette, 2024-09-11:
Since this was filed against Foxy, which is long out of support, I'm going to close this out.  If you are still experiencing problems on Humble or newer, please feel free to open a new bug describing the issue you are seeing, and we can take a look.