ros2/ros2 #1075: Using spcific interface

liyue75, 2020-12-17:
I want to use ros2 with fastdds configuration xml file to specify network interface used by transport.    
In my xml file, I add address 192.168.1.6 which is my network card's ip address to interfaceWhileList, then I run   
a pub and a sub on two terminals using this xml file and they work fine.  
But if I modify the ip address in interfaceWhileList to 127.0.0.1, this time the pub and sub can't work unless I set the network  card which ip is 192.168.1.6 down.  
Why can't I specify localhost to transport when other interface exists?  In fastdds, it works. 
My system: ubuntu18.04 
ros2 version: dashing
dds: fastdds

clalancette, 2021-01-04:
@MiguelCompany @richiware any thoughts on what might be going on here?  Notice that this is on Dashing, so an older version of Fast-RTPS (1.8.x) is in use.

EduPonz, 2021-01-07:
Hi @liyue75 ,

I have two questions regarding your issue:
1. How are you loading the XML file within the ROS 2 stack?
1. You say that with standalone Fast DDS there is no problem (there really shouldn't). Which version of Fast DDS have you used in the standalone test?

Thanks!

liyue75, 2021-01-07:
@EduPonz Thank you for your reply. 
Here is my answer to your question:
1. My way to load XML file is: **export FASTRTPS_DEFAULT_PROFILES_FILE=<path_to_xml_file>** as [https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html](url) says.
2. Fast DDS's version is v2.1.0.

liyue75, 2021-01-07:
@EduPonz 
And I find something strange about interfaceWhitelist in Fast DDS v2.1.0.
There are two hosts in the LAN. They all have wired and wireless cards. Here is their ip schematic diagram:
|  |Host1(pub) | Host2(sub)
---------|------------ | -------------
|eth0(wired card) |192.168.1.5 |192.168.1.6
|wlp0(wireless card) |192.168.0.115 | 192.168.0.144

And they can ping each other no matter which network card specify, for example, from Host1 I can **ping 192.168.0.144 -I eth0 or ping 192.168.0.144 -I wlp0**.

Here's the problem.  If I set useBuiltinTransports=false and  specify 192.168.1.5(eth0) in interfaceWhiteList on **pub**, 
and set useBuiltinTransports=flase and specify 192.168.0.144(wlp0) in interfaceWhiteList on **sub**.

The pub can send data to sub, but I find the interface the pub used to send data is wlp0.
I specify the eth0 on host, but the actual interface used is wlp0. 
It seems the interfaceWhiteListe didn't work.

iuhilnehc-ynos, 2021-01-11:
@liyue75 

IIUC, I think that we should use `<type>UDPv4</type>` instead of `<type>TCPv4</type>`.

liyue75, 2021-01-11:
@iuhilnehc-ynos,
Yes, it's UDPv4, I spelled it wrong in this issue. But in xml file in ros2, it is UDPv4.

iuhilnehc-ynos, 2021-01-11:
dashing with `Fast DDS v2.1.0`? so you first installed `ros-dashing-ros-base/ros-dashing-desktop`, and then build `Fast-DDS` with tag v2.1.0, right? (If yes, I'll continue to try to reproduce this issue.)

For `localhost` used in the white list, I tested with `ros2 master`(https://github.com/ros2/ros2/pull/1058/files), it worked well.
BTW: https://github.com/ros2/ros2/pull/1058/files IS NOT MERGED even if in `ros2 master`. I am a little curious about why you use `dashing` with `Fast DDS v2.1.0`.

<details><summary> log </summary>
<p>

[image]

- whitelist.xml
```shell
<?xml version="1.0" encoding="UTF-8" ?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
    <profiles>
    <transport_descriptors>
        <transport_descriptor>
            <transport_id>mytest</transport_id>
            <type>UDPv4</type>
            <interfaceWhiteList>
                <address>127.0.0.1</address>
            </interfaceWhiteList>
        </transport_descriptor>
    </transport_descriptors>
    <participant profile_name="myparticipant" is_default_profile="true">
        <rtps>
            <userTransports>
            <transport_id>mytest</transport_id>
            </userTransports>
            <useBuiltinTransports>false</useBuiltinTransports>
        </rtps>
    </participant>
    </profiles>
</dds>
```

```
chenlh@OptiPlex-990:~$ ifconfig
eno1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.100  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::dafc:9133:f829:aa35  prefixlen 64  scopeid 0x20<link>
        inet6 2403:700:0:20ad:60a8:dcde:5f2c:d323  prefixlen 64  scopeid 0x0<global>
        inet6 2403:700:0:20ad:d158:e5a7:ecab:fd81  prefixlen 64  scopeid 0x0<global>
        inet6 2403:700:0:20ad:ec87:b7a5:babe:de0c  prefixlen 64  scopeid 0x0<global>
        inet6 2403:700:0:20ad:8401:eb5b:f194:ef11  prefixlen 64  scopeid 0x0<global>
        inet6 2403:700:0:20ad:64fd:59b2:785b:8532  prefixlen 64  scopeid 0x0<global>
        ether 90:b1:1c:77:e9:70  txqueuelen 1000  (Ethernet)
        RX packets 6174694  bytes 2475928056 (2.4 GB)
        RX errors 0  dropped 164255  overruns 0  frame 0
        TX packets 8779446  bytes 11391541715 (11.3 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 20  memory 0xe1a00000-e1a20000  

enx00808e8a2424: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:80:8e:8a:24:24  txqueuelen 1000  (Ethernet)
        RX packets 590389  bytes 85544742 (85.5 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1150132  bytes 146864433 (146.8 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 529465  bytes 10729275760 (10.7 GB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 529465  bytes 10729275760 (10.7 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

chenlh@OptiPlex-990:~$ ps aux | grep demo_nodes_cpp
chenlh    359777  0.0  0.2  31880 19624 pts/1    S+   15:20   0:00 /usr/bin/python3 /home/chenlh/ROS2/local_install/bin/ros2 run demo_nodes_cpp talker
chenlh    359778  0.1  0.4 441956 37080 pts/1    Sl+  15:20   0:00 /home/chenlh/ROS2/local_install/lib/demo_nodes_cpp/talker
chenlh    359794  0.0  0.2  31496 19252 pts/3    S+   15:20   0:00 /usr/bin/python3 /home/chenlh/ROS2/local_install/bin/ros2 run demo_nodes_cpp listener
chenlh    359795  0.1  0.4 441956 37576 pts/3    Sl+  15:20   0:00 /home/chenlh/ROS2/local_install/lib/demo_nodes_cpp/listener
chenlh    360035  0.0  0.0  12104  2540 pts/4    S+   15:29   0:00 grep --color=auto demo_nodes_cpp
chenlh@OptiPlex-990:~$ cat /proc/359778/maps | grep libfastrtps
7f8d0012d000-7f8d0085d000 r--p 00000000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7f8d0085d000-7f8d00fe4000 r-xp 00730000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7f8d00fe4000-7f8d01242000 r--p 00eb7000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7f8d01242000-7f8d01253000 r--p 01114000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7f8d01253000-7f8d0129a000 rw-p 01125000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
chenlh@OptiPlex-990:~$ cat /proc/359795/maps | grep libfastrtps
7feb1d7b4000-7feb1dee4000 r--p 00000000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7feb1dee4000-7feb1e66b000 r-xp 00730000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7feb1e66b000-7feb1e8c9000 r--p 00eb7000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7feb1e8c9000-7feb1e8da000 r--p 01114000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0
7feb1e8da000-7feb1e921000 rw-p 01125000 08:05 2238231                    /home/chenlh/ROS2/local_install/lib/libfastrtps.so.2.1.0

```
</p>
</details>

liyue75, 2021-01-11:
@iuhilnehc-ynos , thank you for your reply.
I used dashing with default fastrtps version 1.8.4.
And I used FastDDS v2.1.0 alone to test the interfaceWhiteList.
Maybe as you tested, if I want to use localhost whitelist I should use latest version of ros2 whose fastrtps' version is v2.1 instead of dashing.

iuhilnehc-ynos, 2021-01-11:
Oh, I see. I think FastDDS v1.8.5 will fix this issue for dashing soon.

iuhilnehc-ynos, 2021-01-11:
Sorry, I can't reproduce the issue about the localhost whitelist on `ros:dashing` docker image, which also seems working.

EduPonz, 2021-01-11:
Hi @liyue75 ,

Could you maybe provide a walk-through example on how de you reproduce the issue both for ROS 2 Dashing and Fast DDS v2.1.0?

liyue75, 2021-01-12:
I'm tring to run the program again, please wait.

EduPonz, 2021-01-12:
No problem, we are here to help! Just ping me when you can. Again, thanks everyone for the iterations on the problem!

liyue75, 2021-01-12:
@EduPonz @iuhilnehc-ynos 
Thank you for your help. I‘m very sorry, the interfaceWhiteList seems to be working fine. 
I tried to rerun the ros2 program and now it can specify the local interface. Maybe my previous program had a problem.
I also tried to run fastddsv2.1.0,  which can also sends data from the specified interface eth0, the problem would probably caused by route setup.
So the whilelist seems working well, sorry to bother you.

But there is another little problem when I running ros2 official example and using whilelist in docker.
Here are the steps to reproduce:
turn off wired and wireless ethernet cards:
  sudo ip link set eth0 down
  sudo ip link set wlp0 down
In one terminal:
  1. docker run -it --rm --network=host ros2 /bin/bash
  2. export FASTRTPS_DEFAULT_PROFILES_FILE=<path_to_xml_file>
  3. ros2 run example_rclcpp_minimal_service service_main
In another terminal:
  1.docker exec -it <container id> /bin/bash
  2. export FASTRTPS_DEFAULT_PROFILES_FILE=<path_to_xml_file>
  3. ros2 run example_rclcpp_minimal_client client_main

The example can work, 
the service will outputs: [INFO] [minimal_service]: request: 41 + 1
and the client will outputs: [INFO] [minimal_client]: result of 41 + 1 = 42
But when they exit and return, both of they will output: 
2021-01-12  14:26:25.581 [RTPS_MSG_OUT Error]  Error: send_to: Network is unreachable -> Function ReleaseInputChannel
This error only occurs when I close down all network card on this computer, and exit the program.
 It seems not a big problem but I wonder why.

EduPonz, 2021-01-12:
Hi @liyue75 ,

> Thank you for your help. I‘m very sorry, the interfaceWhiteList seems to be working fine.

I'm happy to hear that everything works as expected.

> But when they exit, both of they will output:
> 2021-01-12 14:26:25.581 [RTPS_MSG_OUT Error] Error: send_to: Network is unreachable -> Function ReleaseInputChannel
> This error only occurs when I close down all network card on the computer, specify interface 127.0.0.1 and exit the program.
> It seems not a big problem but I wonder why.

I'm not sure I'm understanding your steps correctly, please correct me if I'm wrong:

1. You configure a whitelist including only 127.0.0.1 in both containers
2. You run your application in both containers
3. You close down al network interfaces
4. You exit the applications

Is that correct? I that case, it'd make sense to see that log, since the interface used for communication does not exists anymore. If this is not the case, I think it'd be best to mark this ticket as solved and open a new one with step-by-step instructions to replicate the issue. This way we can keep a better track of it, and other people experiencing the same will have an easier time finding a solution.

Thanks for the iterations and your collaboration!

liyue75, 2021-01-12:
@EduPonz 
Yes, it is. I close down all network interfaces, but the lo is still there, data can transfer locally.

EduPonz, 2021-01-13:
Hi @liyue75 ,

The log you are seeing originates [here](https://github.com/eProsima/Fast-DDS/blob/v1.8.0/src/cpp/transport/UDPTransportInterface.cpp#L482). However, that means that you're using Fast DDS v1.8.0 or prior, since `ReleaseInputChannel` was removed in v1.8.1. v1.8.0 is rather old even for ROS 2 Dashing. I suggest to move to v1.8.4 and give it another shot.

liyue75, 2021-01-13:
@EduPonz Thanks a lot!

liyue75, 2021-01-13:
@EduPonz 
Is it right that FASTDDS v2.1 in latest ros2 can use lo by default if pubs find subs are on the same host?

EduPonz, 2021-01-13:
Hi @liyue75 ,

Fast DDS v2.1.0 is the version you'd find in ROS 2 Rolling when compiling from sources. ROS 2 Foxy ships v2.0.2 (although when compiling from sources, you could set v2.1.0). In any case, they both can be used to transmit on loopback when on the same host. However, this is not the default behaviour, as these versions feature a shared memory transport that boost performance avoiding the network stack. It is still possible to force the loopback of course. For that, you'd need to create a custom UDP transport specifying the whitelist, and then you'll need to disable the builtin transports. You can take a look at:

https://fast-rtps.docs.eprosima.com/en/v2.1.0/fastdds/transport/udp/udp.html#enabling-udp-transport
https://fast-rtps.docs.eprosima.com/en/v2.1.0/fastdds/transport/whitelist.html

If you need further assistance, please do let me know.

liyue75, 2021-01-13:
@EduPonz 
Oh, I see, thank you!

closed 2021-01-13 by liyue75