ros2/ros2 #1077: A questions about xacro command on OSX

Ice-Panda, 2020-12-25:
When I migrated the project from ROS to ROS2, I found that I could not use the xacro command, How do I use xacro command like ROS1 on OSX?

emersonknapp, 2020-12-28:
Hi! This question sounds better suited to [ROS Answers](https://answers.ros.org/questions/) - where the community may be able to help you. I would suggest being much more specific in that question as well, answering some of the following questions:
* What ROS 2 distro are you trying to use?
* How are you installing ROS 2 (build from source, binaries, etc)?
* What is the sequence of commands you run that does not work?

clalancette, 2021-01-04:
Agreed, this sounds like a question for ROS Answers.  As such, I'm going to close this out.

closed 2021-01-04 by clalancette

Ice-Panda, 2021-01-04:
Oh! At first I thought it was a system problem so I didn't post the ROS version.
Now I have solved this problem by manually installing the corresponding version of xacro.
The reason is that xacro installed by default on python3.8 has some problems. 
 [https://github.com/ros-gbp/xacro-release/tree/release/foxy/xacro](url)