ros2/ros2 #1078: The error when pip install the pygraphviz

Gatsby23, 2021-01-06:
ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python@3.9/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/setup.py'"'"'; __file__='"'"'/private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-wheel-hxhvh5vt
       cwd: /private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/
  Complete output (59 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-11-x86_64-3.9
  creating build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/release.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/version.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/graphviz.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/__init__.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/agraph.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  creating build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_setup.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_unicode.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_readwrite.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_string.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/__init__.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_html.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_node_attributes.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_drawing.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_subgraph.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_close.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_edge_attributes.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_clear.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_layout.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_attribute_defaults.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_graph.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_attributes.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  running egg_info
  writing pygraphviz.egg-info/PKG-INFO
  writing dependency_links to pygraphviz.egg-info/dependency_links.txt
  writing top-level names to pygraphviz.egg-info/top_level.txt
  reading manifest file 'pygraphviz.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.png' under directory 'doc'
  warning: no files found matching '*.html' under directory 'doc'
  warning: no files found matching '*.txt' under directory 'doc'
  warning: no files found matching '*.css' under directory 'doc'
  warning: no previously-included files matching '*~' found anywhere in distribution
  warning: no previously-included files matching '*.pyc' found anywhere in distribution
  warning: no previously-included files matching '.svn' found anywhere in distribution
  no previously-included directories found matching 'doc/build'
  writing manifest file 'pygraphviz.egg-info/SOURCES.txt'
  copying pygraphviz/graphviz.i -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/graphviz_wrap.c -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  running build_ext
  building 'pygraphviz._graphviz' extension
  creating build/temp.macosx-11-x86_64-3.9
  creating build/temp.macosx-11-x86_64-3.9/pygraphviz
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/usr/local/include -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/usr/local/opt/openblas/include -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python@3.9/3.9.1_4/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c pygraphviz/graphviz_wrap.c -o build/temp.macosx-11-x86_64-3.9/pygraphviz/graphviz_wrap.o
  pygraphviz/graphviz_wrap.c:3047:12: warning: incompatible pointer to integer conversion returning 'Agsym_t *' (aka 'struct Agsym_s *') from a function with result type 'int' [-Wint-conversion]
      return agattr(g, kind, name, val);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
  1 warning generated.
  clang -bundle -undefined dynamic_lookup -L/usr/local/lib -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -L/usr/local/Cellar/python@3.9/3.9.1/lib -I/usr/local/opt/openblas/include build/temp.macosx-11-x86_64-3.9/pygraphviz/graphviz_wrap.o -L/usr/local/lib -L/usr/local/opt/openssl@1.1/lib -L/usr/local/opt/sqlite/lib -lcdt -lcgraph -o build/lib.macosx-11-x86_64-3.9/pygraphviz/_graphviz.cpython-39-darwin.so
  ld: library not found for -lSystem
  clang-11: error: linker command failed with exit code 1 (use -v to see invocation)
  error: command '/usr/local/opt/llvm/bin/clang' failed with exit code 1
  ----------------------------------------
  ERROR: Failed building wheel for pygraphviz
  Running setup.py clean for pygraphviz
Failed to build pygraphviz
Installing collected packages: pygraphviz
    Running setup.py install for pygraphviz ... error
    ERROR: Command errored out with exit status 1:
[... 44 lines cut ...]
    warning: no files found matching '*.html' under directory 'doc'
    warning: no files found matching '*.txt' under directory 'doc'
    warning: no files found matching '*.css' under directory 'doc'
    warning: no previously-included files matching '*~' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '.svn' found anywhere in distribution
    no previously-included directories found matching 'doc/build'
    writing manifest file 'pygraphviz.egg-info/SOURCES.txt'
    copying pygraphviz/graphviz.i -> build/lib.macosx-11-x86_64-3.9/pygraphviz
    copying pygraphviz/graphviz_wrap.c -> build/lib.macosx-11-x86_64-3.9/pygraphviz
    running build_ext
    building 'pygraphviz._graphviz' extension
    creating build/temp.macosx-11-x86_64-3.9
    creating build/temp.macosx-11-x86_64-3.9/pygraphviz
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/usr/local/include -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/usr/local/opt/openblas/include -I/usr/local/Cellar/graphviz/2.44.1/include/graphviz:/usr/local/Cellar/graphviz/2.44.1/include -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python@3.9/3.9.1_4/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c pygraphviz/graphviz_wrap.c -o build/temp.macosx-11-x86_64-3.9/pygraphviz/graphviz_wrap.o
    pygraphviz/graphviz_wrap.c:3047:12: warning: incompatible pointer to integer conversion returning 'Agsym_t *' (aka 'struct Agsym_s *') from a function with result type 'int' [-Wint-conversion]
        return agattr(g, kind, name, val);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~
    1 warning generated.
    clang -bundle -undefined dynamic_lookup -L/usr/local/lib -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -L/usr/local/Cellar/python@3.9/3.9.1/lib -I/usr/local/opt/openblas/include build/temp.m [... cut ...] n/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-record-p9zbjq7y/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/pygraphviz Check the logs for full command output.

Gatsby23, 2021-01-06:
```
ERROR: Command errored out with exit status 1:
   command: /usr/local/opt/python@3.9/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/setup.py'"'"'; __file__='"'"'/private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-wheel-hxhvh5vt
       cwd: /private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/
  Complete output (59 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-11-x86_64-3.9
  creating build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/release.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/version.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/graphviz.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/__init__.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/agraph.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  creating build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_setup.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_unicode.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_readwrite.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_string.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/__init__.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_html.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_node_attributes.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_drawing.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_subgraph.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_close.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_edge_attributes.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_clear.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_layout.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_attribute_defaults.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_graph.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  copying pygraphviz/tests/test_attributes.py -> build/lib.macosx-11-x86_64-3.9/pygraphviz/tests
  running egg_info
  writing pygraphviz.egg-info/PKG-INFO
  writing dependency_links to pygraphviz.egg-info/dependency_links.txt
  writing top-level names to pygraphviz.egg-info/top_level.txt
  reading manifest file 'pygraphviz.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching '*.png' under directory 'doc'
  warning: no files found matching '*.html' under directory 'doc'
  warning: no files found matching '*.txt' under directory 'doc'
  warning: no files found matching '*.css' under directory 'doc'
  warning: no previously-included files matching '*~' found anywhere in distribution
  warning: no previously-included files matching '*.pyc' found anywhere in distribution
  warning: no previously-included files matching '.svn' found anywhere in distribution
  no previously-included directories found matching 'doc/build'
  writing manifest file 'pygraphviz.egg-info/SOURCES.txt'
  copying pygraphviz/graphviz.i -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  copying pygraphviz/graphviz_wrap.c -> build/lib.macosx-11-x86_64-3.9/pygraphviz
  running build_ext
  building 'pygraphviz._graphviz' extension
  creating build/temp.macosx-11-x86_64-3.9
  creating build/temp.macosx-11-x86_64-3.9/pygraphviz
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/usr/local/include -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/usr/local/opt/openblas/include -I/usr/local/include -I/usr/local/opt/openssl@1.1/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python@3.9/3.9.1_4/Frameworks/Python.framework/Versions/3.9/include/python3.9 -c pygraphviz/graphviz_wrap.c -o build/temp.macosx-11-x86_64-3.9/pygraphviz/graphviz_wrap.o
  pygraphviz/graphviz_wrap.c:3047:12: warning: incompatible pointer to integer conversion returning 'Agsym_t *' (aka 'struct Agsym_s *') from a function with result type 'int' [-Wint-conversion]
      return agattr(g, kind, name, val);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
  1 warning generated.
[... cut ...] pip-install-s12afp4o/pygraphviz_04d902a71217437394a59a7e14f803ca/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/37/34tkdx1505nch17mpqh5zzr00000gn/T/pip-record-p9zbjq7y/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.9/pygraphviz Check the logs for full command output.
```

zmk5, 2021-01-06:
```
ld: library not found for -lSystem
```

This error seems to be the issue. What version of pip are your running? I heard updating pip using `python3 -m pip install -U pip` can sometimes fix these issues.

Gatsby23, 2021-01-06:
> ```
> ld: library not found for -lSystem
> ```
> 
> This error seems to be the issue. What version of pip are your running? I heard updating pip using `python3 -m pip install -U pip` can sometimes fix these issues.

[image]

Gatsby23, 2021-01-06:
> ```
> ld: library not found for -lSystem
> ```
> 
> This error seems to be the issue. What version of pip are your running? I heard updating pip using `python3 -m pip install -U pip` can sometimes fix these issues.

I'm sorry, I don't know what's wrong.

Gatsby23, 2021-01-06:
> ```
> ld: library not found for -lSystem
> ```
> 
> This error seems to be the issue. What version of pip are your running? I heard updating pip using `python3 -m pip install -U pip` can sometimes fix these issues.

The main problem is that when I update the mac osx, it also should reinstall command_line_tools. But there are some stderr, should I figure out what problem it is and solve it?
[image]

zmk5, 2021-01-06:
Cool! I should have mentioned I reinstalled the command line tools when I updated Big Sur. Don't worry about the `stderr` you see during the `colcon build` phase. Right under the one you posted just states that it is a warning and not an error. A real error would have stopped compilation!

AkhilGKrishnan, 2021-01-06:
Building wheel for psycopg2 (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/akhil/Projects/GekayRealEstate/env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/ct/r47t7v0n4sd5w56n18gfht0r0000gn/T/pip-install-ajd0j8gf/psycopg2_43dfb55858574cbfa3862380c3455f6a/setup.py'"'"'; __file__='"'"'/private/var/folders/ct/r47t7v0n4sd5w56n18gfht0r0000gn/T/pip-install-ajd0j8gf/psycopg2_43dfb55858574cbfa3862380c3455f6a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/ct/r47t7v0n4sd5w56n18gfht0r0000gn/T/pip-wheel-ecybs8rw
       cwd: /private/var/folders/ct/r47t7v0n4sd5w56n18gfht0r0000gn/T/pip-install-ajd0j8gf/psycopg2_43dfb55858574cbfa3862380c3455f6a/
  Complete output (81 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.macosx-10.14.6-x86_64-3.8
  creating build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/_json.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/extras.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/errorcodes.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/tz.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/_range.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/_ipaddress.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/__init__.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/psycopg1.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/extensions.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/sql.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  copying lib/pool.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2
  creating build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_transaction.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/dbapi20.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_extras_dictcursor.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_with.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_types_basic.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_bug_gc.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_module.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_psycopg2_dbapi20.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_async.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_dates.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_async_keyword.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/testutils.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_connection.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_copy.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_bugX000.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/__init__.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_cursor.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_types_extras.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_sql.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_fast_executemany.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_green.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_ipaddress.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_cancel.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_quote.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/testconfig.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_errcodes.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_replication.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_lobject.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/test_notify.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  copying tests/dbapi20_tpc.py -> build/lib.macosx-10.14.6-x86_64-3.8/psycopg2/tests
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  building 'psycopg2._psycopg' extension
  creating build/temp.macosx-10.14.6-x86_64-3.8
  creating build/temp.macosx-10.14.6-x86_64-3.8/psycopg
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/Headers -arch arm64 -arch x86_64 -DPSYCOPG_DEFAULT_PYDATETIME=1 -DPSYCOPG_VERSION=2.7.7 (dt dec pq3 ext lo64) -DPG_VERSION_NUM=130001 -DHAVE_LO64=1 -I/Users/akhil/Projects/GekayRealEstate/env/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -I. -I/usr/local/include -I/usr/local/ [... cut ...] tate/env/include -I/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/include/python3.8 -c ast27/Parser/acceler.c -o build/temp.macosx-10.14.6-x86_64-3.8/ast27/Parser/acceler.o
  ast27/Parser/acceler.c:13:10: fatal error: 'pgenheaders.h' file not found
  #include "pgenheaders.h"
           ^~~~~~~~~~~~~~~
  1 error generated.
  error: command 'clang' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for typed-as

AkhilGKrishnan, 2021-01-06:
any solution for above issue

zmk5, 2021-01-06:
Would you be so kind as to format the error message so it is easier to read?

Gatsby23, 2021-01-07:
> Would you be so kind as to format the error message so it is easier to read?

[image]
I'm sorry, there are another problem about yaml-cpp, I have searched some solution, but it doesn't work could you please give me some advice?

zmk5, 2021-01-07:
Not entirely sure what the issue is here since I've never really run into this error. What I assume it may be is that you haven't linked the flags for the homebrew dependencies in your `.bash_profile` or `zsh` file.

Another issue might be that you are missing the `yaml_vendor_cpp` package. Check in the `src/ros2/` within the `ros2_foxy` workspace you created for the source installation and see if you can find that `yaml_vendor_cpp` package in there.

Gatsby23, 2021-01-07:
> Not entirely sure what the issue is here since I've never really run into this error. What I assume it may be is that you haven't linked the flags for the homebrew dependencies in your [...]

The yaml_vendor_cpp package was there, but what falgs should I linked in the zsh files?

zmk5, 2021-01-07:
I have these ones:

```
export PATH="/usr/local/opt/icu4c/bin:$PATH"
export PATH="/usr/local/opt/icu4c/sbin:$PATH"
export LDFLAGS="-L/usr/local/opt/icu4c/lib"
export CPPFLAGS="-I/usr/local/opt/icu4c/include"
export PKG_CONFIG_PATH="${PKG_CONFIG_PATH}:/usr/local/opt/icu4c/lib/pkgconfig"

export Qt5_DIR=/usr/local/opt/qt5/lib/cmake
export CMAKE_PREFIX_PATH=${CMAKE_PREFIX_PATH}:/usr/local/opt/qt
export PATH=$PATH:/usr/local/opt/qt/bin

export PATH="/usr/local/opt/qt/bin:$PATH"
export LDFLAGS="-L/usr/local/opt/qt/lib"
export CPPFLAGS="-I/usr/local/opt/qt/include"
export PKG_CONFIG_PATH="${PKG_CONFIG_PATH}:/usr/local/opt/qt/lib/pkgconfig"

export PATH="/usr/local/opt/libxslt/bin:$PATH"
export LDFLAGS="-L/usr/local/opt/libxslt/lib"
export PKG_CONFIG_PATH="${PKG_CONFIG_PATH}:/usr/local/opt/libxslt/lib/pkgconfig"
export CPPFLAGS="-I/usr/local/opt/libxslt/include"

export PATH="/usr/local/opt/apr/bin:$PATH"
export PATH="/usr/local/opt/apr-util/bin:$PATH"
export PATH="/usr/local/sbin:$PATH"

export PATH="/usr/local/opt/opencv@3/bin:$PATH"
export LDFLAGS="-L/usr/local/opt/opencv@3/lib"
export CPPFLAGS="-I/usr/local/opt/opencv@3/include"
export PKG_CONFIG_PATH="${PKG_CONFIG_PATH}:/usr/local/opt/opencv@3/lib/pkgconfig"

export PATH="/usr/local/opt/curl-openssl/bin:$PATH"
export LDFLAGS="-L/usr/local/opt/curl-openssl/lib"
export CPPFLAGS="-I/usr/local/opt/curl-openssl/include"
export PKG_CONFIG_PATH="${PKG_CONFIG_PATH}:/usr/local/opt/curl-openssl/lib/pkgconfig"
```

Gatsby23, 2021-01-07:
> I have these ones:
>
> ```
> export PATH="/usr/local/opt/icu4c/bin:$PATH"
> export PATH="/usr/local/opt/icu4c/sbin:$PATH"
> export LDFLAGS="-L/usr/local/opt/icu4c/lib"
> export [...]

What does icu4c and libxslt do？ To my point of view, I think the main problem maybe still in the yaml-cpp

zmk5, 2021-01-07:
I just wanted to show you what I meant about setting your flags for dependencies that you download form homebrew. `icu4c` and `libxslt` were old ROS 2 dependencies. I tried looking at the `rosbag2` GitHub repository page and couldn't find an issue with the yaml dependency. Does running `brew doctor` show any linking issues?

clalancette, 2021-01-21:
This ticket has drifted into several tangents, and so far I don't see anything particular wrong with ROS 2 in here.  I'm going to close this issue out; if you are running into problems that are directly ROS 2 related, please open a new issue with just the problem you are facing.  Also, please use text instead of screenshots; we can't search screenshots.  Thanks.

closed 2021-01-21 by clalancette