ros2/ros2 #1080: Sudden NTP sync can break timer callback periodicity

hsgwa, 2021-01-13:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- Version or commit hash:
  - https://github.com/ros2/rmw_fastrtps (master)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Run demo node
```
$ ros2 run demo_nodes_cpp talker
[INFO] [1610531479.711126895] [talker]: Publishing: 'Hello World: 1'
[INFO] [1610531480.711165784] [talker]: Publishing: 'Hello World: 2'
[INFO] [1610531481.711246655] [talker]: Publishing: 'Hello World: 3'
...
```
Adjust system time manually in another terminal.
```
$ sudo timedatectl set-ntp no
$ sudo date -s @`date +%s | xargs -I@ echo @-5 | bc` # adjust 5s before.
```

#### Expected behavior
Most developers may assume timer callback executes periodically with best-effort.
```
[INFO] [1610531479.711126895] [talker]: Publishing: 'Hello World: 33' <- 1s later
[INFO] [1610531480.711165784] [talker]: Publishing: 'Hello World: 34' <- 1s later # timer adjust here
[INFO] [1610531481.711246655] [talker]: Publishing: 'Hello World: 35' <- 1s later
[INFO] [1610531481.711246655] [talker]: Publishing: 'Hello World: 36' <- 1s later
...
```
#### Actual behavior
Timer callback execution waits 5 seconds when the system timer is adjusted to backward.
```
[INFO] [1610531479.711126895] [talker]: Publishing: 'Hello World: 33' <- 1s later
[INFO] [1610531480.711165784] [talker]: Publishing: 'Hello World: 34' <- 1s later # timer adjust here
[INFO] [1610531481.711246655] [talker]: Publishing: 'Hello World: 35' <- 5s later
[INFO] [1610531481.711246655] [talker]: Publishing: 'Hello World: 36' <- 1s later
...
```

#### Additional information
In the rcl layer, the execution time of the timer callback is obtained by steady_time (clock_monotonic_raw), so it looks like timer callback is expected to be executed periodically. ([rclcpp/timer.hpp](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/timer.hpp#L275))
However, condition_variable->wait_for eventually executes futex system call that uses the system clock and can be affected by the system's time adjustment.([rmw_fastrtps/rmw_wait.cpp](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp#L183))

related issue: 
https://github.com/ros2/rcutils/issues/43

For systems that use NTP, this issue will only occur in cases the hardware clock advances fastly.
If the hardware clock advances slowly, the time adjustment is forward direction and timer callback is executed periodically. 

However, some systems can behave differently as expected with following conditions.

1. Use of hardware whose hardware clock advances fast.
2. NTP is inaccessible or a computer has been shut down for along time.
   (After startup, system clock reference hardware clock)
3. An application was executed before NTP synchronization is completed or NTP synchronization is suddenly executed during the operation.

This is an extreme example, but if an automated car that has not been used for several days suddenly adjusts its system time during driving, this could cause sensors and steer uncontrolled for a few seconds.
Incidentally, in my environment, the Raspberry Pi's hardware clock was delayed at a rate of 2.8 seconds per day.

This issue may not be a problem for most applications.
Rather than a specific source code change to ros packages, I think it can be adequately solved with applications or systems implementation.

I considered posting this to tutorials, ros2 design, etc.
I didn't know the appropriate place to post it, so I posted this as an issue.

clalancette, 2021-01-13:
> However, condition_variable->wait_for eventually executes futex system call that uses the system clock and can be affected by the system's time adjustment.([rmw_fastrtps/rmw_wait.cpp](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp#L183))

Are you sure this is the cause of the discontinuity?  That line of code you reference is `conditionVariable`, which turns out to be a `std::condition_variable`.  Looking at the documentation for [std::condition_variable::wait_for](https://en.cppreference.com/w/cpp/thread/condition_variable/wait_for), it says:

```
A steady clock is used to measure the duration.
```

So I have a suspicion that the problem is somewhere else, though I could be wrong.

Regardless, your overall point is true; we haven't done a systematic analysis of all of the timers in the system to make sure that they are using `steady_clock` instead of `system_clock`.  So there definitely could be bugs like this in the stack.  I do think it is worth an analysis, so any help here is welcome.

fujitatomoya, 2021-01-13:
problem confirmed, changing system time causes this issue. (cyclonedds has the same problem)

> In the rcl layer, the execution time of the timer callback is obtained by steady_time (clock_monotonic_raw), so it looks like timer callback is expected to be executed periodically.

I believe that this is correct. `rcl_timer_call` calculates with monotinic raw even if the system time is adjusted.

hsgwa, 2021-01-14:
@clalancette 
Thank you for your comment.
I was overlooking the document.
But I'm still thinking condition_variable is the cause in the case of rmw_fastrtps.

the same issue is observed even with this simple program.

```
// std::condition_variable::wait_for test
#include <chrono>
#include <condition_variable>
#include <iostream>

using namespace std::chrono_literals;

std::condition_variable cv;
std::mutex cv_m;

int main() {
  for (int i = 0; i < 10; i++) {
    std::unique_lock<std::mutex> lk(cv_m);
    cv.wait_for(lk, 1s, [] { return false; });
    std::cerr << "Thread timed out. i == " << i << '\n';
  }
}
```

strace command records following system calls in this program.
```
futex(0x561da4d77168, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1610595789, tv_nsec=71355755}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
write(2, "Thread timed out. i == ", 23) = 23
futex(0x561da4d77180, FUTEX_WAKE_PRIVATE, 1) = 0
(repeat)
```

As far as I can see source code, condition_variable calculates absolute wake-up time (tv_sec and tv_nsec) from the current time and timeout duration with steady_clock as the document describes.
ref. https://github.com/gcc-mirror/gcc/blob/master/libstdc%2B%2B-v3/include/std/condition_variable#L164
And wait_for eventually pass absolute wake-up time to futex system call with FUTEX_CLOCK_REALTIME option.
So futex's sleep is affected by system clock adjustment.

So far as futex system call,  it seems disabling FUTEX_CLOCK_REALTIME option makes discontinuity soleved.

>        FUTEX_CLOCK_REALTIME (since Linux 2.6.28)
>
>              If this option is set, the kernel measures the timeout
>              against the CLOCK_REALTIME clock.
>
>              If this option is not set, the kernel measures the timeout
>              against the CLOCK_MONOTONIC clock. 
> ref: futex(2) manual https://man7.org/linux/man-pages/man2/futex.2.html

Here is a backtrace result which I observed with debug mode.
#7std::condition_variable::wait_for calls #0futex_abstimed_wait_cancelable.
```
(gdb) catch syscall
(gdb) c
(gdb) bt
#0  futex_abstimed_wait_cancelable (private=<optimized out>, abstime=0x7fffffffdce0,
    clockid=<optimized out>, expected=0, futex_word=0x555555559168 <cv+40>)
    at ../sysdeps/nptl/futex-internal.h:320
#1  __pthread_cond_wait_common (abstime=0x7fffffffdce0, clockid=<optimized out>,
    mutex=0x555555559180 <cv_m>, cond=0x555555559140 <cv>) at pthread_cond_wait.c:520
#2  __pthread_cond_timedwait (cond=0x555555559140 <cv>, mutex=0x555555559180 <cv_m>,
    abstime=0x7fffffffdce0) at pthread_cond_wait.c:656
#3  0x00005555555556d8 in __gthread_cond_timedwait (__cond=0x555555559140 <cv>,
    __mutex=0x555555559180 <cv_m>, __abs_timeout=0x7fffffffdce0)
    at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:872
#4  0x0000555555556158 in std::condition_variable::__wait_until_impl<std::chrono::duration<long, std::ratio<1l, 1000000000l> > > (this=0x555555559140 <cv>, __lock=..., __atime=...)
    at /usr/include/c++/9/condition_variable:188
#5  0x0000555555555ebe in std::condition_variable::wait_until<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > (this=0x555555559140 <cv>, __lock=..., __atime=...)
    at /usr/include/c++/9/condition_variable:121
#6  0x00005555555554f4 in std::condition_variable::wait_until<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> >, main()::<lambda()> >(std::unique_lock<std::mutex> &, const std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long, std::ratio<1, 1000000000> > > &, <lambda()>) (this=0x555555559140 <cv>, __lock=..., __atime=..., __p=...)
    at /usr/include/c++/9/condition_variable:138
#7  0x0000555555555491 in std::condition_variable::wait_for<long int, std::ratio<1>, main()::<lambda()> >(std::unique_lock<std::mutex> &, const std::chrono::duration<long, std::ratio<1, 1> > &, <lambda()>) (
    this=0x555555559140 <cv>, __lock=..., __rtime=..., __p=...)
    at /usr/include/c++/9/condition_variable:166
#8  0x0000555555555375 in main () at main.cpp:13
```

Barry-Xu-2018, 2021-01-14:
@hsgwa 

FYI  
This issue has been finished by gcc-10. (Ubuntu 20.04, the default version of gcc is 9).  
I tried to use gcc-10 to build your simple program.  
```
futex(0x5648de754168, FUTEX_WAIT_BITSET_PRIVATE, 0, {tv_sec=118042, tv_nsec=858601644}, FUTEX_BITSET_MATCH_ANY) = -1 ETIMEDOUT (Connection timed out)
...
```

`FUTEX_CLOCK_REALTIME` has been removed in this version.

fujitatomoya, 2021-01-14:
@Barry-Xu-2018 

nice catch 👍 i've confirmed that too.

i think that the fix for libstdc++ is https://gcc.gnu.org/legacy-ml/gcc-patches/2019-10/msg01935.html (against https://gcc.gnu.org/bugzilla/show_bug.cgi?id=41861)

about the fix, how do you think about the following? (simple example works okay w/o affection from system time)

```
    //cv.wait_for(lk, 1s, [] { return false; });
    cv.wait_until(
      lk,
      std::chrono::steady_clock::now() + std::chrono::seconds(1),
      [] { return false; });
```

i will give it a shot to see if it works with rmw_wait.

fujitatomoya, 2021-01-14:
simple example provided by @hsgwa works okay with `cv.wait_until` with NTP time adjustment. but i still cannot get it work on rmw_wait with following patch.

```
diff --git a/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp b/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp
index df35f97..dfe4794 100644
--- a/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp
+++ b/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp
@@ -180,7 +180,8 @@ __rmw_wait(
       auto n = std::chrono::duration_cast<std::chrono::nanoseconds>(
         std::chrono::seconds(wait_timeout->sec));
       n += std::chrono::nanoseconds(wait_timeout->nsec);
-      timeout = !conditionVariable->wait_for(lock, n, predicate);
+      std::chrono::steady_clock::time_point now = std::chrono::steady_clock::now();
+      timeout = !conditionVariable->wait_until(lock, now + n, predicate);
     } else {
       timeout = true;
     }
```

any thoughts?

Barry-Xu-2018, 2021-01-15:
@fujitatomoya 

Maybe related to the different implementation of wait_until.  

- gcc-9 

    https://github.com/gcc-mirror/gcc/blob/releases/gcc-9/libstdc%2B%2B-v3/include/std/condition_variable#L176-L193

- gcc-10
  
    https://github.com/gcc-mirror/gcc/blob/1e151fb43523ff2c60540cff2518dba4ecd64f71/libstdc%2B%2B-v3/include/std/condition_variable#L197-L215

fujitatomoya, 2021-01-15:
@Barry-Xu-2018 

thanks, i wasn't clear for https://github.com/ros2/ros2/issues/1080#issuecomment-760007443, my bad.

https://github.com/ros2/ros2/issues/1080#issuecomment-760007443 is confirmed with gcc-9 ubuntu20.04 environment.

using gcc-9 on ubuntu20.04,

- sample with cv.wait_for (https://github.com/ros2/ros2/issues/1080#issuecomment-759916256) DOES get affected by NTP time adjustment.
- sample with cv.wait_until (https://github.com/ros2/ros2/issues/1080#issuecomment-759987816) DOES NOT get affected by NTP time adjustiment.
- rmw_wait (rmw_fastrtps) with wait_for / wait_until DOES get affected by NTP time adjustment.

i am not sure why it does not work on rmw_wait, but above is the current situation that i see now.

iuhilnehc-ynos, 2021-01-15:
> - sample with cv.wait_for (#1080 (comment)) DOES get affected by NTP time adjustment.

NOT affected by the original sample.

> - sample with cv.wait_until (#1080 (comment)) DOES NOT get affected by NTP time adjustiment.

and then you updated with `wait_until`,  which seems working. (actually, the sample doesn't get affected at the beginning.)

With https://github.com/ros2/ros2/issues/1080#issuecomment-759916256,
the sample can't be affected by NTP time adjustment.

I need to add some more code,
```
#include <thread>

int main() {
  std::thread t([](){std::cout << "to run a thread" << std::endl;});
  t.join();

  ...
```
after adding `std::thread`, and then continue to update with `wait_until`, it still DOES get affected by NTP time adjustment.

> - rmw_wait (rmw_fastrtps) with wait_for / wait_until DOES get affected by NTP time adjustment.

It still gets affected by NTP time adjustment because of https://github.com/ros2/ros2/issues/1080#issuecomment-760597689

Could you please check it again?

fujitatomoya, 2021-01-15:
> Could you please check it again?

right, i don't know what went wrong. but as you mentioned adding `std::thread` gets sample affected by NTP time adjustment.

hsgwa, 2021-01-15:
@Barry-Xu-2018 
Thank you! gcc-10 solved this issue :+1: 

@iuhilnehc-ynos
my comment was lack of information, I think I reproduced your result.

Different behavior may be caused by whether pthread is not linked.
```
# ubuntu 20.04,  g++ v 9.3.0
$ c++ -std=c++14 -pthread main.cpp # case 1) call futex system call. affected.
$ c++ -std=c++14 main.cpp  # case 2) doesn't call futex system call. Not affecteced.
```
case 1 calls system-call to sleep thread.
```
futex(0x55e17cb98180, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x55e17cb98168, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, {tv_sec=1610691831, tv_nsec=666754867}, FUTEX_BITSET_MATCH_ANYY) = -1 ETIMEDOUT (Connection timed out)
write(2, "Thread timed out. i == ", 23Thread timed out. i == ) = 23
(repeat)
```
```
$ top
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
  67957 hasegawa  20   0    6032   1536   1372 S   0.0   0.0   0:00.00 a.out // %CPU is ZERO.
```

case 2 doesn't sleep but waits with while-loop. it waits until timeout. (here: [condition_variable](https://github.com/gcc-mirror/gcc/blob/releases/gcc-9/libstdc++-v3/include/std/condition_variable#L138)) 
```
write(2, "Thread timed out. i == ", 23Thread timed out. i == ) = 23
(repeat)
```
```
$ top
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
  67732 hasegawa  20   0    5888   1504   1348 R 100.0   0.0   0:23.20 a.out  // %CPU is 100.
```

It seems above behavior is implemented as macro inside __gthread_cond_timedwait.
case 2 calls a different function from case 1.
```
#0  __pthread_cond_timedwait (cond=0x555555559140 <cv>, mutex=0x555555559180 <cv_m>,
    abstime=0x7fffffffdcf0) at forward.c:134 <- here is different from #2 https://github.com/ros2/ros2/issues/1080#issuecomment-759916256
   (where is __pthread_cond_timedwait definition..?)
#1  0x00005555555556d8 in __gthread_cond_timedwait (__cond=0x555555559140 <cv>,
    __mutex=0x555555559180 <cv_m>, __abs_timeout=0x7fffffffdcf0)
    at /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:872
#2  0x0000555555556158 in std::condition_variable::__wait_until_impl<std::chrono::duration<long, s
td::ratio<1l, 1000000000l> > > (this=0x555555559140 <cv>, __lock=..., __atime=...)
    at /usr/include/c++/9/condition_variable:188
#3  0x0000555555555ebe in std::condition_variable::wait_until<std::chrono::_V2::steady_clock, std:
:chrono::duration<long, std::ratio<1l, 1000000000l> > > (this=0x555555559140 <cv>, __lock=...,
    __atime=...) at /usr/include/c++/9/condition_variable:121
#4  0x00005555555554f4 in std::condition_variable::wait_until<std::chrono::_V2::steady_clock, std:
```

@fujitatomoya
Thanks.
Though I haven't read the fix for libstdc++, It seems wait_until calls futex systemcall with FUTEX_CLOCK_REALTIME option too.
seeing compile_coomands.json of rmw_fastrtps_cpp, ROS packages build with pthread option so that wait_until version patch(https://github.com/ros2/ros2/issues/1080#issuecomment-760007443) may get affected.

simple sample with wait_until case DOES get affected too in my environment.(https://github.com/ros2/ros2/issues/1080#issuecomment-759987816)

fujitatomoya, 2021-01-15:
> case 1 calls system-call to sleep thread.

this is because libstdc++ actually calls glibc backend in this case, which is `pthread_cond_timedwait` via `__gthread_cond_timedwait`. (https://github.com/gcc-mirror/gcc/blob/669f3c7bc022388ada90b8560f64486cf5a94394/libstdc%2B%2B-v3/include/std/condition_variable#L188-L189)

once glibc `pthread_cond_timedwait` is called, clockid will be `CLOCK_REALTIME`. this is default setting for `pthread_condattr_t`.

```
Thread 1 "a.out" hit Breakpoint 1, __pthread_cond_wait_common (abstime=0x7fffffffe1f0, clockid=0,
    mutex=0x55555555b2c0 <cv_m>, cond=0x55555555b280 <cv>) at pthread_cond_wait.c:656
```

so glibc (eventually system call) uses CLOCK_REALTIME, then it affects the sleep time when system time is adjusted in any way.

> case 2 doesn't sleep but waits with while-loop. it waits until timeout.

this is correct, it does not actually sleep, instead while-loop in userspace. as you pointed, https://github.com/gcc-mirror/gcc/blob/669f3c7bc022388ada90b8560f64486cf5a94394/libstdc%2B%2B-v3/include/std/condition_variable#L137-L139

this happens because `__gthread_cond_timedwait` is forwarded as stub, so `__gthread_cond_timedwait` returns immediately w/o using glibc implementation.

```
(gdb) break __pthread_cond_timedwait
Function "__pthread_cond_timedwait" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 1 (__pthread_cond_timedwait) pending.
(gdb) run
Starting program: /root/ros2_ws/work/a.out

Breakpoint 1, __pthread_cond_timedwait (cond=0x555555559140 <cv>, mutex=0x555555559180 <cv_m>, abstime=0x7fffffffe200)
    at forward.c:134
```

### Additional Information

- original issue: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=41861
- bugfix: https://gcc.gnu.org/legacy-ml/libstdc++/2019-07/msg00045.html
- same problem: https://stackoverflow.com/questions/63631963/condition-variable-workaround-for-wait-until-with-system-time-change

fujitatomoya, 2021-01-15:
i was considering work-around on this issue, but i do not have any solution at this moment.

i've tried to use [native_handle](https://en.cppreference.com/w/cpp/thread/condition_variable/native_handle) to set CLOCK_MONOTONIC. but it does not work.

```
  pthread_condattr_init(&attr);
  pthread_condattr_setclock(&attr, CLOCK_MONOTONIC);
  std::condition_variable cv;
  pthread_cond_init(cv.native_handle(), &attr);
  cv.wait_for(lk, 1s, [] { return false; });
```

any idea?

iuhilnehc-ynos, 2021-01-15:
> any idea?

Actually, I also did that. This is one point of the problem, another is https://github.com/gcc-mirror/gcc/blob/releases/gcc-9/libstdc++-v3/include/std/condition_variable#L117.
```
const __clock_t::time_point __s_entry = __clock_t::now();
                                                                 ^^^^^^^^^  is `chrono::system_clock`
https://github.com/gcc-mirror/gcc/blob/releases/gcc-9/libstdc++-v3/include/std/condition_variable#L68
```

So, it seems there is no workaround.

fujitatomoya, 2021-01-18:
> So, it seems there is no workaround.

probably not with using ubuntu20.04 main. (gcc/g++-10 package is `universe`, not `main`)

i tried to use `gcc-10` and `g++-10` on ubuntu20.04 with `update-alternatives`.

```
update-alternatives --config g++
update-alternatives --config gcc
```

but the problem still stays in `demo_nodes_cpp talker` with system time adjustment. probably there would be more configuration to colcon or cmake. could anyone help to confirm this? 😅

iuhilnehc-ynos, 2021-01-18:
After `update-alternatives --config g++`, please check `g++ --version` first.
then could you rebuild rmw_fastrtps_shared_cpp?  ( I hope you used `RMW_IMPLEMENTATION=rmw_fastrtps_cpp`, we know that the latest `ros2 master` used rmw_cyclonedds_cpp.)

```
$ rm -rf build/rmw_fastrtps_shared_cpp
$ colcon build --packages-select rmw_fastrtps_shared_cpp
```

---

BTW: I asked `Jonathan Wakely`s help to backport these two commits
https://github.com/gcc-mirror/gcc/commit/ad4d1d21ad5c515ba90355d13b14cbb74262edd2
https://github.com/gcc-mirror/gcc/commit/76e0dd66c87720a8a82a6d5f44ba4d8f63a51b1d
into `remotes/origin/releases/gcc-9`
(I think git://gcc.gnu.org/git/gcc.git is the official repository URL.)

iuhilnehc-ynos, 2021-01-18:
> but the problem still stays in demo_nodes_cpp talker with system time adjustment

I think you're right. `cyclonedds` use `pthread_cond_timedwait` directly, whichever gcc version to use, the issue still exists.

fujitatomoya, 2021-01-18:
backporting https://gcc.gnu.org/legacy-ml/libstdc++/2019-07/msg00045.html to `gcc-9/libstd` would be our choice, but it would take really long time. i think it would be nice to have work-around in ros2, which i am not really sure about at this moment...

iuhilnehc-ynos, 2021-01-18:
OK, I think we could use the same fix method as `ros_comm`(ros1) to https://github.com/ros2/rmw_fastrtps/blob/4d38e385b8474cd604ff4f491d058f14b3a49a5d/rmw_fastrtps_shared_cpp/src/types/custom_wait_set_info.hpp#L18.

```
#if _GLIBCXX_RELEASE == 9
// use backported version of condition variable for steady clock
#include "gcc_10_condition_variable.hpp"     // add a new file located in rmw_fastrtps/rmw_fastrtps_shared_cpp/include
#else
#include <condition_variable>
#endif
```

But I really don't like it.

fujitatomoya, 2021-01-18:
I think this is big, so i'd like to have some feedback from the community.

@wjwwood @ivanpauno @clalancette @jacobperron friendly ping,

- this seems to be `gcc-9/libstdc++` bug.
- root cause is described in https://github.com/ros2/ros2/issues/1080#issuecomment-760806890
- so far, there isn't good work-around at this moment.

I'd internally discuss on this too.

wjwwood, 2021-01-19:
Based on the discussion above, there doesn't seem to be a work around, so I'd say the best we can do is put on the affected release pages as a known issue, suggesting people upgrade to gcc10 (or presumably clang/libc++?).

If it hasn't already been done, open an issue against Cyclone.

I don't have any ideas for what else we could do, expect maybe add a test that exercises this, but it would be hard to do...

fujitatomoya, 2021-01-19:
>  I'd say the best we can do is put on the affected release pages as a known issue, suggesting people upgrade to gcc10 (or presumably clang/libc++?).

sounds good. and [libstdc++ patch](https://gcc.gnu.org/legacy-ml/libstdc++/2019-07/msg00045.html) is dependent on `glibc 2.30`. which is okay for ubuntu20.04 focal supporting `glibc 2.31`.

this issue and details have been really dedicated to [rmw_fastrtps](https://github.com/ros2/rmw_fastrtps), could anyone help to move this to [rmw_fastrtps](https://github.com/ros2/rmw_fastrtps) since i do not have permission to do that.

> open an issue against Cyclone.

i will check again, and make issue for cyclonedds accordingly.

fujitatomoya, 2021-01-19:
issue confirmed and reported to rmw_cyclonedds: https://github.com/ros2/rmw_cyclonedds/issues/276

jwakely, 2021-01-19:
> i was considering work-around on this issue, but i do not have any solution at this moment.
>
> i've tried to use [...]

That would have undefined behaviour, don't do silly things like that to the standard library internals.

jwakely, 2021-01-19:
> Are you sure this is the cause of the discontinuity? That line of code you reference is `conditionVariable`, which turns out to be a `std::condition_variable`. Looking at the documentation for [...]

cppreference.com is wrong. The C++ standard says:

> The functions whose names end in `_for` take an argument that specifies a duration. These functions produce
relative timeouts. Implementations should use a steady clock to measure time for these functions.

Note it says "should", which means it's not a hard requirement. Conversely, the `wait_until` functions "should" use the system clock. In the current POSIX standard (and in versions of glibc before 2.30) there was no way to specify a clock to use for a given wait (you could only set it when initializing the condition variable). Older versions of libstdc++ (and libc++) have to decide at construction time which clock to use, and that affects all `wait_for` and `wait_until` calls. Whichever clock is chosen, it can only meet one of the "should" recommendations in the standard. So not really a libstdc++ bug, but a limitation of POSIX pthreads in the current POSIX standard.

Mike Crowe modified POSIX and glibc and libstdc++ to add `pthread_cond_clockwait` to solve this problem, but it's very new and not supported in older toolchains. I do not plan to backport it to gcc-9.

fujitatomoya, 2021-01-19:
@jwakely appreciate for sharing the insights, that helps.

wjwwood, 2021-01-19:
> this issue and details have been really dedicated to rmw_fastrtps, could anyone help to move this to rmw_fastrtps since i do not have permission to do that.

@fujitatomoya would you like this on the rmw_fastrtps repository? Or maybe rcl? As it seems to affect all rmw implementations (perhaps for different reasons)?

Either way, just let me know and I will move it.

> issue confirmed and reported to rmw_cyclonedds: ros2/rmw_cyclonedds#276

Thanks!

ivanpauno, 2021-01-19:
I don't think that the issues in the rmw implementations are worth opening.
`rcl` supports [three kind of clocks](https://github.com/ros2/rcl/blob/0ca8c6162136594e490ba69fbf288de1b5f7073d/rcl/include/rcl/time.h#L59-L65): system time, steady time, ros time.

When you call `rcl_wait()`, the timeout passed to `rmw_wait()` is calculated [like](https://github.com/ros2/rcl/blob/c83119c826c75fbaf6e741658927a953318832ad/rcl/src/rcl/wait.c#L546-L582):

`min(rcl_wait_timeout, minimal_timeout_to_next_timer_trigger)`

but the timer might be using a different clock that the one `rmw_wait()` uses ....

As a consequence:
- if `rmw_wait` uses the system clock, a rcl timer using steady time isn't going to work fine (issue happens when system time is adjusted backwards)... 
- if `rmw_wait` uses steady clock, a rcl timer using system time isn't going to work fine (issue happens when system time is adjusted forward)...
- if `rmw_wait` uses any of steady/system clock, a rcl timer using ros time (with RTF > 1) isn't going to work fine.

So it doesn't seem to matter what clock `rmw_wait` uses, either timers using steady time or timers using system time aren't going to work "completely fine".

wjwwood, 2021-01-19:
I think the only thing that it makes sense to use with `rmw_wait` is steady clock. The fact that `rmw_wait` was being impacted by changes to the system clock was definitely a surprise.

> if rmw_wait uses steady clock, a rcl timer using system time isn't going to work fine (issue happens when system time is adjusted forward)...

That's a good point, we may need to do something here. It's possible that we need to wait on system timers separately (like in another thread) and interrupt the wait with a guard condition.

> if rmw_wait uses any of steady/system clock, a rcl timer using ros time (with RTF > 1) isn't going to work fine.

That already requires that we interrupt wait to check the timers any time the `/clock` topic is updated, i.e. any time the time has changed.

> So it doesn't seem to matter what clock rmw_wait uses

I agree that either option will cause a problem, but I do think steady clock makes more sense.

wjwwood, 2021-01-19:
Actually, we currently document that wait should operate on system time...

https://github.com/ros2/rmw/blob/eeb5859def4af28c3aa231dfe7be30fbc31bcfaf/rmw/include/rmw/rmw.h#L2368

So maybe we need to evaluate that again, but either way.

ivanpauno, 2021-01-19:
> I agree that either option will cause a problem, but I do think steady clock makes more sense.

> Actually, we currently document that wait should operate on system time...
>
> https://github.com/ros2/rmw/blob/eeb5859def4af28c3aa231dfe7be30fbc31bcfaf/rmw/include/rmw/rmw.h#L2368
>
> So maybe we need to evaluate that again, but either way.

I think we shouldn't document exactly what clock is being used, because it's hard to enforce that's the case for all rmw implementations, all platforms, etc ...
We could recommend implementations to use one, but I'm not sure if that matters much.

> It's possible that we need to wait on system timers separately (like in another thread) and interrupt the wait with a guard condition.

A bit of out the scope of this issue, but probably handling all timers in a separated thread is a good idea.

extra note: if we want to still handle some/all the timers in `rcl_wait()`, we shouldn't be recalculating the timeout in each `rcl_wait()` call because of performance reasons, but cache that calculation and update it when a timer is added/removed to the waitset. 

> That already requires that we interrupt wait to check the timers any time the /clock topic is updated, i.e. any time the time has changed.

So, I think that the waitset is going to be awaken by the clock message (supposing that you're waiting both on the "/clock" subscription and the timer in the same waitset, which is usually the case).
The timer isn't going to be marked as ready immediately  but only after the subscription callback is executed and `rcl_wait()` is called again.
I don't recall exactly, but I think there were some issues when the RTF was greater than 1 (maybe @hidmic mentioned that to me (?)).

wjwwood, 2021-01-19:
> I think we shouldn't document exactly what clock is being used, because it's hard to enforce that's the case for all rmw implementations, all platforms, etc ...
We could recommend implementations to use one, but I'm not sure if that matters much.

I actually don't see why this would be the case. We required already that systems have a steady clock and a system clock. It's possible we don't control which clock the very lowest item uses. But either way, I think it's important that the calling code know which clock is being used, either because we documented it as a requirement or by self reporting if we leave it flexible, e.g. similar to `std::chrono` clocks having `is_steady`.

> A bit of out the scope of this issue, but probably handling all timers in a separated thread is a good idea.

Perhaps, but we'd need a thread for each kind of timer, at least.

> extra note: if we want to still handle some/all the timers in rcl_wait(), we shouldn't be recalculating the timeout in each rcl_wait() call because of performance reasons, but cache that calculation and update it when a timer is added/removed to the waitset.

I don't think that's possible because the time left until the "next" timer will change every time and which timer is "next" may change each time as well. I don't think there's anything you can cache realistically.

wjwwood, 2021-01-19:
> I agree that either option will cause a problem, but I do think steady clock makes more sense.

Just to expand on this too, I think a steady clock makes sense because the timeout is always expressed as a duration, i.e. we don't have a `wait_until`. It's possible to measure a duration using a non-steady clock, but I think it's surprising in this case. A duration doesn't imply a specific epoch or even rate, so I guess my feeling is a steady clock is least surprising.

wjwwood, 2021-01-19:
Also, we (or at least I) discussed with someone about making it possible to wait based on ROS time. I can't find the conversation for this, maybe it was not on GitHub... but aside from the obvious circular issues (wait would need ROS Time to update, and ROS Time would need wait to update), we had the same issue that we don't control the clock being used at the "bottom" of the stack.

ivanpauno, 2021-01-19:
> I don't think that's possible because the time left until the "next" timer will change every time and which timer is "next" may change each time as well. I don't think there's anything you can cache realistically.

Ah yeah :man_facepalming:, I was thinking only to cache the minimal period, but that won't work.

ivanpauno, 2021-01-19:
> Just to expand on this too, I think a steady clock makes sense because the timeout is always expressed as a duration, i.e. we don't have a wait_until. It's possible to measure a duration using a [...]

:+1: it's definitely less surprising.

> I actually don't see why this would be the case. We required already that systems have a steady clock and a system clock. It's possible we don't control which clock the very lowest item uses. But [...]

I mentioned that because it depends on what synchronization mechanism `rmw_wait()` uses behind the scenes, if that mechanism supports using different clocks, and if that's consistent across different platforms.

IIUC, all rmw implementations (at least fastrtps and cyclonedds) are using a `std::condition_variable`.
The cpp standard is only saying that `wait_for` "should" use a steady clock, so I don't think we would be able to enforce anything more than a "should".
The idea of having a "is_steady" method is interesting, but I don't think that exist for e.g. the clock that `std::condition_variable` is using, so the only way to implement that would be to provide the result manually for each rmw/platform (which is a maintenance issue).

I understand the value of having exactly documented what clock is being used, and if that's possible I agree that the best choice for a timeout duration is a steady clock. But I'm not sure we would be able to guarantee that.

fujitatomoya, 2021-01-20:
@wjwwood @ivanpauno 

thanks for the consideration,

> @fujitatomoya would you like this on the rmw_fastrtps repository? Or maybe rcl? As it seems to affect all rmw implementations (perhaps for different reasons)?

yes, but it depends on requirement. cyclonedds can support steady time instead of system time since it uses native handlers. (see https://github.com/ros2/rmw_cyclonedds/issues/276#issuecomment-762695763)

> if rmw_wait uses steady clock, a rcl timer using system time isn't going to work fine (issue happens when system time is adjusted forward)...

could you enlighten why this is not gonna work? implementation knows only duration, but no idea about what clock type is used. even if system time is adjusted in forward or backward, implementation does not even know about that.

> I think the only thing that it makes sense to use with rmw_wait is steady clock. The fact that rmw_wait was being impacted by changes to the system clock was definitely a surprise.

totally agree, because of the interface for `rmw_wait` is `duration` (not `time point`).

> Actually, we currently document that wait should operate on system time...

if that is the current requirement, this is not an issue. but as you mentioned, we'd like to discuss on this, since this affects user experience. (reflection time and behavior for robots) probably discuss in [Middleware WG](https://github.com/ros2/middleware_working_group) would be the best place?

> Just to expand on this too, I think a steady clock makes sense because the timeout is always expressed as a duration, i.e. we don't have a wait_until. It's possible to measure a duration using a [...]

> 👍 it's definitely less surprising.

👍

> IIUC, all rmw implementations (at least fastrtps and cyclonedds) are using a std::condition_variable.

cyclonedds uses native handlers such as `pthread_xxx`, so it can actually supports steady clock if that is the requirement. (see https://github.com/ros2/rmw_cyclonedds/issues/276#issuecomment-762695763)

ivanpauno, 2021-01-20:
> could you enlighten why this is not gonna work? implementation knows only duration, but no idea about what clock type is used. even if system time is adjusted in forward or backward, implementation does not even know about that.

Suppose you only have one timer using the clock.
When you call `rcl_wait()` that timer is going to be triggered in 1s, so the `rmw_wait()` timeout is adjusted to 1s.
If `rmw_wait()` is waiting using the steady clock and the system clock is adjusted forward 1s, you will actually end up waiting 2 (*system time*) seconds.

i.e.: the clock that `rmw_wait()` is using must be the same that the clock of the timer you're using, if not the logic `rcl_wait()` doesn't really work.

By the way, I don't think that periodic timers using the system clock are really useful, so the case described in this issue is probably more important.

fujitatomoya, 2021-01-21:
> When you call rcl_wait() that timer is going to be triggered in 1s, so the rmw_wait() timeout is adjusted to 1s. If rmw_wait() is waiting using the steady clock and the system clock is adjusted forward 1s, you will actually end up waiting 2 (system time) seconds.

I see, thanks!

I think that in this case, system clock is forwarded, so even calling `rcl_wait` with timeout = 1sec, it will never be in sleep since timeout for `rmw_wait` is 0sec. when calling `rcl_timer_get_time_until_next_call` will return with `time_until_next_call` zero or probably negative, because system clock is forwarded and it already reaches time point that cyclic expires.

https://github.com/ros2/rcl/blob/c83119c826c75fbaf6e741658927a953318832ad/rcl/src/rcl/wait.c#L572-L579

then,

https://github.com/ros2/rcl/blob/c83119c826c75fbaf6e741658927a953318832ad/rcl/src/rcl/wait.c#L592

so timeout argument for `rmw_wait` is going to be zero.

> By the way, I don't think that periodic timers using the system clock are really useful, so the case described in this issue is probably more important.

:+1:

ivanpauno, 2021-01-21:
> so timeout argument for rmw_wait is going to be zero.

I mean, the forwarding happens just after the calculation of the timeout passed to rmw time.

hidmic, 2021-01-25:
Oof, catching up.

> I think we shouldn't document exactly what clock is being used, because it's hard to enforce that's the case for all rmw implementations, all platforms, etc ...

> I think it's important that the calling code know which clock is being used, either because we documented it as a requirement or by self reporting if we leave it flexible, e.g. similar to std::chrono clocks having is_steady.

I agree with @wjwwood here. A time sensitive API against an unspecified clock isn't particularly useful. 

> Actually, we currently document that wait should operate on system time...

Indeed. An outcome of the discussion at https://github.com/ros2/rcl/issues/687. Not necessarily ideal or even desired, but to document the current state of things.

 > I don't recall exactly, but I think there were some issues when the RTF was greater than 1

Indeed. An RTF > 1 cannot be met as long as we measure wait timeouts against steady or system clocks. ROS 1 worked around the problem by polling at an arbitrary, hopefully faster rate essentially (see [here](https://github.com/ros/ros_comm/blob/08fcc096f4e822b56bfb5a30b0a09ec45f0802aa/clients/roscpp/include/ros/timer_manager.h#L587-L592)). I personally think we shouldn't be trying to solve the clock distribution problem all over again. A variation of NTP or PTP running on top of the middleware with some device to deal with clock pause/resume should do.

clalancette, 2022-02-03:
This very interesting discussion petered out a year ago.  I don't think much has changed since then (except that we are going to switch Jammy to 22.04, which should have the original bug fixed).

Is the next step here to document what we have, and/or make a design document about time in ROS 2?

ivanpauno, 2022-02-03:
> Is the next step here to document what we have, and/or make a design document about time in ROS 2?

I think that rmw_wait() documentation should said that the implementation "should" use the steady clock.
I don't think that we need to fix the fact that fastrtps was using the system clock in previous releases, as that seems hard to fix.

@fujitatomoya @hsgwa @Barry-Xu-2018 @iuhilnehc-ynos do you think that's a good resolution for this issue?

There's also the problem I pointed out here https://github.com/ros2/ros2/issues/1080#issuecomment-763607158, but I don't think that one is that important.

fujitatomoya, 2022-02-03:
@ivanpauno 

> I think that rmw_wait() documentation should said that the implementation "should" use the steady clock.

i am good to go 👍 that is actually something we can do in ROS 2 aspect.
we can also keep https://github.com/ros2/rmw_cyclonedds/issues/276 alive since that is not a recommendation after this documentation. i think cyclonedds uses native handler in C, so that this could be addressed.

about the root cause of this, https://github.com/ros2/ros2/issues/1080#issuecomment-760806890, we need to wait it out. (i think there is nothing we can do.)

(BTW, @Barry-Xu-2018 @iuhilnehc-ynos are on Lunar New Year Holiday!)

fujitatomoya, 2022-04-28:
@ivanpauno 

as described https://github.com/ros2/ros2/issues/1080#issuecomment-760806890, we confirmed that this has been fixed on humble release with `rnw_fastrtps`. (Ubuntu 22.04)

fujitatomoya, 2022-05-10:
@hsgwa it would be appreciated if you try rolling(or humble) with Ubuntu22.04 to make sure that original problem has been fixed.

hsgwa, 2022-05-10:
@fujitatomoya 
I have confirmed that the bug is fixed in the following environments.

- Ubuntu22.04
- rolling
- rmw_fastrtps_cpp
- (docker ros2:nightly [DIGEST:9c3250e5675f])

Thanks for the bug fix!
For fastdds, this issue may be closed.

fujitatomoya, 2022-05-10:
@hsgwa thanks for checking.

- this issue seems to be dedicated to `rmw_fastrtps`, which has been confirmed the fix on Ubuntu22.04.
- for `rmw_cyclonedds`, problem still can be observed. we have https://github.com/ros2/rmw_cyclonedds/issues/276 to manage this issue. since cyclonedds uses native handler, it is up to the implementation of cyclonedds.
- doc for rmw fix, https://github.com/ros2/rmw/pull/316 has been addressed.

i think that we can close this issue. @ivanpauno @clalancette what do you think?

clalancette, 2022-05-10:
Yep, that sounds good.  I'll close this.

closed 2022-05-10 by clalancette