ros2/ros2 #1082: Unable to use ros2 after installing using debian-packages

shlokgoel, 2021-01-21:
I have installed ROS 2 (ros-dashing-desktop) via debian packages as per the following instructions:

 https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/

I am using Ubuntu 18.04. 

After installation, I source the ros-dashing workspace using the following command: 

     source /opt/ros/dashing/setup.bash

When I try to run the ros 2 demo example using the following command: 

    ros2 run demo_nodes_cpp talker

I get the following error: 

>  `   ros2 run demo_nodes_cpp talker 
> Failed to load entry point 'security': No module named 'ament_index_python' 
> Failed to load entry point 'run': No module named 'ament_index_python' 
> Failed to load entry point 'launch': No module named 'ament_index_python' 
> Failed to load entry point 'info': No module named 'rclpy' 
> Failed to load entry point 'list': No module named 'ament_index_python' Failed to load entry point 'send_goal': No module named 'action_msgs' 
> Failed to load entry point 'show': No module named 'ament_index_python' 
> Failed to load entry point 'list': No module named 'rclpy' Failed to load entry point 'load': No module named 'rclpy' 
> Failed to load entry point 'standalone': No module named 'rclpy' Failed to load entry point 'types': No module named 'rclpy' Failed to load entry point 'unload': No module named 'rclpy' Failed to load entry point 'start': No module named 'rclpy' Failed to load entry point 'status': No module named 'rclpy' Failed to load entry point 'stop': No module named 'rclpy' Failed to load entry point 'get': No module named 'rclpy' Failed to load entry point 'list': No module named 'rclpy' Failed to load entry point 'nodes': No module named 'rclpy' Failed to load entry point 'set': No module named 'rclpy' Failed to load entry point 'list': No module named 'ament_index_python' Failed to load entry point 'package': No module named 'ament_index_python' Failed to load entry point 'packages': No module named 'ament_index_python' Failed to load entry point 'show': No module named 'ament_index_python' Failed to load entry point 'info': No module named 'rclpy' Failed to load entry point 'list': No module named 'rclpy' Failed to load entry point 'delete': No module named 'rcl_interfaces' Failed to load entry point 'get': No module named 'rcl_interfaces' Failed to load entry point 'list': No module named 'rcl_interfaces' Failed to load entry point 'set': No module named 'rcl_interfaces' Failed to load entry point 'create': No module named 'ament_index_python' Failed to load entry point 'executables': No module named 'ament_index_python' Failed to load entry point 'list': No module named 'ament_index_python' Failed to load entry point 'prefix': No module named 'ament_index_python' Failed to load entry point 'call': No module named 'rclpy' Failed to load entry point 'list': No module named 'rclpy' Failed to load entry point 'list': No module named 'ament_index_python' Failed to load entry point 'package': No module named 'ament_index_python' Failed to load entry point 'packages': No module named 'ament_index_python' Failed to load entry point 'show': No module named 'ament_index_python' Failed to load entry point 'bw': No module named 'rclpy' Failed to load entry point 'delay': No module named 'rclpy' Failed to load entry point 'echo': No module named 'rclpy' Failed to load entry point 'hz': No module named 'rclpy' Failed to load entry point 'info': No module named 'rclpy' Failed to load entry point 'list': No module named 'rclpy' Failed to load entry point 'pub': No module named 'rclpy' usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ... ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'run' (choose from 'action', 'component', 'daemon', 'extension_points', 'extensions', 'lifecycle', 'msg', 'multicast', 'node', 'param', 'pkg', 'service', 'srv', 'topic')`

*I am able to succesfully run the ros2 demo if I install the ros2-dashing by building from source. However, I need to install ros-dashing inside my opt/ros/ workspace hence I need to install it using the Debian Packages* 

(https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/)

**PS. I get the same error even if I try to enter the command ros2 after sourcing the ros-dashing workspace.** 

Any idea how I can resolve this? 

Thank you!

closed 2021-01-22 by shlokgoel

DavidATapia, 2021-02-25:
I'm having this exact same issue (but with foxy) since `0.9.2-1focal.20210217.003839`, which is still the candidate release version as of today.

Why was this issue closed?  Is there a workaround?

Thanks.

clalancette, 2021-02-25:
Please ask questions like this on https://answers.ros.org .  You're far more likely to get a response there.