ros2/ros2 #1086: Deadlock on service call when client and service are launched in the same component container

zygfrydw, 2021-02-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - apt
- Version or commit hash:
  - foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
1. add libraries https://github.com/zygfrydw/ros2_service_issue/tree/master/srv_client to workspace and compile
2. run command 
``` bash
ros2 launch src/ros2_service_issue/srv_client/deadlock_config.launch.py 
```

#### Expected behavior
Expected console output
```
[INFO] [launch]: All log files can be found below /home/autoware/.ros/log/2021-02-01-09-28-46-969254-zwieszok-93633
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [93665]
[INFO] [component_container-2]: process started with pid [93667]
[component_container-1] [INFO] [1612171727.499984107] [client_container]: Load Library: /as_drive/ws/install/srv_client/lib/libsrv_client_lib.so
[component_container-1] [INFO] [1612171727.501412150] [client_container]: Found class: rclcpp_components::NodeFactoryTemplate<srv_client::SrvClient>
[component_container-1] [INFO] [1612171727.501484517] [client_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<srv_client::SrvClient>
[component_container-2] [INFO] [1612171727.503452952] [srv_service]: Load Library: /as_drive/ws/install/srv_service/lib/libsrv_service_lib.so
[component_container-2] [INFO] [1612171727.504916458] [srv_service]: Found class: rclcpp_components::NodeFactoryTemplate<srv_service::SrvService>
[component_container-2] [INFO] [1612171727.504993702] [srv_service]: Instantiate class: rclcpp_components::NodeFactoryTemplate<srv_service::SrvService>
[component_container-2] [WARN] [1612171727.505202179] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/srv_client' in container '/client_container'
[WARNING] [component_container-2]: there are now at least 2 nodes with the name /srv_service created within this launch context
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/srv_service' in container '/srv_service'
[component_container-1] [INFO] [1612171728.014497629] [srv_client]: Service avaliable - making a call
[component_container-2] [INFO] [1612171728.014914026] [srv_service]: Incoming request
[component_container-1] [INFO] [1612171728.015279827] [srv_client]: Received result: hello - serice response
[component_container-1] [INFO] [1612171728.514497335] [srv_client]: Service avaliable - making a call
[component_container-2] [INFO] [1612171728.514953284] [srv_service]: Incoming request
[component_container-1] [INFO] [1612171728.515394042] [srv_client]: Received result: hello - serice response
[component_container-1] [INFO] [1612171729.014504878] [srv_client]: Service avaliable - making a call
[component_container-2] [INFO] [1612171729.014999843] [srv_service]: Incoming request
[component_container-1] [INFO] [1612171729.015488860] [srv_client]: Received result: hello - serice response
```
#### Actual behavior
Actual console output is 
```
[INFO] [launch]: All log files can be found below /home/autoware/.ros/log/2021-02-01-09-29-31-995960-zwieszok-94127
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [94158]
[component_container-1] [INFO] [1612171772.532506940] [client_container]: Load Library: /as_drive/ws/install/srv_client/lib/libsrv_client_lib.so
[component_container-1] [INFO] [1612171772.533759314] [client_container]: Found class: rclcpp_components::NodeFactoryTemplate<srv_client::SrvClient>
[component_container-1] [INFO] [1612171772.533823989] [client_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<srv_client::SrvClient>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/srv_client' in container '/client_container'
[component_container-1] [INFO] [1612171772.545901260] [client_container]: Load Library: /as_drive/ws/install/srv_service/lib/libsrv_service_lib.so
[component_container-1] [INFO] [1612171772.546907394] [client_container]: Found class: rclcpp_components::NodeFactoryTemplate<srv_service::SrvService>
[component_container-1] [INFO] [1612171772.546985878] [client_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<srv_service::SrvService>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/srv_service' in container '/client_container'
[component_container-1] [INFO] [1612171773.042005660] [srv_client]: Service avaliable - making a call
```
And the node is hanging

#### Additional information
When I am running Service client and server in separate component containers, it is working correctly as presented in working behavior:
```
from launch import LaunchDescription
from launch_ros.actions import ComposableNodeContainer, Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    return LaunchDescription(
        [
            ComposableNodeContainer(
                name="client_container",
                namespace="",
                package="rclcpp_components",
                executable="component_container",
                output="screen",
                composable_node_descriptions=[
                    ComposableNode(
                        package="srv_client",
                        plugin="srv_client::SrvClient",
                        name="srv_client",
                    ),
                ],
            ),
            ComposableNodeContainer(
                name="srv_service",
                namespace="",
                package="rclcpp_components",
                executable="component_container",
                output="screen",
                composable_node_descriptions=[
                    ComposableNode(
                        package="srv_service",
                        plugin="srv_service::SrvService",
                        name="srv_service",
                    ),
                ],
            ),
        ]
    )

```

however, when I am running both nodes in a single container, there is **a deadlock**:

```
from launch import LaunchDescription
from launch_ros.actions import ComposableNodeContainer, Node
from launch_ros.descriptions import ComposableNode

def generate_launch_description():
    return LaunchDescription(
        [
            ComposableNodeContainer(
                name="client_container",
                namespace="",
                package="rclcpp_components",
                executable="component_container",
                output="screen",
                composable_node_descriptions=[
                    ComposableNode(
                        package="srv_client",
                        plugin="srv_client::SrvClient",
                        name="srv_client",
                    ),
                    ComposableNode(
                        package="srv_service",
                        plugin="srv_service::SrvService",
                        name="srv_service",
                    ),

                ],
            )
        ]
    )

```

iuhilnehc-ynos, 2021-02-01:
I think you could try to use `executable="component_container_mt"`.

zygfrydw, 2021-02-01:
Thank you - indeed it solves the problem!

closed 2021-02-01 by zygfrydw