ros2/ros2 #1092: Provide a way of loading node parameters from a file regardless of the node namespace

v-lopez, 2021-02-23:
## Feature request

#### Feature description
Discussion originated here: https://github.com/ros2/ros2cli/pull/590#issuecomment-782101072

As an example I'll use ros2_control parameters, as specified here: https://github.com/ros-controls/ros2_control_demos/blob/master/ros2_control_demo_robot/controllers/rrbot_forward_controller_position.yaml

If you have multiple identical robots, their controller_manager and controller parameters will be identical as well. The same applies for the majority of the nodes running on the robot. 

But as mentioned on the discussion linked above, the key in the parameters file is a fully qualified name (namespace + node name), therefore if you have 2 identical robots, you need to generate 2 files with different keys in the parameters file. 

All this is doable, either by duplicating files manually, or by some launch file code that adds the proper namespace to those keys. But since this will be an extremely common (if not the default) behavior, there should be an easier way.

#### Implementation considerations
I believe @ivanpauno [suggestion](https://github.com/ros2/ros2cli/pull/590#issuecomment-782107723) is the proper way:
>Ideally, `my_robot` would refer to any node named `my_robot` and `/namespace/my_robot` would refer specifically to one.

ivanpauno, 2021-02-23:
Another way to achieve the same is to allow more wildcards e.g.: `/**/my_robot` would match any node named `my_robot` (and any namespace).
That option sounds better, as it's backwards compatible (`my_robot` is currently the same than the absolute `/my_robot`).

Now I remember this has also been proposed in https://github.com/ros2/rclcpp/issues/1265

ivanpauno, 2021-02-23:
Also, https://github.com/ros2/design/pull/303 was documenting https://github.com/ros2/rclcpp/issues/1265 in the design docs.