ros2/ros2 #1093: Various ros2 commands such as `ros2 topic list` and `ros2 node list` just hang, no errors given

DavidATapia, 2021-02-25:
## Bug report

- **Operating System**:
  Ubuntu 20.04 (focal-20210119)

- **Installation type**:
  Debians

- **Version or commit hash**:
  `0.9.2-1focal.20210217.003839`
taken from:
```
apt-cache policy ros-foxy-ros-base
ros-foxy-ros-base:
  Installed: 0.9.2-1focal.20210217.003839
  Candidate: 0.9.2-1focal.20210217.003839
  Version table:
 *** 0.9.2-1focal.20210217.003839 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
```

- **DDS implementation**:
  Default

#### Steps to reproduce issue
- Install ROS2 from debian packages using [these instructions](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/).
- then install `ros-foxy-ros-base`

#### Expected behavior
`ros2 topic list` should give:
```
/parameter_events
/rosout
```

`ros2 param list` and `ros2 node list` should at least return and not hang.

#### Actual behavior
No output -- hangs and does not return

#### Additional information

Validated by installing ROS2 in a fresh Docker container from Dockerhub labeled as `ubuntu:focal-20210119`.

fujitatomoya, 2021-02-25:
I cannot reproduce this issue on my environment.

```
root@92be57c7870d:/# apt-cache policy ros-foxy-ros-base
ros-foxy-ros-base:
  Installed: 0.9.2-1focal.20210217.003839
  Candidate: 0.9.2-1focal.20210217.003839
  Version table:
 *** 0.9.2-1focal.20210217.003839 500
        500 http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
root@92be57c7870d:~# source /opt/ros/foxy/setup.bash
root@92be57c7870d:~# ros2 topic list
/parameter_events
/rosout
root@92be57c7870d:~# ros2 param list
root@92be57c7870d:~# ros2 node list
```

closed 2021-02-26 by DavidATapia

gcullie, 2021-03-15:
I'm having a similar issue.  Hangs with the call:  _topicList.py: ln25: **with NodeStrategy(args) as node**_  
Opening issue in ros2cli project.

austin-InDro, 2023-01-06:
I'm having the same issue on my system.
```
Linux austin-greisman 5.15.0-56-generic #62-Ubuntu SMP Tue Nov 22 19:54:14 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
ros-humble-ros2cli/jammy,now 0.18.4-1jammy.20221207.031440 amd64 [installed,automatic]
ros-humble-ros2controlcli/jammy 2.18.0-1jammy.20221207.043415 amd64

ros-humble-cyclonedds-dbgsym/jammy 0.9.1-1jammy.20221108.154039 amd64
ros-humble-cyclonedds/jammy,now 0.9.1-1jammy.20221108.154039 amd64 [installed,automatic]

```
It seems the only temporary fix is restarting my system.
`ros2 daemon stop && ros2 daemon start` also hangs...

I first noticed this issue while I was debugging C++ code. It seemed to be causing issues with Cyclone