ros2/ros2 #1094: Using Python Virtual Environments

jandaa, 2021-02-26:
## Bug report

I'm trying to use a python virtual environment in my colcon workspace to separate system level and node level dependencies. I'm following the instructions given in this [tutorial](https://index.ros.org/doc/ros2/Tutorials/Using-Python-Packages/#installing-via-a-virtual-environment) in the ros2 documentation (Specifically installing via a virtual environment). However, when I check what python interpreter is being used inside the node, it returns the system level one instead of the virtual environment. This can be further verified by installing a package in the virtual environment that is not present in the system environment and try importing it from inside the node. 

I have tried doing a clean build with and without sourcing the virtual environment in the build shell. I have also tried using virtualenv as in the tutorial and also python -m venv. I also call my virtual environment "venv" as in the tutorial.

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Binaries
- Version or commit hash: Foxy Fitzroy
- DDS implementation: (Default) Cyclone DDS
- Client library (if applicable): N/A

#### Steps to reproduce issue
1. Follow the steps in the [tutorial](https://index.ros.org/doc/ros2/Tutorials/Using-Python-Packages/#installing-via-a-virtual-environment)
2. Create a ros2 python package with ```cd src && ros2 pkg create venv --build-type ament_python```
3. Create a python node called ```venv.py``` under src/venv/venv with the following code
```
import rclpy
from rclpy.node import Node
import sys

class ExampleNode(Node):
    def __init__(self):
        super().__init__('sensors')
        self.get_logger().info("ExampleNode started")
        self.get_logger().info(f"python path: {sys.executable}")

def main(args=None):
    rclpy.init(args=args)
    node = ExampleNode()
    rclpy.spin(node)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
4. Add the node to setup.py with
```
entry_points={
      'console_scripts': [
          'venv = venv.venv:main'
      ],
  },
```
5. Build using
```
source ./venv/bin/activate
source /opt/ros/foxy/setup.bash
colcon build
source install/setup.sh
```
6. Run using ```ros2 run venv venv ```

#### Expected behavior
The returned python path should show ```${workspaceFolder}/venv/bin/python``` and any packages installed in the virtual environment should be accessible inside the node. 

#### Actual behavior
Returns the system base python ```/usr/bin/python3``` and breaks when trying to import a package that is installed in the virtual environment but not at system level.

YunfeiZHAO, 2021-05-14:
same problem with anaconda

PymZoR, 2021-07-17:
Same problem on galactic

clalancette, 2021-07-19:
In short, python venvs with Python installed in the venv are never going to work using the installed ROS 2 binaries.  The problem boils down to the fact that building a Python package as a Debian hard-codes the default python path in the console script (namely `/usr/bin/python3`).

However, what *should* work is having a Python venv with just pip packages installed.  That is, if you rely on the system version of Python, but just make a venv where you pip install a bunch of dependencies, that should work just fine.  That said, ROS 2 expects certain versions of pip dependencies/APIs to be present, so it may be the case that you cannot use newer versions of pip packages while doing this.

jandaa, 2021-07-19:
Thank you so much @clalancette for your response, that makes sense! What you describe as using just the pip packages would work just fine. In this case, how would we point our environment to these separate site packages contained in our venv?

clalancette, 2021-07-19:
> Thank you so much @clalancette for your response, that makes sense! What you describe as using just the pip packages would work just fine. In this case, how would we point our environment to these separate site packages contained in our venv?

Just running `source venv/bin/activate`, then `source /opt/ros/foxy/setup.bash` should do it, though I haven't personally tried this out yet.

theunkn0wn1, 2021-08-11:
> Just running `source venv/bin/activate`, then `source /opt/ros/foxy/setup.bash` should do it, though I haven't personally tried this out yet.

I tried doing precisely this, though `ros2 run` still seems to be using either wrong interpreter or is otherwise disregarding venv packges and thus isn't finding the pip-installed packages the node requires.

 - the venv was created using the same interpreter ROS2 Foxy was installed with
  - the venv was sourced first
  - Foxy's setup was sourced second
  - `colcon build ` executed
  - the python node's setup was sourced third
  - `ros2 run` finds and attempts to execute the node, but fails because it fails to find the pip-installed package.
  
Console output provided.
[console.log](https://github.com/ros2/ros2/files/6970953/console.log)

clalancette, 2021-08-11:
It's possible that sourcing `install/setup.zsh` is overwriting the PYTHONPATH that the venv setup.  Since you don't actually need the venv to build, then I'll suggest trying the other order:

* source /opt/ros/foxy/setup.zsh
* colcon build
* source install/local_setup.zsh
* source venv/bin/activate
* ros2 run pkg exe

theunkn0wn1, 2021-08-11:
> It's possible that sourcing `install/setup.zsh` is overwriting the PYTHONPATH that the venv setup. Since you don't actually need the venv to build, then I'll suggest trying the other order:
>
>     * [...]

```
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 12:57:43
$ source /opt/ros/foxy/setup.zsh

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 12:57:51
$ colcon build
Starting >>> osiris_turret
Finished <<< osiris_turret [0.50s]          

Summary: 1 package finished [0.66s]

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 12:57:54
$ source install/local_setup.zsh

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 12:57:56
$ source venv/bin/activate
(venv) 
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 12:57:59
$ ros2 run osiris_turret service
Traceback (most recent call last):
  File "/home/orion/projects/skool/rover/next/science/osiris_turret_ros2/install/osiris_turret/lib/osiris_turret/service", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3254, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3237, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 3266, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 584, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 901, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 787, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'turret_python_interface' distribution was not found and is required by osiris-turret
(venv) 
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 12:58:02
$ 
```

theunkn0wn1, 2021-08-11:
@aprotyas it seems `venv` isn't adding anything to the `PYTHONPATH`, but rather extending `PATH` and setting the `VIRTUAL_ENV` key.
```
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:16:50
$ printenv  | grep -i venv

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:16:54
$ printenv  | grep -i pythonpath

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:16:59
$ source /opt/ros/foxy/setup.zsh

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:03
$ printenv  | grep -i pythonpath
PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:06
$ printenv  | grep -i venv      

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:14
$ source ./install/local_setup.zsh

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:21
$ printenv  | grep -i venv

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:24
$ printenv  | grep -i pythonpath  
PYTHONPATH=/home/orion/projects/skool/rover/next/science/osiris_turret_ros2/install/osiris_turret/lib/python3.8/site-packages:/opt/ros/foxy/lib/python3.8/site-packages

~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:26
$ source venv/bin/activate
(venv) 
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:31
$ printenv  | grep -i pythonpath
PYTHONPATH=/home/orion/projects/skool/rover/next/science/osiris_turret_ros2/install/osiris_turret/lib/python3.8/site-packages:/opt/ros/foxy/lib/python3.8/site-packages
(venv) 
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:34
$ printenv  | grep -i venv      
PATH=/home/orion/projects/skool/rover/next/science/osiris_turret_ros2/venv/bin:/opt/ros/foxy/bin:/home/orion/gems/bin:/home/orion/.cargo/bin:/home/orion/.cargo/bin:/home/orion/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
VIRTUAL_ENV=/home/orion/projects/skool/rover/next/science/osiris_turret_ros2/venv
PS1=(venv) 
(venv) 
~/projects/skool/rover/next/science/osiris_turret_ros2 ⌚ 13:17:37
$ 

```

clalancette, 2021-08-12:
OK, I see what is happening here.  The issue is in the script that python generates when we do the install.  For instance, in the original case in this PR, the build process ends up generating `install/venv/lib/venv/venv`.  Inside of that script, the preamble looks something like:

```
#!/usr/bin/python3
# EASY-INSTALL-ENTRY-SCRIPT: 'mytest==0.0.0','console_scripts','mytest'
import re
import sys
```

Because that is a hard-coded path to `/usr/bin/python3`, it isn't even looking inside of the virtual environment for packages.  What that really should be is `#!/usr/bin/env python3` so that it would pick up the venv properly.

I'm really not sure how we can fix that; that script is totally generated by Python itself.  We'll have to look into whether there is a way to have `EASY-INSTALL-ENTRY-SCRIPT` honor the venv, in a cross-platform way.

theunkn0wn1, 2021-08-12:
One possible solution  would be for `ros2 run` to detect if the specified executable is a python target, and run the entry-point using the environment's interpreter directly instead of depending on the shebang line. While this is a workaround, it would resolve the immediate issue.

I am not entirely familiar with how ROS2's py build process works, but if `EASY-INSTALL-ENTRY-SCRIPT` is referring to `easy-install`, [that is depricated](https://setuptools.readthedocs.io/en/latest/deprecated/easy_install.html).

Another possible option I am considering is running a post-processing script over the generated entry-points to programmatically replace the generated shebang lines.

theunkn0wn1, 2021-08-12:
Caused by https://github.com/colcon/colcon-core/issues/181
Fixed by, but not merged, https://github.com/colcon/colcon-core/pull/183.
I tried the fix in https://github.com/colcon/colcon-core/pull/188 but that seems to not effect the core issue.

yliu0062, 2021-09-25:
I think I also encounter this issue, after running this "sudo apt install ros-foxy-desktop", the pythonpath will have additional path "opt/ros/foxy/lib/python3.8/site-packages", then the right virtual python env can't be recognized correctly.

theunkn0wn1, 2021-09-25:
Here are some steps to creating a suitable venv:

1. source your ros distro's setup
2. create the venv, enabling system packages by symlink. `python3 -m venv venv  --system-site-packages --symlinks`
3. source the venv's setup script

In terms of creating Python ROS2 packages, create it as normal but add apply the fix in https://github.com/colcon/colcon-core/pull/183 to your setup.cfg

When building your nodes, be sure you've sourced both ros's setup as well as the venv's, BEFORE invoking `colcon build`.

miRemid, 2022-02-22:
thanks to @theunkn0wn1 , it works

devrite, 2022-02-22:
Since we are working with (potentially single vens). Instead of adding the "fixed" version, colcon already extracts the correct interpreter if built from your workspace or when it is reinstalled in your venv using something like

`pip install --force-reinstall colcon-core setuptools==$(pip list --no-index --format=json |  jq -r '.[] | select(.name=="setuptools").version')`

Of course you can not auto maintain multiple installations along side it (single colcon install). You need it for every venv. Which is a reasonable compromise, I guess. The above setuptools limit is only needed if an upgraded version causes issues. For example the current (py)torch still accesses features of distutils at runtime which have been removed in newer versions. But pip auto upgrades it for some reason although it is not required. Only-if-needed does still install a newer version.

When colcon core is being rebuilt it will your sys.executable path and add it to your python scripts as shebang. In case of a venv or any python env this is the interpreter path of your executable. No need to manually set the PYTHONPATH or using the specific fix. If you want to work with a single install of colcon though this may be the version to go.

mshahbionaut, 2022-08-13:
This thread was super helpful. I couldn't get @theunkn0wn1 's solution to work for me, but I might be doing something different (like using `virtualenv` instead of `venv`, not sure how/why that would be different)

My application is simple for now, and I can just rely on the system python and install my packages there. However, is there a long term cleaner solution even possible for this? Could we try to update [documentation here](https://docs.ros.org/en/humble/How-To-Guides/Using-Python-Packages.html) to reflect the current behavior and/or the fix?

Thanks!

jdlangs, 2023-03-15:
Would this also be fixed by addressing [colcon using the deprecated method](https://github.com/colcon/colcon-core/issues/454) of installing by directly invoking `setup.py`?

jdlangs, 2023-04-11:
Just to follow up on this, I found everything works well if I just run `python -m colcon` instead of `colcon` when the virtualenv is active. No need to change any files in your packages.

xiaolong73, 2023-06-24:
I found a way that you can use virtual python's package like torch.
```
import sys
sys.path.append('{YOUR VIRTUAL PYTHON}/lib/python3.10/site-packages')
import torch
```

theunkn0wn1, 2023-06-24:
> I found a way that you can use virtual python's package like torch.
> 
> ```
> import sys
> sys.path.append('{YOUR VIRTUAL PYTHON PATH}/lib/python3.10/site-packages')
> import torch
> ```

Note: path hacks make the code very brittle, break isolation, and make environments difficult to reproduce.
I encourage you to find a better solution.

Ryanf55, 2023-09-02:
> 2\. python3 -m venv venv  --system-site-packages --symlinks

This doesn't work, still can't find `lark`, despite it being installed.
```
--- stderr: nav2_msgs                                                                                                                                                                                                                       
Traceback (most recent call last):                                                                                                                                                                                                          
  File "/home/ryan/Development/ros2_rolling/install/rosidl_generator_type_description/lib/rosidl_generator_type_description/rosidl_generator_type_description", line 21, in <module>                                                        
    from rosidl_generator_type_description import generate_type_hash
  File "/home/ryan/Development/ros2_rolling/install/rosidl_generator_type_description/lib/python3.11/site-packages/rosidl_generator_type_description/__init__.py", line 24, in <module>
    from rosidl_parser.parser import parse_idl_file
  File "/home/ryan/Development/ros2_rolling/install/rosidl_parser/lib/python3.11/site-packages/rosidl_parser/parser.py", line 20, in <module>
    from lark import Lark
```
```
(.rolling) ryan@ryan-B650-970:~/Development/nav2_ws/src/navigation2$ python3 -m pip list | grep lark
lark                                 1.1.5
```

> python -m colcon

This doesn't work because:
```
$ which colcon
/usr/bin/colcon
```

devrite, 2023-09-02:
> This doesn't work, still can't find `lark`, despite it being installed.
>
> ```
> --- stderr: [...]

Have you tried reinstalling colcon-core to your env?

Ryanf55, 2023-09-02:
> Have you tried reinstalling colcon-core to your env?

I've removed the use of `colcon` through apt and will now try through python instead. 

Here's the latest progress of using `venv` with Ubuntu 23 and ROS 2 humble: 
https://github.com/Ryanf55/ros-on-lunar/blob/add-lunar-dockerfile/Dockerfile

TZECHIN6, 2023-09-04:
So... is it recommended to use venv in ros2 development? [ros2 document](https://docs.ros.org/en/humble/How-To-Guides/Using-Python-Packages.html#installing-via-a-package-manager) shows a way to use venv, but simply its not working....

jdlangs, 2023-09-14:
> > python -m colcon
> 
> This doesn't work because:
> 
> ```
> $ which colcon
> /usr/bin/colcon
> ```

The purpose of using `python -m colcon` is exactly that it *doesn't* invoke `/usr/bin/colcon`. If you run colcon explicitly through your virtual env interpreter version then setuptools will use that one inside the generated console scripts instead of `/usr/bin/python3`.

I'm not sure if it was mentioned above, but it also works to `pip install colcon-common-extensions` into your virtualenv and make sure `which colcon` points to that one. In both cases, the main limitation is that you have to have the virtualenv active when you build the workspace.

jjdengjj, 2024-04-20:
Try this solution:
1. Modify "setup.cfg" in your ros2 python module as follows: 
```
[build_scripts]
executable = /usr/bin/env python3
```
2. Rebuild your ros2 package, and then source the setup.bash in the terminal.
3. Activate the corresponding virtual environment by source the activate file.
4. Run your module with ```ros2 run .... ```, which will be able to identify the right virtual python environment and installed package in it.

hugoyvrn, 2024-04-23:
Thank you @jjdengjj for your quick reply. I understood what you said, and it works fine for me. 

But I do want to use the `--symlink-install` flag. When I do, my python node runs on the default python, not in the virtual env as I expect.

Hawxxer, 2024-06-21:
Sorry but how can such a fundamental issue be 3 years old? Is this not applicant for all users, do we overlook sth? Are majority just install their packages on the plain machine? Are they just using rosdep?

dghw, 2024-08-02:
I think the 'Installing via a virtual environment' section of https://docs.ros.org/en/jazzy/How-To-Guides/Using-Python-Packages.html should be removed or point to this issue to prevent wasted effort. 

I am confused on best practices now to integrate an external/custom Python package with ROS2 code, specifically one that is not installable via package manager (apt/pip).

Is the only way forward to strip the package in to a module within a ROS2 node and maintain the standalone package/module version separately...? Any tips from others who have met this issue?

alaurenzi, 2024-09-20:
Dear maintainers,
is there any plan to address this issue?

With Ubuntu24 + Jazzy (and therefore Python 3.12) it is not possible to pip install software outside of a virtualenv without "breaking system packages" as you know. This makes solving this issue pretty critical. Or am I missing something?

clalancette, 2024-09-21:
> is there any plan to address this issue?

Yes and no.  Yes, there is an open set of PRs that I need to get back to that will improve the situation.

However, it will probably not be possible to solve this in the general case.  In particular, if you have a custom version of Python and/or pip packages installed that are compiled in a venv, that is pretty much fundamentally incompatible with ROS 2 binaries.  It may possible to make that scenario work in a pure from-source build, which I'll look into when I open the documentation PR that corresponds to https://github.com/ros2/ros2/pull/1524

victorcarvesk, 2024-10-30:
> Dear maintainers, is there any plan to address this issue?
> 
> With Ubuntu24 + Jazzy (and therefore Python 3.12) it is not possible to pip install software outside of a virtualenv without "breaking system packages" as you know. This makes solving this issue pretty critical. Or am I missing something?

I’m currently facing this same issue (with the same setup), and all the proposed solutions seem to be more of a workaround than a robust solution. Why is it not possible to just specify the shebang in the Python scripts inside an `ament_python` package and be done with it? I tried this approach, but the shebang seems to be ignored in `ament_python` packages.

Using Python scripts in `ament_cmake` packages with shebangs (`#!/usr/bin/env python3`) inside virtual environments works seamlessly. However, in theory, `ament_python` packages should be better suited for handling pure Python packages.

Considering that all Linux systems (and possibly other systems) have an environment variable pointing to a Python interpreter, it's strange that ROS simply ignores it and uses a predefined interpreter.

ANogin, 2024-11-28:
> Dear maintainers, is there any plan to address this issue?
> 
> With Ubuntu24 + Jazzy (and therefore Python 3.12) it is not possible to pip install software outside of a virtualenv without "breaking system packages" as you know. This makes solving this issue pretty critical. Or am I missing something?

I came here exactly because of this. IMHO the right solution for the specific Ubuntu24 issue is the opposite of "allow ROS2 to work with arbitrary pre-existing venv" - instead, can we implement a "use ROS install as the venv"? Namely, what I am advocating for is to allow `ament_python_install_package` to specify a set of `requirements.txt` files and have those packages pip-installed into the *same* `site-packages` that the package itself is being installed under. 

For now, I am able to do it manually using
```
colcon build
source install/setup.bash
PYTHONUSERBASE="$(ros2 pkg prefix foo)" pip3 install --user --break-system-packages -r src/foo/requirements.txt
```
which installs those requirements not already satisfied elsewhere into `install/foo/lib/python3.12/site-packages` and this way, whatever the `PYTHONPATH` includes the foo, it will include its dependencies, and everything would work.

Does this approach make sense? Is it workable?

P.S. I find it weird that colcon creates a separate `lib/python3.12/site-packages` per package as opposed to using a shared one for all packages in the same workspace, but whatever...

amjad-haider, 2025-02-03:
Hi found a workaround for this situation.

Steps:
1. Create your ROS2 Workspace.
2. Create your virtual environment using `python3 -m venv <name of your environment>`.
3. Create your package in `src` folder inside your workspace with the help of `ros2 pkg create`. 
4. If your virtual environment is inside the workspace execute this command `touch <name of your environment>/COLCON_IGNORE`.
5. Activate your virtual environment and install your custom libraries.
6. In the terminal execute `export PYTHONPATH="<path to your virtual environment>/lib/python<python-version>/site-packages"`
7. Create your custom nodes in your package.
8. Change directory to workspace and execute `colcon build`.
9. Run your custom node.

Note:
- If you face some error with `pytest` while building comment those lines in `setup.py` ideally should be installed in the virtual environment.

Do comment if this works.

JunzheJosephZhu, 2025-02-04:
I managed to do a colcon build for multiple python packages with venv and symlink-install. Here's the guide I wrote:
1. install venv https://docs.ros.org/en/rolling/How-To-Guides/Using-Python-Packages.html
(Warning: cursor is breaking python venv for some reason... try not to create the venv inside cursor, or select interpreter) 
2. remove em, since this conflicts with empy
3. add this to setup.cfg: https://github.com/ros2/ros2/issues/1094#issuecomment-2067759726
    [build_scripts]
    executable = /usr/bin/env python3
4. sudo apt install jq, then in the venv install colcon and setuptools with compatible version:
    pip install --force-reinstall colcon-core setuptools==$(pip list --no-index --format=json |  jq -r '.[] | select(.name=="setuptools").version')
    pip install colcon-common-extensions
    check the correct colcon is used with       python -c "import colcon; print(colcon.__file__)" Maybe just change stuptools to 58.2.0. Empy 
5. pip install empy==3.3.4 lark catkin_pkg https://github.com/carla-simulator/ros-bridge/issues/712#issuecomment-2011659025
6. Source venv before build. In the build command, run python -m colcon instead of colcon
7. when running, no need to source venv

FelipeH92, 2025-02-05:
I managed to do it with a setup exactly like @amjad-haider a couple weeks back and forgot to comment, with the difference that I added the normal Python Path and Path also to use global python packages when they were not in the virtual env.

```
export PYTHONPATH=<path to your virtual environment>/lib/python3.10/site-packages:$PYTHONPATH
export PATH=<path to your virtual environment>/bin:$PATH
```

Achllle, 2025-02-17:
The cautionary note in [the how-to guide for using python packages](https://docs.ros.org/en/jazzy/How-To-Guides/Using-Python-Packages.html#using-python-packages-with-ros-2) mentions: 

> If you intend to use something like virtualenv or pipenv, make sure to use the system interpreter.

That means: `python3 -m virtualenv --system-site-packages my_venv`, which seems to solve the problem.

Achllle, 2025-02-17:
The cautionary note in [the how-to guide for using python packages](https://docs.ros.org/en/jazzy/How-To-Guides/Using-Python-Packages.html#using-python-packages-with-ros-2) mentions: 

> If you intend to use something like virtualenv or pipenv, make sure to use the system interpreter.

That means creating the virtualenv with `python3 -m virtualenv --system-site-packages my_venv`. Unfortunately that doesn't seem to solve the problem.

caro33, 2025-02-17:
> Hi found a workaround for this situation. Tested this with ROS2 Jazzy on Ubuntu 24.04 with python3.12, should work with other ROS-Distros as well.
> 
> Steps:
> 
> 1. Create your ROS2 Workspace.
> 2. Create your virtual environment using `python3 -m venv <name of your environment>`.
> 3. Create your package in `src` folder inside your workspace with the help of `ros2 pkg create`.
> 4. If your virtual environment is inside the workspace execute this command `touch <name of your environment>/COLCON_IGNORE`.
> 5. Activate your virtual environment and install your custom libraries.
> 6. In the terminal execute `export PYTHONPATH="<path to your virtual environment>/lib/python<python-version>/site-packages"`
> 7. Create your custom nodes in your package.
> 8. Change directory to workspace and execute `colcon build`.
> 9. Run your custom node.
> 
> Note:
> 
> * If you face some error with `pytest` while building comment those lines in `setup.py` ideally should be installed in the virtual environment.
> 
> Do comment if this works.

Sadly, After performing step 6, when I do colcon build, the following error occurs. Do you have any idea how to fix it? 
```
AttributeError: type object 'Callable' has no attribute '_abc_registry'
```

gc-robotics, 2025-02-20:
> Hi found a workaround for this situation. Tested this with ROS2 Jazzy on Ubuntu 24.04 with python3.12, should work with other ROS-Distros as well.
>
> Steps:
>
> 1. Create your ROS2 Workspace.
> 2. Create [...]

Unfortunately, this is not working for me:

a) venv activated at the very start.
b) successful colcon build from within the venv

```
(venv) janko@rasptank:~/ros2$ colcon build
Starting >>> hello_pubsub
Starting >>> test_venv
Finished <<< test_venv [14.9s]                                               
Finished <<< hello_pubsub [15.0s]            

```

c) sourced jazzy; sourced workspace
d) exported PYTHONPATH and PATH

```
(venv) janko@rasptank:~/ros2$ export PYTHONPATH=/home/janko/ros2/venv/lib/python3.12/site-packages:$PYTHONPATH
(venv) janko@rasptank:~/ros2$ export PATH=/home/janko/ros2/venv/bin:$PATH
```
e) execute package test_venv

```
(venv) janko@rasptank:~/ros2$ ros2 run test_venv test
[...]
  File "/home/janko/ros2/install/test_venv/lib/python3.12/site-packages/test_venv/test.py", line 1, in <module>
    import ws_281x
ModuleNotFoundError: No module named 'ws_281x'
[ros2run]: Process exited with failure 1
```

failed for dependency unmet. 

f) re-install pip package in venv

```
(venv) janko@rasptank:~/ros2$ python3 -m pip install rpi_ws281x
Requirement already satisfied: rpi_ws281x in ./venv/lib/python3.12/site-packages (5.0.0)
```

but the library is already there. 

Have I missed any of the required steps?

Mechazo11, 2025-03-04:
> [@123swk123](https://github.com/123swk123) I think the issue is due to the `--symlin-install` option.
> 
> [@jjdengjj](https://github.com/jjdengjj) It works for me without the `--symlin-install` option. But not with the `--symlin-install` option. Do you have a clue?

@123swk123  Thanks for the heads-up. I am not sure why to using the symlink flag causes this issue. But after getting rid of it, now I can use ros2 run for my python nodes inside my virtual enviornments

victorcarvesk, 2025-04-19:
I wouldn't recommend using `--break-system-packages` (self‑explanatory) or manually adjusting `PYTHONPATH` (it can break the self‑managed Python/ROS overlay mechanism and must be updated whenever you rename or move the workspace).

A simpler, more robust workaround until ROS provides a built‑in solution:

_Don't forget to source the ROS environment_

1. **Create the virtual environment with the --system-site-packages flag and activate it:**  
   ```bash
   python3 -m venv .venv --system-site-packages
   source .venv/bin/activate
   ```  
   _Colcon depends on other packages that are already in the system and are not part of ROS._

2. **Specify a dynamic python3 shebang in the `setup.cfg` file (add these 2 lines at the end of the file):**  
   ```ini
   [build]
   executable=/usr/bin/env python3
   ```  
   _Ensures auto‑generated scripts will use `/usr/bin/env` to locate the correct python3 interpreter at runtime rather than a hard‑coded path._

With these two simple steps, your workspace will be set up cleanly, predictably, and consistently.

---

**Tip:** to generate a `requirements.txt` containing only the packages in your virtual environment (excluding system‑wide ones), run:
```bash
pip list --local --format=freeze > requirements.txt
```

gavanderhoorn, 2025-04-19:
Or use [locusrobotics/ament_virtualenv](https://github.com/locusrobotics/ament_virtualenv) and/or [locusrobotics/catkin_virtualenv](https://github.com/locusrobotics/catkin_virtualenv)?

Betavion, 2025-05-12:
> Try this solution:
>
>     1. Modify "setup.cfg" in your ros2 python module as follows:
>
>
> ```
> [build_scripts]
> executable = /usr/bin/env python3
> ```
>
>     2. Rebuild your ros2 package, and then source [...]

This fixed my problem. Thanks @jjdengjj .

liuchangji, 2025-05-21:
I solved this.

# BuIild
```
conda activate your_environment_name
pip install rospkg
pip install -U colcon-common-extensions
python -m colcon build
```

# Run
If within conda env doesn't matter.

```
source install/setup.bashrc
ros2 run your_package your_nodes
```

Everything works fine so far.

ksomml, 2025-08-05:
> A simpler, more robust workaround until ROS provides a built‑in solution:
> 
> _Don't forget to source the ROS environment_
> 
> 1. **Create the virtual environment with the --system-site-packages flag and activate it:**
>    python3 -m venv .venv --system-site-packages
>    source .venv/bin/activate
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    _Colcon depends on other packages that are already in the system and are not part of ROS._
> 2. **Specify a dynamic python3 shebang in the `setup.cfg` file (add these 2 lines at the end of the file):**
>    [build]
>    executable=/usr/bin/env python3
>        
>          
>        
>    
>          
>        
>    
>        
>      
>    _Ensures auto‑generated scripts will use `/usr/bin/env` to locate the correct python3 interpreter at runtime rather than a hard‑coded path._
> 
> With these two simple steps, your workspace will be set up cleanly, predictably, and consistently.
> 
> **Tip:** to generate a `requirements.txt` containing only the packages in your virtual environment (excluding system‑wide ones), run:
> 
> pip list --local --format=freeze > requirements.txt

This was a quick fix for this problem wich still exists in 2025. Thanks! Hope there will be a internal solution soon.