ros2/ros2 #1096: [Foxy backport] Supporting ROS_LOCALHOST_ONLY for rmw_fastrtps

mitsudome-r, 2021-03-03:
## Feature request
There was a fix applied to rmw_fastrtps in master to fix the issues launching multiple nodes with ROS_LOCALHOST_ONLY set to 1(true). However, the issue still exists in foxy so I would like have them backported.

#### Feature description
https://github.com/ros2/ros2/issues/1052
https://github.com/ros2/rmw_fastrtps/pull/476
Applying changes made in the above pull reqest to Foxy.

Please close this issue if it is already ongoing somewhere else.

fujitatomoya, 2021-03-03:
i think this is not exactly feature but bug, so i guess it is okay to have https://github.com/ros2/ros2/issues/1052 backported into foxy. CC: @clalancette @jacobperron

fujitatomoya, 2021-03-04:
https://github.com/eProsima/Fast-DDS/pull/1520 is merged in https://github.com/eProsima/Fast-DDS/tree/2.0.x, so it is already available in foxy develop branch.

https://github.com/ros2/rmw_fastrtps/pull/514 is foxy backport for this issue. i confirmed that using https://github.com/ros2/ros2/blob/foxy/ros2.repos, problem can be fixed with this PR.

fujitatomoya, 2021-03-04:
@mitsudome-r 

no pressure but it would be appreciated if you could check https://github.com/ros2/rmw_fastrtps/pull/514 can solve the problem on your environment.

fujitatomoya, 2021-03-08:
@mitsudome-r 

backport PR is merged. could we close this issue?

jacobperron, 2021-03-08:
@fujitatomoya Thanks for resolving this ticket!

I'm going to close it since it seems fixed.

@mitsudome-r Please comment if there is still an issue.

closed 2021-03-08 by jacobperron