ros2/ros2 #1100: Create rolling branch for this repo

christophebedard, 2021-03-06:
## Feature request

This is more of a meta feature. Not sure if it was brought up before.

#### Feature description

It would be nice to be able to link to a `ros2.repos` file using simply: `https://raw.githubusercontent.com/ros2/ros2/$DISTRO/ros2.repos`.

Right now this isn't possible because we're using `master` here, and there is no `rolling` branch.

#### Implementation considerations

Simplest solution is having a `rolling` that is updated along with `master`, or the opposite (switching to `rolling` as the default branch and updating `master` to be equivalent). I admit that it would be a bit of a pain.

Other solution is to switch to `rolling` altogether. This would probably break a lot of stuff though.

clalancette, 2021-03-08:
My opinion of this is that it is not worth the churn.  Here are my reasons:

1. The name `rolling` actually isn't totally the same thing as what `master` is here.  `master` is all of the latest source versions.  But the Rolling distribution actually lags behind what's in the master branches.  That is, it is possible to merge a PR into the sources, and then not have a release of that code into Rolling for a few weeks.  So `master` and `rolling` aren't exactly synonyms.
1. As you mention we have a lot of baked-in knowledge that `master` is the default branch for the ros2.repos file.  While we could eventually find and fix all of them, it's a game of whack-a-mole that will probably last a while.

The benefit doesn't seem to outweigh those downsides in my opinion.

@ros2/team other thoughts on this?

jacobperron, 2021-03-08:
The same argument as (1) can be said for any of the other branches. E.g. a change lands in a `foxy` source branch, but is not released into Foxy until a time much later. This is a thing that isn't well documented (related: https://github.com/ros2/ros2_documentation/pull/1120).

However, I agree that the benefit doesn't seem to outweigh the disruption. I'm not aware of a way to redirect any URLs from the `master` branch to a new `rolling` branch, which would make the transition easier.

christophebedard, 2021-03-09:
Alright, that makes sense. Closing.

closed 2021-03-09 by christophebedard

christophebedard, 2021-03-09:
> I'm not aware of a way to redirect any URLs from the master branch to a new rolling branch, which would make the transition easier.

Although not a redirect per se, I just found this: https://github.com/google/mirror-branch-action. Seems like it should be pretty easy to set up.

```yaml
# .github/workflows/mirror.yml
on:
  push:
    branches:
      - 'master'
jobs:
  mirror_job:
    runs-on: ubuntu-latest
    name: Mirror master branch to rolling branch
    steps:
    - name: Mirror action step
      id: mirror
      uses: google/mirror-branch-action@v1.0
      with:
        github-token: ${{ secrets.GITHUB_TOKEN }}
        source: 'master'
        dest: 'rolling'
```

If people do end up relying on this, there's always a risk of it stopping working and breaking stuff, though. Anyway, just thought I'd point it out. Feel free to ignore.