ros2/ros2 #1103: Error with rcutils when running colcon build on Arch Linux

winstxnhdw, 2021-03-13:
## Bug report

**Required Info:**

- Operating System:
  - Arch Linux (Not on a ARM system)
- Installation type:
  - https://wiki.archlinux.org/index.php/ROS
  - https://docs.ros.org/en/foxy/Installation/Linux-Development-Setup.html
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Followed steps as per https://wiki.archlinux.org/index.php/ROS

When running:
```
colcon build --merge-install --install-base /opt/ros2/foxy
```

I get this error:
```
Failed   <<< rcutils [38.8s, exited with code 2]
Aborted  <<< rviz_assimp_vendor [2min 52s]                                                            
Aborted  <<< fastrtps [3min 58s]                          
Aborted  <<< rviz_ogre_vendor [5min 51s]
Summary: 115 packages finished [6min 6s]
1 package failed: rcutils
3 packages aborted: fastrtps rviz_assimp_vendor rviz_ogre_vendor
6 packages had stderr output: fastcdr fastrtps mimick_vendor osrf_testing_tools_cpp rcutils test_osrf_testing_tools_cpp
190 packages not processed
```

When I ignore the warnings and try to source anyways:
```
. ~/ros2_ws/install/local_setup.bash

# OR

source /opt/ros2/foxy/setup.bash
```
I get:
```
bash: /home/winston/ros2_ws/install/local_setup.bash: No such file or directory
```

#### Expected behavior
Colcon should complete with no errors.

#### Actual behavior
Fails with this error:
```
--- stderr: rcutils                                                                                                                                                                           
In file included from /home/winston/ros2_ws/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:35,
                 from /home/winston/ros2_ws/src/ros2/rcutils/test/test_filesystem.cpp:22:
/home/winston/ros2_ws/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:158:32: error: ‘__xstat’ was not declared in this scope; did you mean ‘_xstate’?
  158 |   MOCKING_UTILS_PATCH_TYPE(ID, __xstat) __xstat_mock_;
      |                                ^~~~~~~
/home/winston/ros2_ws/src/ros2/rcutils/test/./mocking_utils/patch.hpp:316:51: note: in definition of macro ‘MOCKING_UTILS_PATCH_TYPE’
  316 |   decltype(mocking_utils::make_patch<id, decltype(function)>("", nullptr))
      |                                                   ^~~~~~~~
/home/winston/ros2_ws/src/ros2/rcutils/test/./mocking_utils/filesystem.hpp:158:32: error: ‘__xstat’ was not declared in this scope; did you mean ‘_xstate’?
  158 |   MOCKING_UTILS_PATCH_TYPE(ID, __xstat) __xstat_mock_;
      |                                ^~~~~~~
/home/winston/ros2_ws/src/ros2/rcutils/test/./mocking_utils/patch.hpp:316:51: note: in definition of macro ‘MOCKING_UTILS_PATCH_TYPE’
  316 |   decltype(mocking_utils::make_patch<id, decltype(function)>("", nullptr))
[... 33 lines cut ...]
      |     ^~~~~~~~~~~~~
In file included from /usr/include/c++/10.2.0/functional:59,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest-printers.h:103,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest-matchers.h:46,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/internal/gtest-death-test-internal.h:39,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest-death-test.h:41,
                 from /opt/ros2/foxy/src/gtest_vendor/include/gtest/gtest.h:62,
                 from /home/winston/ros2_ws/src/ros2/rcutils/test/test_filesystem.cpp:15:
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<5>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<5>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
  601 |       function<_Res(_ArgTypes...)>::
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<6>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<6>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<7>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<7>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
/usr/include/c++/10.2.0/bits/std_function.h:601:7: error: ‘std::function<_Res(_ArgTypes ...)>::function(_Functor) [with _Functor = mocking_utils::filesystem::FileSystem<ID>::FileSystem<8>::<lambda(auto:1&& ...)>; <template-parameter-2-2> = void; <template-parameter-2-3> = void; _Res = __dirstream*; _ArgTypes = {const char*}]’, declared using local type ‘mocking_utils::filesystem::FileSystem<ID>::FileSystem<8>::<lambda(auto:1&& ...)>’, is used but never defined [-fpermissive]
make[2]: *** [CMakeFiles/test_filesystem.dir/build.make:82: CMakeFiles/test_filesystem.dir/test/test_filesystem.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1162: CMakeFiles/test_filesystem.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:160: all] Error 2
---
```

#### Additional information
- `vcs pull` shows that all sources are up to date.
- All packages are up to date when I performed `yay -Syu`
- I just want to build a default workspace. Nothing special.

clalancette, 2021-03-14:
This was likely fixed by https://github.com/ros2/rcutils/pull/330 on the master branch.  Can you follow the instructions at https://docs.ros.org/en/rolling/Installation/Linux-Development-Setup.html to see if it builds on rolling?

winstxnhdw, 2021-03-14:
When I run: 
```
rosdep install --from-paths src --ignore-src --rosdistro rolling -y --skip-keys "console_bridge fastcdr fastrtps rti-connext-dds-5.3.1 urdfdom_headers"
```
I get:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ament_cmake_nose: No definition of [python3-nose] for OS [arch]
ament_pyflakes: No definition of [pyflakes3] for OS [arch]
sros2: No definition of [python3-cryptography] for OS [arch]
rosidl_parser: No definition of [python3-lark-parser] for OS [arch]
rqt_msg: No definition of [python3-catkin-pkg-modules] for OS [arch]
rqt_shell: No definition of [python3-catkin-pkg-modules] for OS [arch]
ament_pep257: No definition of [pydocstyle] for OS [arch]
google_benchmark_vendor: No definition of [benchmark] for OS [arch]
rosidl_cmake: No definition of [python3-empy] for OS [arch]
rcl_logging_log4cxx: No definition of [python3-empy] for OS [arch]
rosidl_runtime_py: No definition of [python3-yaml] for OS [arch]
ros2cli: No definition of [python3-pkg-resources] for OS [arch]
rqt_plot: No definition of [python3-catkin-pkg-modules] for OS [arch]
python_cmake_module: No definition of [python3-dev] for OS [arch]
ros1_bridge: No definition of [python3-yaml] for OS [arch]
rosidl_adapter: No definition of [python3-empy] for OS [arch]
tracetools_trace: No definition of [python3-lttng] for OS [arch]
rqt_reconfigure: No definition of [python3-yaml] for OS [arch]
ament_package: No definition of [python3-flake8] for OS [arch]
ament_clang_format: No definition of [python3-yaml] for OS [arch]
ament_cmake_core: No definition of [python3-catkin-pkg-modules] for OS [arch]
rosidl_generator_py: No definition of [python3-numpy] for OS [arch]
tracetools_read: No definition of [python3-babeltrace] for OS [arch]
test_launch_ros: No definition of [python3-yaml] for OS [arch]
rcl_logging_spdlog: No definition of [spdlog] for OS [arch]
rcl_logging_noop: No definition of [python3-empy] for OS [arch]
ros2pkg: No definition of [python3-pkg-resources] for OS [arch]
ament_flake8: No definition of [python3-flake8] for OS [arch]
rqt_gui: No definition of [python3-catkin-pkg-modules] for OS [arch]
ament_clang_tidy: No definition of [python3-yaml] for OS [arch]
qt_gui: No definition of [python3-catkin-pkg-modules] for OS [arch]
rcutils: No definition of [python3-empy] for OS [arch]
rviz_ogre_vendor: No definition of [libfreetype6-dev] for OS [arch]
ros2doctor: No definition of [python3-rosdistro-modules] for OS [arch]
launch_ros: No definition of [python3-yaml] for OS [arch]
ros2service: No definition of [python3-yaml] for OS [arch]
launch_testing: No definition of [python3-mock] for OS [arch]
ros2topic: No definition of [python3-yaml] for OS [arch]
rqt_publisher: No definition of [python3-catkin-pkg-modules] for OS [arch]
osrf_pycommon: No definition of [python3-mock] for OS [arch]
ament_xmllint: No definition of [libxml2-utils] for OS [arch]
launch: No definition of [python3-lark-parser] for OS [arch]
uncrustify_vendor: No definition of [uncrustify] for OS [arch]
spdlog_vendor: No definition of [spdlog] for OS [arch]
```

winstxnhdw, 2021-03-14:
> This was likely fixed by [ros2/rcutils#330](https://github.com/ros2/rcutils/pull/330) on the master branch. Can you follow the instructions at https://docs.ros.org/en/rolling/Installation/Linux-Development-Setup.html to see if it builds on rolling?

Thanks! I no longer get that error but now I got this error for multiple packages. The error message is the same but for different packages each time.

```
--- stderr: rmw                                                                                                                                                                                              
CMake Error at test/CMakeLists.txt:2 (find_package):
  By not providing "Findosrf_testing_tools_cpp.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "osrf_testing_tools_cpp", but CMake did not find one.

  Could not find a package configuration file provided by
  "osrf_testing_tools_cpp" with any of the following names:

    osrf_testing_tools_cppConfig.cmake
    osrf_testing_tools_cpp-config.cmake

  Add the installation prefix of "osrf_testing_tools_cpp" to
  CMAKE_PREFIX_PATH or set "osrf_testing_tools_cpp_DIR" to a directory
  containing one of the above files.  If "osrf_testing_tools_cpp" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< rmw [1.03s, exited with code 1]
Aborted  <<< rcpputils [7.86s]
Aborted  <<< rosidl_typesupport_introspection_c [0.98s]
Aborted  <<< rosidl_generator_c [1.16s]                                                                               
Aborted  <<< rviz_rendering [13.7s]  
```

clalancette, 2021-03-15:
That package should be built by the colcon build process, so I'm not sure why it is not being found.

Unfortunately Arch Linux isn't one of our supported platforms, and we don't really do any testing there.  So building on it is going to be a bit of an adventure.  We're happy to take patches to fix things, however.

winstxnhdw, 2021-03-15:
> That package should be built by the colcon build process, so I'm not sure why it is not being found.
>
> Unfortunately Arch Linux isn't one of our supported platforms, and we don't really do any [...]

Perfectly understandable. I think I will just try running a Docker image instead. Thank you for your time and help.

closed 2021-03-15 by winstxnhdw