ros2/ros2 #1107: dyld library not loaded (QtWidgets); Reason: Image not found

A-Padmanabhan, 2021-03-18:
## Bug report

**Required Info:**

- Operating System:

  - MacOS Big Sur (11.2.3)
  
- Installation type:

  - ROS2 Foxy (binary)

#### Steps to reproduce issue

Executing "ros2 run turtlesim turtlesim_node" results in the following error: 

  dyld: Library not loaded: /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Path/to/directory/ros2_foxy/install/turtlesim/lib/turtlesim/turtlesim_node
  Reason: image not found

#### Additional information

The same workspace and installation worked perfectly in the previous version of the OS (11.2.2) without any glitches. There was no update/changes, either to the ros2 or Qt directories, between the two versions. I understand Big Sur is not officially supported for Foxy, but since it worked well on one version (Rviz, Gazebo and all in 11.2.2) I'm just a little hopeful.

clalancette, 2021-03-18:
If you can identify and fix the issue, we're happy to review a patch for it.  But given that it isn't one of the supported platforms, we won't likely be spending any time on it ourselves.

Spicedwater, 2021-05-07:
Hi @AniPadmanabhan, I had this problem too and found this thread helpful:
`https://stackoverflow.com/questions/17703510/dyld-library-not-loaded-reason-image-not-found`

First I would make sure you have qt5 installed using `brew list qt@5` and if you haven't, install it, `brew install qt@5`.

The installed packages should be located in `/usr/local/Cellar/qt@5`

Using `otool -L /Path/to/directory/ros2_foxy/install/turtlesim/lib/turtlesim/turtlesim_node` will show you where it is looking for the faulty image.  Then use `install_name_tool -change /old/path /new/path /path/to/exe` to change the location ros2 searches for the image.  

I had 3 errors when running this myself (QtWidgets, QtGUI, and QtCore)

I hope this helps!

Hongkuan-Zhou, 2021-05-09:
Yes, exactly @Spicedwater. It works for me. I just changed this  path `/usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets ` to `/usr/local/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets ` with the command `install_name_tool -change xxx xxx xxx`

chetanpm, 2021-11-24:
> otool -L /Path/to/directory/ros2_foxy/install/turtlesim/lib/turtlesim/turtlesim_node

What is the path to the executable? is it the Qt5 exec?

wgrand, 2021-12-29:
> Hi @AniPadmanabhan, I had this problem too and found this thread helpful: `https://stackoverflow.com/questions/17703510/dyld-library-not-loaded-reason-image-not-found`
>
> First I would make sure you [...]

Certainly I am not understanding some fundamentals, but what is `/path/to/exe` in this case? On macOS Monterrey (12.1), I put in `install_name_tool -change /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets /usr/local/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets`, but I get `Usage: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool [-change old new] ... [-rpath old new] ... [-add_rpath new] ... [-delete_rpath old] ... [-id name] input`.

Spicedwater, 2021-12-29:
> > Yes, exactly @Spicedwater. It works for me. I just changed this path `/usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets ` to `/usr/local/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets ` with the command `install_name_tool -change xxx xxx xxx`
> 
> What is the path to the executable? is it the Qt5 exec?

Hi, the executable in this case is the ros2 executable.  In my example that is the turtlesim_node exe.  Change this for the executable that you are trying to run and hope for the best!

good luck!

Spicedwater, 2021-12-29:
> > Using `otool -L /Path/to/directory/ros2_foxy/install/turtlesim/lib/turtlesim/turtlesim_node` will show you where it is looking for the faulty image. Then use `install_name_tool -change /old/path /new/path /path/to/exe` to change the location ros2 searches for the image.
> 
> Certainly I am not understanding some fundamentals, but what is `/path/to/exe` in this case? On macOS Monterrey (12.1), I put in:
> 
> ```
> install_name_tool -change /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets /usr/local/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets
> ```
> 
> But, I get:
> 
> ```
> Usage: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool [-change old new] ... [-rpath old new] ... [-add_rpath new] ... [-delete_rpath old] ... [-id name] input
> “

Hi Wgrand, 
 
 I would see the above answer and make sure you are putting the executable in your command.  The command as you have put it doesn’t make any sense as the command doesn’t know what executable to change.

wgrand, 2021-12-29:
Thanks for getting back to me @Spicedwater. That is what I am asking. What is the executable? @chetanpm asked the same question.

wgrand, 2021-12-29:
> > > Yes, exactly @Spicedwater. It works for me. I just changed this path `/usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets ` to `/usr/local/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets ` with the command `install_name_tool -change xxx xxx xxx`
> > 
> > 
> > What is the path to the executable? is it the Qt5 exec?
> 
> Hi, the executable in this case is the ros2 executable. In my example that is the turtlesim_node exe. Change this for the executable that you are trying to run and hope for the best!
> 
> what this command is doing is changing where the executable is looking for it’s dependencies. The error arises when the executable looks where it has been told to look, but cannot find what it has been told to look for. We are just telling the executable to look in a new place.
> 
> If this doesn’t work, I will be back at my work computer in a few days and can look into it
> 
> good luck!

Thank you @Spicedwater! Here is what I did:

```
install_name_tool -change /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets /usr/local/opt/qt@5/lib/QtWidgets.framework/Versions/5/QtWidgets /path/to/turtlesim_node
```

And repeated the same command by replacing `QtWidgets` with `QtGui` and `QtCore`.

wuyue92tree, 2022-07-07:
This work for me.
```
brew unlink qt
brew install qt@5
brew link --force qt@5

cd /usr/local/opt
rm qt
ln -s ../Cellar/qt@5/5.15.5 qt
```

ifbbprochris, 2022-09-03:
@wuyue92tree  it works for me, thanks