ros2/ros2 #1110: fresh container and installation - cant start examples

MightyMirko, 2021-03-19:
Linux rosmaster-VirtualBox 5.8.0-45-generic #51~20.04.1-Ubuntu SMP Tue Feb 23 13:46:31 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

binaries 
- Version or commit hash:

https://github.com/ros2/ros2/releases/download/release-foxy-20201211/ros2-foxy-20201211-linux-focal-amd64.tar.bz2

#### Steps to reproduce issue
Follow installation doc:
https://docs.ros.org/en/foxy/Installation/Linux-Install-Binary.html#

#### Expected behavior

> You should see the talker saying that it’s Publishing messages and the listener saying I heard those messages. This verifies both the C++ and Python APIs are working properly. Hooray!

#### Actual behavior
```

rosmaster@rosmaster-VirtualBox:~/ros2_foxy$ . ~/ros2_foxy/ros2-linux/setup.bash
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/opt/rti.com/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
rosmaster@rosmaster-VirtualBox:~/ros2_foxy$ ros2 run demo_nodes_cpp talker
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to initialize rcl init options: failed to load shared library of rmw implementation: /home/rosmaster/ros2_foxy/ros2-linux/lib/librmw_fastrtps_cpp.so Exception: LoadLibrary error: libtinyxml2.so.6: cannot open shared object file: No such file or directory, at /home/jenkins-agent/workspace/ci_packaging_linux/ws/src/ros2/rcutils/src/shared_library.c:84, at /home/jenkins-agent/workspace/ci_packaging_linux/ws/src/ros2/rmw_implementation/rmw_implementation/src/functions.cpp:62, at /home/jenkins-agent/workspace/ci_packaging_linux/ws/src/ros2/rcl/rcl/src/rcl/init_options.c:61
```

#### Additional information
Fresh container ...

clalancette, 2021-03-19:
The proximate reason is that you probably don't have `libtinyxml2` installed.  But the underlying reason for that is probably because the rosdep command to install dependencies failed.  I notice that we have a slight bug in the installation instructions there; the rosdep command should be:

```
rosdep install --from-paths ~/ros2_foxy/ros2-linux/share --ignore-src --rosdistro foxy -y --skip-keys "console_bridge fastcdr fastrtps osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers"
```

Can you try that out and see if it works better for you?  That command should end up installing a lot of packages.

closed 2021-03-22 by clalancette

MightyMirko, 2021-03-23:
Hi thank you.

Unfortunately, there is still an error:

<pre><font color="#4E9A06"><b>osmaster@rosmaster-VirtualBox</b></font>:<font color="#3465A4"><b>~/ros2_foxy</b></font>$ rosdep install --from-paths ~/ros2_foxy/ros2-linux/share --ignore-src --rosdistro foxy -y --skip-keys &quot;console_bridge fastcdr fastrtps osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers&quot;
<b>executing command [sudo -H apt-get install -y python3-flake8]</b>
[sudo] Passwort für rosmaster: 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib pyflakes
  pyflakes3 python-pkg-resources python-pyflakes python2 python2-minimal
  python2.7 python2.7-minimal python3-mccabe python3-pycodestyle
  python3-pyflakes
Vorgeschlagene Pakete:
  python-setuptools python2-doc python-tk python2.7-doc binfmt-support
Die folgenden NEUEN Pakete werden installiert:
  libpython2-stdlib libpython2.7-minimal libpython2.7-stdlib pyflakes
  pyflakes3 python-pkg-resources python-pyflakes python2 python2-minimal
  python2.7 python2.7-minimal python3-flake8 python3-mccabe
  python3-pycodestyle python3-pyflakes
0 aktualisiert, 15 neu installiert, 0 zu entfernen und 4 nicht aktualisiert.
Es müssen 4.146 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 18,2 MB Plattenplatz zusätzlich benutzt.
Holen:1 http://de.archive.ubuntu.com/ubuntu focal-updates/universe amd64 libpython2.7-minimal amd64 2.7.18-1~20.04.1 [335 kB]
Holen:2 http://de.archive.ubuntu.com/ubuntu focal-updates/universe amd64 python2.7-minimal amd64 2.7.18-1~20.04.1 [1.285 kB]
Holen:3 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python2-minimal amd64 2.7.17-2ubuntu4 [27,5 kB]
Holen:4 http://de.archive.ubuntu.com/ubuntu focal-updates/universe amd64 libpython2.7-stdlib amd64 2.7.18-1~20.04.1 [1.887 kB]
Holen:5 http://de.archive.ubuntu.com/ubuntu focal-updates/universe amd64 python2.7 amd64 2.7.18-1~20.04.1 [248 kB]
Holen:6 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 libpython2-stdlib amd64 2.7.17-2ubuntu4 [7.072 B]
Holen:7 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python2 amd64 2.7.17-2ubuntu4 [26,5 kB]
Holen:8 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python-pkg-resources all 44.0.0-2 [129 kB]
Holen:9 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python-pyflakes all 2.1.1-2 [48,5 kB]
Holen:10 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 pyflakes all 2.1.1-2 [3.304 B]
Holen:11 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python3-pyflakes all 2.1.1-2 [48,6 kB]
Holen:12 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 pyflakes3 all 2.1.1-2 [3.340 B]
Holen:13 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python3-mccabe all 0.6.1-3 [8.488 B]
Holen:14 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python3-pycodestyle all 2.5.0-2 [36,7 kB]
Holen:15 http://de.archive.ubuntu.com/ubuntu focal/universe amd64 python3-flake8 all 3.7.9-2 [51,8 kB]
Es wurden 4.146 kB in 7 s geholt (624 kB/s).                                   
Vormals nicht ausgewähltes Paket libpython2.7-minimal:amd64 wird gewählt.
(Lese Datenbank ... 211096 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../0-libpython2.7-minimal_2.7.18-1~20.04.1_amd64.deb ...
Entpacken von libpython2.7-minimal:amd64 (2.7.18-1~20.04.1) ...
Vormals nicht ausgewähltes Paket python2.7-minimal wird gewählt.
Vorbereitung zum Entpacken von .../1-python2.7-minimal_2.7.18-1~20.04.1_amd64.deb ...
Entpacken von python2.7-minimal (2.7.18-1~20.04.1) ...
Vormals nicht ausgewähltes Paket python2-minimal wird gewählt.
Vorbereitung zum Entpacken von .../2-python2-minimal_2.7.17-2ubuntu4_amd64.deb ...
Entpacken von python2-minimal (2.7.17-2ubuntu4) ...
Vormals nicht ausgewähltes Paket libpython2.7-stdlib:amd64 wird gewählt.
Vorbereitung zum Entpacken von .../3-libpython2.7-stdlib_2.7.18-1~20.04.1_amd64.deb ...
Entpacken von libpython2.7-stdlib:amd64 (2.7.18-1~20.04.1) ...
Vormals nicht ausgewähltes Paket python2.7 wird gewählt.
Vorbereitung zum Entpacken von .../4-python2.7_2.7.18-1~20.04.1_amd64.deb ...
Entpacken von python2.7 (2.7.18-1~20.04.1) ...
Vormals nicht ausgewähltes Paket libpython2-stdlib:amd64 wird gewählt.
Vorbereitung zum Entpacken von .../5-libpython2-stdlib_2.7.17-2ubuntu4_amd64.deb ...
Entpacken von libpython2-stdlib:amd64 (2.7.17-2ubuntu4) ...
libpython2.7-minimal:amd64 (2.7.18-1~20.04.1) wird eingerichtet ...
python2.7-minimal (2.7.18-1~20.04.1) wird eingerichtet ...
Linking and byte-compiling packages for runtime python2.7...
python2-minimal (2.7.17-2ubuntu4) wird eingerichtet ...
Vormals nicht ausgewähltes Paket python2 wird gewählt.
(Lese Datenbank ... 211843 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../0-python2_2.7.17-2ubuntu4_amd64.deb ...
Entpacken von python2 (2.7.17-2ubuntu4) ...
Vormals nicht ausgewähltes Paket python-pkg-resources wird gewählt.
Vorbereitung zum Entpacken von .../1-python-pkg-resources_44.0.0-2_all.deb ...
[... 58 lines cut ...]
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket python3-catkin-pkg-modules kann nicht gefunden werden.
[... cut ...] t_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers&quot;
<b>executing command [sudo -H apt-get install -y python3-catkin-pkg-modules]</b>
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
E: Paket python3-catkin-pkg-modules kann nicht gefunden werden.
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y python3-catkin-pkg-modules] failed
</pre>

clalancette, 2021-03-23:
That looks like your apt sources are incorrect somehow.  I'd go back over the install instructions and make sure that your /etc/apt/sources.list.d/ros2-latest.list matches what is there.  If you are still having problems, please open a question on https://answers.ros.org