ros2/ros2 #1118: Accept a PR to add ros/angles to ros2.repos?

SteveMacenski, 2021-03-25:
Would maintainers accept a PR to add this to the repos file? I believe this has sufficient general use that it should belong.

clalancette, 2021-03-25:
The question for me is why it is needed.  That is, what use-case do we want to support that isn't being handled by the package being released for both the latest release (Foxy) and the development release (Rolling)?

SteveMacenski, 2021-03-25:
Nav2 has a [heavy dependency](https://github.com/ros-planning/navigation2/search?q=angles) on angles and it would be nice to have it included by default in the docker images we rely on for CI to speed up builds. It's also a low-overhead package that has alot of marginal utility to users. Nav2 also makes use of master builds for our development and CI. So most Nav2 users and all developers will be building `ros2` from `ros2.repos`, the less things we have in the underlay workspace, the faster development of Nav2 can occur.  

It came to my attention that [image_common](https://github.com/ros2/ros2/pull/884) was added recently. I'm not familiar with the burden of need / criteria to have packages included here so I figured it would be good to ask. Of our [underlay dependencies](https://github.com/ros-planning/navigation2/blob/main/tools/underlay.repos) we require for Nav2 on top of the `ros2.repos`, angles is the other that makes most sense to me to have included (though `bond` and `gazebo_ros_pkgs` seem like good potential targets as well).

I won't die on the hill about this any way it goes, just for convenience and curiosity + I think its useful for folks to have these.

clalancette, 2021-03-25:
> It came to my attention that [image_common](https://github.com/ros2/ros2/pull/884) was added recently. I'm not familiar with the burden of need / criteria to have packages included here so I figured [...]

image_common went in because rviz2 wants to grow a dependency on it: https://github.com/ros2/rviz/pull/523 .

In general, I don't think we should be adding additional dependencies to ros2.repos unless they are needed in the core.  There are a couple of different reasons for that:
1.  We can't pull "everything" into ros2.repos.  Every time we add something, it adds more burden to the core.  This makes compiling and testing times longer, but more importantly puts more burden on the ROS 2 core team.  For every package that we add, it now becomes a de-facto release-critical blocker for doing anything in ROS 2 core.  So the core team essentially has to maintain every single package that is in ros2.repos.
2. The long term plan is to remove ros2.repos completely.  On Linux, it is more-or-less redundant with `rosdep` and `rosinstall_generator`, and we'd much rather switch to that.  But we haven't switched because a) it takes time and effort, and b) we don't have a good story on macOS and Windows.

Anyway, that's my thinking on it.  Others can chime in if they think differently.

SteveMacenski, 2021-03-25:
OK, I'm fine with them not being added - just wanted to check in on this process to see if it made sense :smile: 

On 1) OK, thanks sense, no worries. I figured `angles` was low effort and high impact, but I respect that position
On 2) Remove it completely? Where would the master list of "ros2 core packages" exist if not the `.repos` file?

I can close this - thanks for the info.

closed 2021-03-25 by SteveMacenski

clalancette, 2021-03-25:
> On 2) Remove it completely? Where would the master list of "ros2 core packages" exist if not the `.repos` file?

It's been a while since we discussed it, but if I remember properly it would come from the variant files: https://github.com/ros2/variants/blob/master/desktop/package.xml