ros2/ros2 #1124: Specify which network interface a ros node should use

Kaju-Bubanja, 2021-04-06:
## Feature request
Allow ros to specify which network interfaces the underlying middleware should use.
I think it would be a nice feature to be able to specify on a per node basis on which network interface the node should communicate on.

#### Feature description
Currently when having multiple network interfaces I receive the message 

`image_publ: using network interface enp2s0 (udp/10.11.12.1) selected arbitrarily from: enp2s0, enp3s0, enp4s0, br-5fd3d27662e1, docker0
`

which sometimes works out and other times doesn't. This feature would allow us to specify the interface no matter what middleware is used. Something along these lines or specified in the yaml config:

```
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    base_config_path = os.path.join(get_package_share_directory('my_project'), 'config')
    image_publisher_config0 = os.path.join(base_config_path, "camera", "image_publisher0.yaml")
    return LaunchDescription([
        Node(
            package='image_publisher',
            executable='image_publisher_node',
            namespace='cam0',
            parameters=[image_publisher_config0],
            networks=["enp2s0", "enp3s0"],
        ),
    ])
```

#### Implementation considerations
If the middleware doesn't support the idea of network interfaces the option can simply be ignored by the middlware and a warning can be emitted.

EduPonz, 2021-04-06:
Hi @Kaju-Bubanja ,

Although I understand that you'd like a ROS 2 API, I'd like to point out that [`rmw_fastrtps`](https://github.com/ros2/rmw_fastrtps) already allows for specifying a list of interfaces that the middleware is allowed to use. You can do that by setting a XML file to configure a whitelist of interfaces for [Fast DDS](https://github.com/eProsima/Fast-DDS). I'm leaving a couple of link that you might find useful:

* [Fast DDS whitelisting](https://fast-dds.docs.eprosima.com/en/latest/fastdds/transport/whitelist.html)
* [Configure rmw_fastrtps_cpp using XML](https://github.com/ros2/rmw_fastrtps/blob/master/README.md)

gavanderhoorn, 2021-04-06:
Also on ROS Answers: [How to specify the network interface ros2 uses for communication?](https://answers.ros.org/question/373132).

Kaju-Bubanja, 2021-04-06:
This is actually my question and has no answer yet

Kaju-Bubanja, 2021-04-06:
> Also on ROS Answers: [How to specify the network interface ros2 uses for communication?](https://answers.ros.org/question/373132).

This is actually my question and has no answer yet

Kaju-Bubanja, 2021-04-06:
> Hi @Kaju-Bubanja ,
>
> Although I understand that you'd like a ROS 2 API, I'd like to point out that [`rmw_fastrtps`](https://github.com/ros2/rmw_fastrtps) already allows for specifying a list of [...]

Thank you very much for the links. I was reading: https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/ros2_configure.html but it was less detailed then the above link. I thought I open the feature request and if nobody finds it important enough it can just sit open. I guess worst case people search for it and find the useful links you posted.

Kaju-Bubanja, 2021-04-07:
Also thinking about it, this would indeed be useful on setups with many nodes on different interfaces(like mine actually is). Because right now I have to split my one launch file into multiple ones and all call them seperately

fake-name, 2022-07-01:
This 1000% needs to be exposed through ros.

fujitatomoya, 2022-07-06:
Related to https://discourse.ros.org/t/unconfigured-dds-considered-harmful-to-networks/25689/25