ros2/ros2 #1127: Fci__nightly-release_ubuntu_focal_amd64 times out

jacobperron, 2021-04-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - https://github.com/ros2/ros2/blob/foxy/ros2.repos

Example failure: https://build.ros2.org/view/Fci/job/Fci__nightly-release_ubuntu_focal_amd64/257

jacobperron, 2021-04-08:
This commit is suspicious since the job has timed out twice in the middle of `test_launch_ros/actions/test_lifecycle_node.py`:

https://github.com/ros2/launch_ros/commit/f9bb0cab453e47ea72206fbdfc3940291c99383f

However, in previous builds we were already close to the time limit of 5 hours and I think recent changes have simply pushed us over the limit.

Qualitatively, tests in the Foxy job appear to run slightly longer than tests in the [Rolling equivalent](https://build.ros2.org/view/Rci/job/Rci__nightly-release_ubuntu_focal_amd64/). The only difference between the jobs (besides the source code) is that the Rolling job [skips tests marked with xfail](https://github.com/ros2/ros_buildfarm_config/blob/21d6136fb867b79b2682c8faa818cba768705f77/rolling/ci-nightly-release.yaml#L14). We might consider doing the same for Foxy.

j-rivero, 2021-04-09:
I've detailed in the [buildfarmer issue](https://github.com/osrf/buildfarmer/issues/174) the sequence of problems with the timeout. I'm also in favour of thinking that we are close to the limit. Note that serveral builds failed with timeout in different points of the build:
 * https://build.ros2.org/view/Fci/job/Fci__nightly-release_ubuntu_focal_amd64/257/
 * https://build.ros2.org/view/Fci/job/Fci__nightly-release_ubuntu_focal_amd64/256/
 * https://build.ros2.org/view/Fci/job/Fci__nightly-release_ubuntu_focal_amd64/252/

clalancette, 2022-09-07:
The only recent failures of this job seem to be infrastructure related, so I'm going to close this out as "fixed", one way or another.  Feel free to reopen if you think this was closed in error.

closed 2022-09-07 by clalancette