ros2/ros2 #1132: Add setuptools helper to deduplicate package.xml and setup.py metadata

christophebedard, 2021-04-20:
## Feature request

#### Feature description

See original issue: https://github.com/ament/ament_package/issues/122

ROS 1's `catkin_pkg` has [`catkin_pkg.python_setup.generate_distutils_setup`](https://github.com/ros-infrastructure/catkin_pkg/blob/ad07ab2570006a72994657ab3f3267a9e84469aa/src/catkin_pkg/python_setup.py#L43). It can be used to extract some metadata from a `package.xml` file to avoid having to maintain/sync the same data in a `setup.py` file for the same package (e.g. package description, maintainers, authors, etc.).

This is currently not available for ROS 2 & `setuptools`.

#### Implementation considerations

A first implementation was proposed in https://github.com/ament/ament_package/pull/126. However, there was a critical problem which made it unusable/fail on the ROS 2 builfarm: https://github.com/ament/ament_package/pull/126#issuecomment-820606420. The initial implementation was therefore reverted.

As mentioned in the discussion that I linked to above, it is not quite clear where such a method/helper should live, which is why this feature request/issue has been opened on ros2/ros2. Also, some changes might have to be made to the buildfarm.

Other relevant issues/PRs:

* https://github.com/ros2/ros2cli/pull/611
* https://github.com/ros2/ros2_documentation/pull/1250

DLu, 2021-04-22:
To make explicit the [question I asked here](https://github.com/ament/ament_package/pull/126#issuecomment-821278684)

To quote @nuclearsandwich 
> sourcedeb jobs which create our debian source packages are designed to run without any dependencies beyond the Debian toolchain

@nuclearsandwich : Why do we need to build debians without any external dependencies?

nuclearsandwich, 2021-04-24:
Thanks for restating the question. It wasn't my intention to drop the matter, I was waiting for an issue like this one (and the end-of-week energy push) to give an answer.

> https://github.com/ros2/ros2_documentation/pull/1385

> Why do we need to build debians without any external dependencies?

To make sure we're everyone who comes across the discussion is on the same page: Binary packages are by necessity produced with the declared dependencies available. Debian (and other platform) source packages are produced in separate jobs before binary packages and these jobs do not include any package-specific dependencies because packages should only require the necessary Debian toolchain packages to prepare source packages and not other external packages. This also helps these jobs run efficiently since there's no need for them to be created in package-specific container environments like the binary jobs are. It also helps keep ROS packages closer to standard behavior so that those who are building or reproducing a ROS environment independently aren't tripped up by hidden requirements. As far as I know, Debian doesn't have a mechanism to express dependencies which are required to produce a source package from base ingredients beyond those required by Debian generally.

The cost of generally installing dependencies during source package creation would be exceedingly high relative to the convenience feature that this would enable. But I do see the value in the convenience feature (and others which we considered in the past).

There is a significant difference between catkin_pkg and ament_package / [ament_python](https://github.com/ament/ament_python) that also has implications here:
The ament modules are released within each ROS distribution rather than as part of the bootstrap packages which means that using them requires sourcing the desired ROS distro.
Bloom packages already do this thanks to the template system and those using ROS packages will likely already be sourcing setup files or have the module sources in their Colcon workspaces.
There is no explicit fallout from this fact that comes to mind.
It means that updating the source package producing containers requires the "target" rosdistro for that source package to be in context, something which may not have been true previously but adding it is just work-- I don't see any risk or problem in doing so.

---

One of the goals for improvement between ament and catkin was to allow language-native build systems to be better used. To that end, I recognize the convenience of a helper which produces valid metadata for both setuptools and ROS package.xml. But one concern I have is whether the helper as it was, would perform as-expected in context as solely a Python package. Colcon is aware of ROS packaging and so knows from package.xml to include the helper module in the dependencies when building a Python package. Likewise the ROS buildfarm packaging pipeline will ensure that dependencies from package.xml are propagated into the platform-specific metadata and made available during package build. But what is the solution for the plain Python context? How are dependencies of a package's setup.py itself meant to be communicated in Python packages generally? I've always thought that it was considered something of an anti-pattern.

---

I'm still skeptical of putting anything in setup.py that isn't provided by setuptools or the Python standard library but here is what I would propose as a path forward:

* We'll take <https://github.com/ament/ament_python> out of mothballs with myself and @audrow (with his consent) as maintainers and cherry-pick the helper from https://github.com/ament/ament_package/pull/126 into it.
* Once we've done that and some initial testing we'll release that into Rolling and make changes to ros_buildfarm so that ament_python, and only ament_python, is available during sourcedeb and sourcerpm builds alongside catkin_pkg so that setup.py can be invoked as part of that process.
* If all goes well in Rolling (and there's enough provisional adoption that we are confident in the utility) I don't see a reason not to backport the new ament_python package into Foxy and Galactic so it can be used by packages there as well.

This approach gets the feature back into an unstable distribution swiftly without rendering our infrastructure "broken" until support for this is introduced. And if things go well it does not preclude the changes being made available in our stable distributions.

---

#### Why ament_python and not ament_package

ament_package is the base of all ROS 2 distributions. It contains template files that will be used by all build systems targeting ament packages. In its early days it had Python modules for processing package.xml but all of that was eventually rolled back into catkin_pkg and it became a bit of a template file repository feeding ament_cmake. In my opinion it makes sense to restrict the contents of this base package to just things which are necessary for and potentially common to any ament build system to avoid keeping multiple copies of the same data in multiple ament implementations.

ament_python was originally put together to create helper modules for python packages conforming to ament standards/guidelines but fell into this same trap and at the time changing the buildfarm to accommodate this was not pursued. Making the affordance for a specific package of helper modules supporting packages with the ament_python b [... cut ...] usly defined along with the new one contributed by @DLu out together very appropriate.

For completeness, I'll also mention [ros2/rpyutils](https://github.com/ros2/rpyutils) which occupies a place analogous to [ros2/rcutils](https://github.com/ros2/rcutils) and contains utilities commonly useful to ROS packages but not ROS specific. I don't think it belongs here either necessarily because ament and the ament_python build type may be used by Python packages that are also not strictly ROS-related.

clalancette, 2021-04-26:
Sorry, I'm having a bit of trouble reconciling:

> I'm still skeptical of putting anything in setup.py that isn't provided by setuptools or the Python standard library

with:

>     * We'll take https://github.com/ament/ament_python out of mothballs with myself and @audrow (with his consent) as maintainers and cherry-pick the helper from [ament/ament_package#126](https://github.com/ament/ament_package/pull/126) into it.

Aren't those in direct opposition to each other?  Shouldn't we discuss whether we really want to do that before moving forward with any plan here?

nuclearsandwich, 2021-04-26:
> Aren't those in direct opposition to each other? Shouldn't we discuss whether we really want to do that before moving forward with any plan here?

My proposal pays what I think is due respect to the momentum of precedent. I let a concrete quality of life improvement override an abstract personal reservation. Put another way: I've proposed what I think is the best way to deliver this. If there is instead consensus that it creates more problems than it solves and we cannot find a method that works in all cases where it would be expected to, then the proposal for how we do it loses relevance.

audrow, 2021-04-26:
> We'll take https://github.com/ament/ament_python out of mothballs with myself and @audrow (with his consent) as maintainers and cherry-pick the helper from ament/ament_package#126 into it.

I'm okay with this, if this is the way we decide to proceed.

jacobperron, 2021-04-27:
From what I've read so far, it doesn't sound like the extra work of installing user-specified dependencies for source deb jobs is worth the benefit. The proposal to install *only* ament_python for these jobs sounds like a reasonable compromise, giving users optional access to these utilities. However, I lean slightly towards not adding magic dependencies that may trip up others trying to reproduce our environment.

I believe @sloretz suggested this offline:
Alternatively, we could have a developer tool to help with generating (or validating) the contents of `setup.py` from a `package.xml`. This could be run explicitly by the developer (periodically) or as commit hook in the source repo. It's a bit more manual of a process, but circumvents any issues with installing dependencies on the buildfarm.

emersonknapp, 2021-04-27:
> Alternatively, we could have a developer tool to help with generating (or validating) the contents of setup.py from a package.xml

To think through this possibility -- Something like a `setup.py.in` template file that is the "true" source, and a tool is invoked to render it into `setup.py`, which is also checked in. Is that kind of the idea?

christophebedard, 2021-04-27:
> However, I lean slightly towards not adding magic dependencies that may trip up others trying to reproduce our environment.

from my point of view, the whole buildfarm environment is a magic environment in itself, so I don't see how this would change much :sweat_smile:

> Alternatively, we could have a developer tool to help with generating (or validating) the contents of setup.py from a package.xml. This could be run explicitly by the developer (periodically) or as [...]

That's an interesting alternative though. I'd personally like to avoid having to do anything manually, including having to set up a per-repo hook.

What if this was simply included in `catkin_prepare_release`? i.e. it reads and updates the `setup.py` file when updating the `package.xml` file to bump the version. We already run this command manually every release.

clalancette, 2021-04-27:
> What if this was simply included in `catkin_prepare_release`? i.e. it reads and updates the `setup.py` file when updating the `package.xml` file to bump the version. We already run this command manually every release.

We've considered modifying `catkin_prepare_release` to edit setup.py for the versions before, which would make releases of pure Python packages much easier.  We could expand that concept to have it sync "everything" from `package.xml` into the `setup.py`, which is fairly natural.

However, it is not without its pitfalls.  In particular, you can't guarantee that this is going to work; `setup.py` can be arbitrary Python code, so there are no guarantees that a simple string substitution will work.  That being said, I think it would work for 80% of the cases out there, so it is probably worthwhile doing (with some warnings if `catkin_prepare_release` failed to substitute strings for some reason).

jacobperron, 2021-04-27:
> Something like a setup.py.in template file that is the "true" source, and a tool is invoked to render it into setup.py, which is also checked in. Is that kind of the idea?

It could involve a template file, though I was thinking something more along the lines of what Chris mentioned: having a tool simply update an existing setup.py file in-place.

clalancette, 2021-04-29:
I just found out something else interesting (completely by accident).  If you look at https://github.com/ros-visualization/qt_gui_core/blob/3b859c4cd754c602038a4cc222d5da13a222c043/qt_dotgraph/setup.py for instance, you'll see that it uses `catkin_pkg.python_setup.generate_distutils_setup`.  Curious, I looked in https://build.ros2.org/view/Rsrc_uF/job/Rsrc_uF__qt_dotgraph__ubuntu_focal__source/8/console , and found that `python3-catkin-pkg-modules` is already injected into all debian source builds.

Looking around the ROS 2 core codebase, it looks like we use this in a few different places, though only sparingly.  And in all current cases where we use this, we use it in an `ament_cmake` package that calls `ament_python_install_package` to do python stuff.

Nevertheless, I think it is worth trying this functionality in a "pure Python" package and seeing if it works.

nuclearsandwich, 2025-04-03:
> If I'm not wrong, the major problem is importing non-default libraries in setup.py. Even other Python libraries fight with it (e.g. matplotlib) because it's a kind of chicken-and-egg problem. A systematic solution seems to be pyproject.toml builds, but that's kind of a different issue.

I don't see it as "a different" issue, but as the way forward to solve this problem without coming up with an arbitrary (and ROS-specific) compromise in packages that are only using setup.py. I'll get to how after addressing your questions.

> It seems colcon+rosdep would properly support custom imports in setup.py in their current state

Can you clarify where rosdep fits into the story you're hoping to enable? rosdep completely ignores setup.py and only examines package.xml to determine package dependencies. If a "no-op" / not failing is considered proper support, I guess so. But perhaps I'm missing another element of the workflow.

Colcon's "support" here again is passive. When you build python packages with colcon, whatever is in your environment is available in the build. So if your environment includes whatever you import in setup.py, it'll get picked up but managing dependencies in the environment is explicitly out-of-scope for colcon so it's not going to do any checking or enforcement of dependencies outside the current workspace.

> I've tested locally adding --no-pre-clean (-nc) to gbp-buildpackage and that seems to do the job. I can have any kind of gibberish in setup.py and the source deb gets built. It can even be invalid syntax. Or is there some reason the clean target would actually need to be called?

Although we default to building debs in a pristine environment on the buildfarm, I'd be hesitant to disable cleaning in bloom templates since packages can be built outside of just the buildfarm. I'd possibly be fine with supplying our own override for dh_clean in the python templates that accomplishes the same thing without reading setup.py.

I want to argue that there is actually value in invalid setup.py failing early but there's no analogue in our cmake-based build types: The dh_clean step for cmake does not depend on cmake at all.

> Or are there cases where other things than python3 setup.py clean would be needed during sourcedeb phase?

It's actually only the first thing that failed only. I'm not sure whether there are others. It could be tested with `generate_release_script.py` and a mocked rosdistro locally. I didn't try to unblock and see past that phase at the time since once that first failure occurred other collaborators lost favor in the solution and decided to remove ament_python from setup.py in favor of the copied logic approach used today.

---

We've been working on the post-setuptools support in ROS tooling for some time and while we've been unable to reach full parity. I would encourage investing more in this tooling than in trying to further enable setuptools-based workflows when they've been pelting us with deprecation warnings for years now and it's only a matter of time before the current workflows just break down.

The standards-based builds also afford us new opportunities to _genuinely solve_ some of the ambiguous chicken-and-egg problems with customizing setup.py. When I first read the specification of build backends in PEP-518 my immediate thought was "now we can build ament_python _for real_. @cottsay and I have sketched this out as
* Pick an existing python build backend, e.g. flit
* Wrap flit's build with tools that use catkin-pkg to read package information from package.xml
* Release that as ament_python allow it to be used for Python-only packages, allowing it to be used for both "native" ROS tooling and "native" Python tooling++.

++ Dependency "resolution" in the Python tooling is somewhat nebulous here. In the ROS world we can implicitly punt dependency resolution to rosdep ahead of time. I don't know if it would be worth, in a Python context, adding an additional ament_python dependency on rosdep and using it to resolve dependencies live during package configuration/build, but it is at least possible.

We've been stuck with rolling out [colcon-python-project](https://github.com/colcon/colcon-python-project/tree/devel) because the symlink install specification in PEP660 does not specify how to handle non-python source files, which we rely on for launch files, ament_index resources, etc, and many of the current build tools just ignore these. But I think it's alright to start introducing a standards-based alternative (and even recommending it) with these trade-offs still in place.

I don't think setuptools is worth investing in. Especially as a way to help bridge people migrating from ROS 1, because at some point they'll need to migrate again from setuptools to a standards-based workflow. We would really welcome help on proving out the resurrection of ament_python as a PEP-compliant build backend that can be used in ROS packages (and work as well as it can for Python packages outside of ROS tooling and workflows). If that sounds like something you're interested in working on. Let's get you set up as an Infrastructure Committer and we can flesh out the sketch above into an actual working prototype.

peci1, 2025-04-03:
Thanks, I'll reply more thorouhly later. Just a quick one now

> Can you clarify where rosdep fits into the story you're hoping to enable? rosdep completely ignores setup.py and only examines package.xml to determine package dependencies. If a "no-op" / not [...]

This means the standard approach end-users should be following: 

```
cd src
vcs import ...
rosdep install --from-paths . --ignore-src
cd ..
colcon build
```
colcon depends on rosdep installing all `<buildtool_depend>` packages to succeed.