ros2/ros2 #1139: Failed to build  qt_gui_cpp

Sission, 2021-04-27:
## Bug report

**Required Info:**

- Operating System:
  - MacOS big sur
- Installation type:
  - Source build
- Version or commit hash:
  - Follow this tutorial https://docs.ros.org/en/foxy/Installation/macOS-Development-Setup.html

#### Steps to reproduce issue

```
colcon build --packages-select qt_gui_cpp
```

#### Actual behavior
```
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                         
CMake Warning at /Users/sission/ros2_foxy/install/python_qt_binding/share/python_qt_binding/cmake/sip_helper.cmake:27 (message):
  SIP binding generator NOT available.
Call Stack (most recent call first):
  src/qt_gui_cpp_sip/CMakeLists.txt:56 (include)

CMake Error at src/CMakeLists.txt:10 (message):
  No Python binding generator found.

---
Failed   <<< qt_gui_cpp [1.20s, exited with code 1]

Summary: 0 packages finished [2.33s]
  1 package failed: qt_gui_cpp
  1 package had stderr output: qt_gui_cpp
```

## Feature request

#### Feature description
I noticed that https://github.com/ros2/ros2/issues/623 has a similar issue. However, when I 

```
sission@Shichens-MacBook-Pro ros2_foxy % type -a python3
python3 is /usr/local/opt/python@3.8/bin/python3
python3 is /usr/local/bin/python3
python3 is /usr/bin/python3
sission@Shichens-MacBook-Pro ros2_foxy % type -a sip
sip not found
sission@Shichens-MacBook-Pro ros2_foxy % brew install sip
Warning: Treating sip as a formula. For the cask, use homebrew/cask/sip
Warning: sip 6.0.3_1 is already installed and up-to-date.
To reinstall 6.0.3_1, run:
  brew reinstall sip

sission@Shichens-MacBook-Pro ros2_foxy % brew install --cask sip
==> Downloading https://sipapp.io/updates/v2/sip-2.5.3.zip
######################################################################## 100.0%
==> Installing Cask sip
==> Moving App 'Sip.app' to '/Applications/Sip.app'
🍺  sip was successfully installed!
sission@Shichens-MacBook-Pro ros2_foxy % colcon build --packages-select qt_gui_cpp               

Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                             
Traceback (most recent call last):
  File "/Users/sission/ros2_foxy/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 8, in <module>
    from PyQt5 import QtCore
ModuleNotFoundError: No module named 'PyQt5'
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp [4.58s, exited with code 2]

sission@Shichens-MacBook-Pro ros2_foxy % pip3 install pyqt5
Collecting pyqt5
  Downloading PyQt5-5.15.4-cp36.cp37.cp38.cp39-abi3-macosx_10_13_intel.whl (7.0 MB)
     |████████████████████████████████| 7.0 MB 4.9 MB/s 
Collecting PyQt5-sip<13,>=12.8
  Downloading PyQt5_sip-12.8.1-cp38-cp38-macosx_10_9_x86_64.whl (63 kB)
     |████████████████████████████████| 63 kB 7.4 MB/s 
Collecting PyQt5-Qt5>=5.15
  Downloading PyQt5_Qt5-5.15.2-py3-none-macosx_10_13_intel.whl (40.5 MB)
     |████████████████████████████████| 40.5 MB 26.8 MB/s 
Installing collected packages: PyQt5-sip, PyQt5-Qt5, pyqt5
Successfully installed PyQt5-Qt5-5.15.2 PyQt5-sip-12.8.1 pyqt5-5.15.4
sission@Shichens-MacBook-Pro ros2_foxy % colcon build --packages-select qt_gui_cpp

Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp                             
Traceback (most recent call last):
  File "/Users/sission/ros2_foxy/install/python_qt_binding/share/python_qt_binding/cmake/sip_configure.py", line 9, in <module>
    import sipconfig
ModuleNotFoundError: No module named 'sipconfig'
make[2]: *** [sip/qt_gui_cpp_sip/Makefile] Error 1
make[1]: *** [src/qt_gui_cpp_sip/CMakeFiles/libqt_gui_cpp_sip.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< qt_gui_cpp [0.75s, exited with code 2]

Summary: 0 packages finished [1.97s]
  1 package failed: qt_gui_cpp
  1 package had stderr output: qt_gui_cpp

Summary: 0 packages finished [6.03s]
  1 package failed: qt_gui_cpp
  1 package had stderr output: qt_gui_cpp

```

#### Implementation considerations

kristoferB, 2021-05-06:
This is the same problem as ros-visualization/python_qt_binding#103. 

You can try the updated instructions https://docs.ros.org/en/rolling/Installation/macOS-Development-Setup.html, where you colcon build --symlink-install --packages-skip-by-dep python_qt_binding, instead.

clalancette, 2021-05-06:
Yeah, I'm going to close this as a duplicate of https://github.com/ros-visualization/python_qt_binding/issues/103

closed 2021-05-06 by clalancette

mdx8, 2022-01-09:
colcon build --symlink-install --packages-ignore qt_gui_cpp rqt_gui_cpp