ros2/ros2 #1144: colcon build error: No 'rosidl_typesupport_c' found  [Ubuntu 20.04 ROS2 Foxy]

newcanopies, 2021-04-30:
## Bug report

**Required Info:**

- Ubuntu 20.04
- binary
- ?
- DDS implementation:
  - middleware name    : rmw_fastrtps_cpp
- Client library (if applicable):
  - ?

#### Steps to reproduce issue

building control_msgs, on Ubuntu 20.04 ROS2 Foxy, server glx vendor string: SGI

~/ws_ROS2$ `colcon build --symlink-install --packages-select control_msgs`
```
Starting >>> control_msgs
--- stderr: control_msgs                         
CMake Error at /opt/ros/foxy/share/rosidl_typesupport_c/cmake/get_used_typesupports.cmake:35 (message):
  No 'rosidl_typesupport_c' found
Call Stack (most recent call first):
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_c-extras.cmake:8 (get_used_typesupports)
  /opt/ros/foxy/share/rosidl_typesupport_c/cmake/rosidl_typesupport_cConfig.cmake:41 (include)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< control_msgs [3.05s, exited with code 1]

Summary: 0 packages finished [3.39s]
  1 package failed: control_msgs
  1 package had stderr output: control_msgs
```

`rm -rf build install log` is done at every new colcon build attempt (the contents of `~/ws_ROS2/build` shown for colcon sequence)
[image]

#### Expected behavior

#### Actual behavior

#### Additional information

what should the `COLCON/AMENT/CMAKE_PREFIX_PATH` be ?

$ env
```
SHELL=/bin/bash
ROS_VERSION=2
GAZEBO_MASTER_URI=http://localhost:11345
COLORTERM=truecolor
NVM_INC=/home/user/.nvm/versions/node/v12.20.1/include/node
TERM_PROGRAM_VERSION=1.55.2
WSL_DISTRO_NAME=Ubuntu-20.04
ROS_PYTHON_VERSION=3
NAME=machine
PWD=/home/user
LOGNAME=user
VSCODE_GIT_ASKPASS_NODE=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/node
HOME=/home/user
GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:
LANG=en_US.UTF-8
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
LS_COLORS= . . .
AMENT_PREFIX_PATH=/opt/ros/foxy
GIT_ASKPASS=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/extensions/git/dist/askpass.sh
GAZEBO_RESOURCE_PATH=/usr/share/gazebo-11:/usr/share/gazebo-11:
NVM_DIR=/home/user/.nvm
LESSCLOSE=/usr/bin/lesspipe %s %s
PYTHONPATH=/opt/ros/foxy/lib/python3.8/site-packages
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=user
VSCODE_GIT_IPC_HANDLE=/tmp/vscode-git-b1d11a6d9d.sock
DISPLAY=xxx.xx.xx.1:0.0
SHLVL=1
NVM_CD_FLAGS=
LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins
XDG_RUNTIME_DIR=/run/user/1000
ROS_LOCALHOST_ONLY=0
WSLENV=VSCODE_WSL_EXT_LOCATION/up
VSCODE_GIT_ASKPASS_MAIN=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/extensions/git/dist/askpass-main.js
GAZEBO_MODEL_PATH=/usr/share/gazebo-11/models/mise_staark_model_description/models:/usr/share/gazebo-11/models:
XDG_DATA_DIRS=/usr/local/share:/usr/share:/var/lib/snapd/desktop
PATH=/home/user/.vscode-server/bin/3c4e3df9e89829dce27b7b5c24508306b151f30d/bin:/home/user/.local/bin:/opt/ros/foxy/bin:/home/user/.nvm/versions/node/v12.20.1/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Python27/:/mnt/c/Python27/Scripts:
 . . . 
NVM_BIN=/home/user/.nvm/versions/node/v12.20.1/bin
HOSTTYPE=x86_64
ROS_DISTRO=foxy
OGRE_RESOURCE_PATH=/usr/lib/x86_64-linux-gnu/OGRE-1.9.0
TERM_PROGRAM=vscode
VSCODE_IPC_HOOK_CLI=/tmp/vscode-ipc-eca0215f-7319-4401-933b-61239177a1d8.sock
_=/usr/bin/env
```

newcanopies, 2021-04-30:
`sudo apt install ros-foxy-rosidl-typesupport-c `

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-foxy-rosidl-typesupport-c is already the newest version (1.0.2-1focal.20210415.021333).
ros-foxy-rosidl-typesupport-c set to manually installed.
The following packages were automatically installed and are no longer required:
  python3-xkit ubuntu-drivers-common
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 8 not upgraded.
```

nuclearsandwich, 2021-05-20:
@newcanopies I'm not able to reproduce this problem when building control_msgs in a `ros:foxy-ros-base` container which uses our binary packages running on Ubuntu 20.04.

Did you set up your environment by sourcing the setup script at `/opt/ros/foxy/setup.bash` (assuming bash is your interactive shell)?
Have you made any other manual changes to the environment after sourcing that setup script and if so do you get the same problem when in a "clean" ROS 2 shell?

newcanopies, 2021-05-27:
thanks @nuclearsandwich 
`control_msgs` does build successfully in a fresh workspace dir with no other packages, 
so the build fail must be some sort of conflict with one of other ROS2 package pictured above in the master workspace `ws_ROS2`

newcanopies, 2021-05-27:
thanks @nuclearsandwich 
`control_msgs` does build successfully in a fresh workspace dir with no other packages, 
so the build fail could be arising from some sort of conflict with one of other core ROS2 packages pictured above.

closed 2021-05-27 by newcanopies

Marshall-Wen-Amba, 2024-02-02:
Hi @newcanopies 

Nice to meet you~~
Recently, I also met this issue described in below link. You mentioned that it might be "so the build fail could be arising from some sort of conflict with one of other core ROS2 packages pictured above". Do you have any further idea on this? Like where the conflict comes from.

https://github.com/ros/meta-ros/issues/1099