ros2/ros2 #1148: macOS build from source in ROS2 Galactic

kmamykin, 2021-06-02:
This is a meta issue to track the progress on various issues related to building ROS2 Galactic on macOS.

Background discourse discussion:
https://discourse.ros.org/t/macos-support-in-ros-2-galactic-and-beyond/17891/23

#### Steps to reproduce issue

Following documentation steps to build from source on macOS

#### Expected behavior

Documentation steps should result in a working installation of all tools up to `desktop` variant, including `rviz`.

#### Actual behavior

Various issue, specifically related to Big Sur and/or Apple silicon (M1 cpu) break the build.

## Issues and workarounds

- [ ] Issue with SIP, Qt@5, and PyQt5 (https://github.com/ros-visualization/python_qt_binding/issues/103)

More to add here

YVbakker, 2021-06-02:
Awesome! I've just started an experimental build of ros2 galactic on M1, and these are my findings thus far:
Perhaps I should make different issues for these but I haven't had the time yet.

osrf_testing_tools_cpp fails because vendor/bombela/backward-cpp is not up to date and m1 compatibility was only patched recently.
Manually cloning https://github.com/bombela/backward-cpp into /vendor/bombela fixes this issue.
---
yaml_cpp_vendor fails because https://github.com/jbeder/yaml-cpp/archive/0f9a586ca1dc29c2ecb8dd715a315b93e3f40f79.zip cannot be downloaded from github (error 500)
Changing the external url (and other parameters) to this, works:
    yaml-cpp-master
    URL https://github.com/jbeder/yaml-cpp/archive/refs/heads/master.zip
    URL_MD5 bf01ea6531690b4028f07ea3d2b2e8c1
---
rviz_ogre_vendor fails because libraries aren’t built for arm64 yet.
this can be fixed by changing a couple lines in the CMakeLists.txt (see diff) in package
basically change x86_64 to arm64, disable PCH and disable ASSIMP plugin
---
--- stderr: rviz_rendering                                                                                            
/Users/yvbakker/ros2_galactic/install/rviz_ogre_vendor/opt/rviz_ogre_vendor/include/OGRE/RenderSystems/GL/OgreGLPrerequisites.h:67:13: error: 'GL/glew.h' file not found with <angled> include; use "quotes" instead
#   include <GL/glew.h>
---
rmw_fastrtps_shared_cpp fails to build because:
tests fail to build:
specifically this test:
src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/test/test_dds_attributes_to_rmw_qos.cpp fail
error: use of undeclared identifier 'RMW_DURATION_INFINITE'
this identifier comes from rmw/time.h
which is included in rmw/types.h
which is included in rmw/rmw.h
which is included in rmw_fastrtps_shared_cpp/qos.hpp
which is included in test_dds_attributes_to_rmw_qos.cpp

for some reason the GTEST macro cannot resolve this constant

temporary solution: disable tests by commenting them out in CMakeLists.txt

For this last one there is an issue on: https://github.com/ros2/rmw_fastrtps/issues/538

christophebedard, 2021-06-02:
> osrf_testing_tools_cpp fails because vendor/bombela/backward-cpp is not up to date and m1 compatibility was only patched recently.
> Manually cloning [bombela/backward-cpp](https://github.com/bombela/backward-cpp) into /vendor/bombela fixes this issue.

Yeah, https://github.com/bombela/backward-cpp was patched, but there's an open PR for osrf_testing_tools_cpp: https://github.com/osrf/osrf_testing_tools_cpp/pull/64 (just to link to it from this meta issue).

kmamykin, 2021-06-03:
@YVbakker awesome! Thanks for adding the issues you encountered. I tried to consolidate them at the top. Some of them I haven't seen in my build.
Could you track down and may be even open new issues to the corresponding upstream projects?

traversaro, 2021-06-03:
If you are interested, there are some macOS-related changes in the patches that we have (for foxy, for now) in the RoboStack project: https://github.com/RoboStack/ros-foxy/tree/master/patch . However, not sure how many of them are relevant when installing dependencies via Homebrew as opposed to install them via conda-forge.

zmk5, 2021-06-13:
Anything I can do to help? I have Galactic installed and working on macOS Big Sur.

kmamykin, 2021-06-13:
@zmk5 
There is a PR I opened that addresses several rviz related issues by upgrading Ogre (https://github.com/ros2/rviz/pull/700). What's left there is to get rid of deprecation warnings. I am a bit stuck there, made the code changes but now there is an assertion thrown I can't figure out where. Feel free to take a stub at that.

Lookin into how to fix `-packages-skip-by-dep python_qt_binding` in the build would also be worthwhile.

Otherwise pick anything you feel you can tackle. Post a comment here if there is an update to the meta tracking issue.

erickisos, 2022-02-12:
Just in case, mimick_vendor is giving some errors on the compilation stage because we have a reference to a previous commit in which macOS M1 was not supported, the solution is to replace in the CMakeLists.txt the `mimick_version` to at least, this commit: https://github.com/ros2/Mimick/commit/4c742d61d4f47a58492c1afbd825fad1c9e05a09

mengchaoheng, 2023-04-22:
run 
```
colcon build \
  --symlink-install \
  --merge-install \
  --event-handlers console_cohesion+ console_package_list+ \
  --packages-skip-by-dep python_qt_binding \
  --cmake-args \
    --no-warn-unused-cli \
    -DBUILD_TESTING=OFF \
    -DINSTALL_EXAMPLES=ON \
    -DCMAKE_OSX_SYSROOT=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk \
    -DCMAKE_OSX_ARCHITECTURES="arm64" \
    -DOpenCV_DIR=/usr/local/lib/cmake/opencv4 \
    -DCMAKE_INCLUDE_PATH="/opt/homebrew/include/" \
    -DCMAKE_LIBRARY_PATH="/opt/homebrew/lib/" \
    -DCMAKE_PREFIX_PATH=$(brew --prefix):$(brew --prefix qt@5) \
    -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl
```
output is 
```
[ 74%] Generating Python code for ROS interfaces
[ 74%] Built target unique_identifier_msgs__py
[ 77%] Building C object CMakeFiles/unique_identifier_msgs__python.dir/rosidl_generator_py/unique_identifier_msgs/msg/_uuid_s.c.o
[ 80%] Linking C shared library rosidl_generator_py/unique_identifier_msgs/libunique_identifier_msgs__python.dylib
ld: warning: ignoring file /usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/config-3.9-darwin/libpython3.9.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_PyImport_ImportModule", referenced from:
      _unique_identifier_msgs__msg__uuid__convert_to_py in _uuid_s.c.o
  "_PyObject_CallObject", referenced from:
      _unique_identifier_msgs__msg__uuid__convert_to_py in _uuid_s.c.o
  "_PyObject_GetAttrString", referenced from:
      _unique_identifier_msgs__msg__uuid__convert_from_py in _uuid_s.c.o
      _unique_identifier_msgs__msg__uuid__convert_to_py in _uuid_s.c.o
  "__Py_Dealloc", referenced from:
      __Py_DECREF in _uuid_s.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rosidl_generator_py/unique_identifier_msgs/libunique_identifier_msgs__python.dylib] Error 1
make[1]: *** [CMakeFiles/unique_identifier_msgs__python.dir/all] Error 2
make: *** [all] Error 2
---
--- stderr: unique_identifier_msgs
ld: warning: ignoring file /usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/lib/python3.9/config-3.9-darwin/libpython3.9.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_PyImport_ImportModule", referenced from:
      _unique_identifier_msgs__msg__uuid__convert_to_py in _uuid_s.c.o
  "_PyObject_CallObject", referenced from:
      _unique_identifier_msgs__msg__uuid__convert_to_py in _uuid_s.c.o
  "_PyObject_GetAttrString", referenced from:
      _unique_identifier_msgs__msg__uuid__convert_from_py in _uuid_s.c.o
      _unique_identifier_msgs__msg__uuid__convert_to_py in _uuid_s.c.o
  "__Py_Dealloc", referenced from:
      __Py_DECREF in _uuid_s.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rosidl_generator_py/unique_identifier_msgs/libunique_identifier_msgs__python.dylib] Error 1
make[1]: *** [CMakeFiles/unique_identifier_msgs__python.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< unique_identifier_msgs [9.87s, exited with code 2]
Aborted  <<< builtin_interfaces [11.3s]                  
Aborted  <<< std_srvs [13.1s]                            
Aborted  <<< rmw_dds_common [13.4s]                      
Aborted  <<< lifecycle_msgs [21.6s]                       
Aborted  <<< rviz_ogre_vendor [6min 34s]                                             

Summary: 142 packages finished [6min 48s]
  1 package failed: unique_identifier_msgs
  5 packages aborted: builtin_interfaces lifecycle_msgs rmw_dds_common rviz_ogre_vendor std_srvs
  48 packages had stderr output: ament_clang_format ament_clang_tidy ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pclint ament_pep257 ament_pycodestyle ament_pyflakes ament_uncrustify ament_xmllint builtin_interfaces cyclonedds domain_coordinator fastrtps foonathan_memory_vendor google_benchmark_vendor iceoryx_posh iceoryx_utils launch launch_testing launch_xml launch_yaml libyaml_vendor lifecycle_msgs mimick_vendor osrf_pycommon pybind11_vendor rcl_logging_log4cxx rcpputils rmw_dds_common rosidl_cli rpyutils rti_connext_dds_cmake_module rviz_ogre_vendor shared_queues_vendor std_srvs tracetools_read tracetools_trace uncrustify_vendor unique_identifier_msgs yaml_cpp_vendor
  157 packages not processed
[407.789s] ERROR:colcon.colcon_notification.desktop_notification.terminal_notifier:Could not find the colcon-terminal-notifier.app in the install prefix '/Users/mch/test_ros2/.venv'
```

mengchaoheng, 2023-04-23:
@YVbakker I made the changes according to the instructions, why rviz_ogre_vendor still reports an error
```
[ 38%] Linking CXX shared library ../lib/macosx/libOgreMain.dylib
Undefined symbols for architecture arm64:
  "_zzip_dir_close", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::~ZipArchive() in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipArchive::unload() in OgreZip.cpp.o
  "_zzip_dir_open_ext_io", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::load() in OgreZip.cpp.o
  "_zzip_dir_read", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::load() in OgreZip.cpp.o
  "_zzip_dir_stat", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_dirhandle", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_error", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_file_close", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::~ZipDataStream() in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::~ZipDataStream() in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::close() in OgreZip.cpp.o
  "_zzip_file_open", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_file_read", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_init_io", referenced from:
      Ogre::ZipArchiveFactory::createInstance(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) in OgreZip.cpp.o
  "_zzip_seek", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::skip(long) in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::seek(unsigned long) in OgreZip.cpp.o
  "_zzip_strerror_of", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_tell", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::tell() const in OgreZip.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [lib/macosx/libOgreMain.1.12.10.dylib] Error 1
make[4]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-v1.12.10-prefix/src/ogre-v1.12.10-stamp/ogre-v1.12.10-build] Error 2
make[1]: *** [CMakeFiles/ogre-v1.12.10.dir/all] Error 2
make: *** [all] Error 2
---
--- stderr: rviz_ogre_vendor
2023-04-23 09:20:46.078 xcodebuild[14144:3654108] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionSentinelHostApplications for extension Xcode.DebuggerFoundation.AppExtensionHosts.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
2023-04-23 09:20:46.078 xcodebuild[14144:3654108] Requested but did not find extension point with identifier Xcode.IDEKit.ExtensionPointIdentifierToBundleIdentifier for extension Xcode.DebuggerFoundation.AppExtensionToBundleIdentifierMap.watchOS of plug-in com.apple.dt.IDEWatchSupportCore
Undefined symbols for architecture arm64:
  "_zzip_dir_close", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::~ZipArchive() in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipArchive::unload() in OgreZip.cpp.o
  "_zzip_dir_open_ext_io", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::load() in OgreZip.cpp.o
  "_zzip_dir_read", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::load() in OgreZip.cpp.o
  "_zzip_dir_stat", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_dirhandle", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
  "_zzip_error", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_file_close", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::~ZipDataStream() in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::~ZipDataStream() in OgreZip.cpp.o
      Ogre::(anonymous namespace)::ZipDataStream::close() in OgreZip.cpp.o
  "_zzip_file_open", referenced from:
      Ogre::(anonymous namespace)::ZipArchive::open(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) const in OgreZip.cpp.o
  "_zzip_file_read", referenced from:
      Ogre::(anonymous namespace)::ZipDataStream::read(void*, unsigned long) in OgreZip.cpp.o
[... 1 lines cut ...]
[... cut ...] t_c rosidl_typesupport_cpp rosidl_typesupport_fastrtps_c rosidl_typesupport_introspection_cpp rpyutils rviz_ogre_vendor sensor_msgs_py shared_queues_vendor sros2 test_launch_ros tf2_ros_py tf2_tools topic_monitor tracetools_launch tracetools_read tracetools_trace urdfdom yaml_cpp_vendor
  6 packages not processed
[301.372s] ERROR:colcon.colcon_notification.desktop_notification.terminal_notifier:Could not find the colcon-terminal-notifier.app in the install prefix '/Users/mch/test_ros2/.venv'
```

pfavr2, 2023-08-19:
I’ve managed to build ROS 2 rolling on a M2 MacBook. This required a few changes to CMakeList.txt files and some other workarounds. I would like to submit this and hopefully help get M1 Macs as supported target for ROS 2.

Any hints on how to get started contributing?

Best regards,

Peter

clalancette, 2023-08-21:
> Any hints on how to get started contributing?

Please start with http://docs.ros.org/en/rolling/The-ROS2-Project/Contributing.html , which covers the steps to get started contributing.

pfavr2, 2023-09-03:
Here is a script which successfully builds ros2_rolling on Mac M1:

https://github.com/pfavr2/install_ros2_rolling_on_mac_m1

XCode and HomeBrew should be installed before install.sh is run.

It builds almost everything. Rviz2 is working also. No need to change SIP (system integrity protection).

Summary: 333 packages finished [15min 16s] (on Macbook Pro with M2 Pro cpu and 32GB RAM).

mengchaoheng, 2023-09-03:
> Here is a script which successfully builds ros2_rolling on Mac M1:
> 
> 
> 
> https://github.com/pfavr2/install_ros2_rolling_on_mac_m1
> 
> 
> 
how to use? I don't see any guide for installing ros on the readme.md

pfavr2, 2023-09-03:
> @pfavr2 how to use? I don't see any guide for installing ros on the readme.md

Just run install.sh script in a terminal:

. ./install.sh

clalancette, 2024-09-11:
Since this was filed against Galactic, which is long out of support, I'm going to close this out.  If you are still having issues with Humble or newer, please feel free to open a new bug.

closed 2024-09-11 by clalancette