ros2/ros2 #1149: ROS 2 Humble Release Checklist

audrow, 2021-06-03:
Pre-kickoff (during the last stage of previous release)
---
- [x] Choose a ROS boss: @audrow
- [x] Choose a distribution name: [Humble Hawksbill](https://discourse.ros.org/t/ros-2-h-name-brainstorming/20065)

Kickoff (start in January)
---
- [ ] Commission a logo
- [ ] Commission a turtlesim icon
- [ ] Add to [REP-2000](https://www.ros.org/reps/rep-2000.html) with proposed platforms and minimum requirements, request feedback from community via discourse
  Example from Foxy: ros-infrastructure/rep#217
- [ ] Create the detailed release plan

### Earlier
- [ ] Update [ROS 2 documentation](https://github.com/ros2/ros2_documentation/) with expected release date, release notes
  Example from Foxy: ros2/ros2_documentation#445
- [ ] Update MSVC 2021

Mid-development cycle (Alpha, Beta?)
---
- [ ] Build ROS 2 on 22.04; fix any issues
- [ ] Cut Rolling over to 22.04
- [ ] Update Windows packages
  e.g., bump Python, bump OpenCV, pinned pip packages, Visual Studio Code, etc...
  - [ ] Bump [REP-2000](https://www.ros.org/reps/rep-2000.html)
- [ ] Encourage regular package releases after API changes
  This helps keep the debian packaging green and us alert to packaging regressions
- [ ] Upgrade all vendored packages to the "latest" (whatever makes sense)
- [ ] Do first round of testing

Late development cycle (Migration from Rolling)
---
- Regularly release new changes
- [ ] Announce the API and feature freeze on Discourse
- [ ] Release all packages into Rolling
- [ ] Add the new distro to rosdistro: Example ros/rosdistro#29211
- [ ] Create a Google Group for the new distro: https://groups.google.com/forum/#!forum/ros2-buildfarm-galactic
- [ ] Add new rosdistro as an option to target on [ci.ros2.org](https://ci.ros2.org/): ros2/ci#571
- [ ] If this is a new Ubuntu platform:
  - [ ] Setup new apt repos
  - [ ] Update CI infrastructure to support new platforms
    Example from Foxy: ros2/ci#376 and ros2/ci#414
  - [ ] Create Docker images for the new platforms in [REP-2000](https://www.ros.org/reps/rep-2000.html) for all targets
    Example from Foxy: osrf/multiarch-docker-image-generation#32
  - [ ] Add newly supported platforms to list of platforms for each Python package (in `stdeb.cfg`)
    Use [`ros_release_python`](https://github.com/ros-infrastructure/ros_release_python)
    Goal is to make the currently released versions available in the new distros using the `--include` option
    In order of dependency:
    - [ ] colcon
    - [ ] catkin_pkg
      Example from Foxy: ros-infrastructure/catkin_pkg#271
    - [ ] rospkg
      Example from Foxy: ros-infrastructure/rospkg#184
    - [ ] rosdistro
      Example from Foxy: ros-infrastructure/rosdistro#148
    - [ ] rosdep
      Example from Foxy: ros-infrastructure/rosdep#734
    - [ ] bloom
      Example from Foxy: ros-infrastructure/bloom#558
    - [ ] rosinstall_generator
      Example from Foxy: ros-infrastructure/rosinstall_generator#62
    - [ ] catkin_tools
      Example from Foxy: catkin/catkin_tools#564
    - [ ] vcstool
      Example from Foxy: dirk-thomas/vcstool#112
- [ ] Run migration from Rolling to Humble
  - [ ] Script to take Rolling packages at commit and release into new distro: ros/rosdistro#29221 
    This is a one time deal resulting in two separate distributions with no automated syncs between them
- [ ] Create distro branches
  - [ ] After rosdistro freeze, ensure each repository has a branch for the new distro
    Suggest creating a workspace with a custom repos file, pruning third-party packages and others that don’t need a new branch. Then run:
    ```
    vcs custom --git --args checkout -b ROSDISTRO
    vcs custom --git --args push origin ROSDISTRO
    ```
  - [ ] Create `ros2.repos` files for the new distro, one for the release and one for development
    This command should get you part way there for the release repos file:
    ```
    vcs export --exact-with-tags
    ```
  - [ ] Ensure rosdistro points to the new branches: ros/rosdistro#29267
    This script might be useful (no warranty): https://gist.github.com/jacobperron/ba6cc225aefdcee822261b434620766f
  - [ ] Update buildfarm CI config to use new distro repos file: ros2/ros_buildfarm_config#157
  - [ ] Update ros2_documentation install pages:
    Example from Foxy ros2/ros2_documentation#752
  - [ ] Sync release devel branches with branched versions
- [ ] Enable jobs on [build.ros2.org](http://build.ros2.org/) (after branch)
  - [ ] Add new distribution to the buildfarm configuration: ros2/ros_buildfarm_config#155
    - [ ] Copy files from previous distro and do some renaming (separate commits for easy review)
    - [ ] Update job priorities (e.g. where they differ between previous two distros)
      Tip: run ./list_priorities.py in root of repo to view a priority list
    - [ ] Reduce package count for sync
  - [ ] Deploy the jobs after configuration changes are merged
    - [ ] On your local machine, install pip dependency jenkinsapi: pip3 install jenkinsapi
    - [ ] Generate a buildfarm token for [build.ros2.org](http://build.ros2.org/) (if you don't already have one):
      * Go to [build.ros2.org](http://build.ros2.org/)
      * Login and click on your name (top-right)
      * Under "Configure -> API Token", add a new token
      * Copy the token to add to the file in the next step
      * Create a file ~/.buildfarm/jenkins.ini with the following content (if you don't already have it):
        ```
        [DEFAULT]
        username=admin
        password=changeme
        [http://build.ros2.org]
        username=YOUR_USERNAME
        # Jenkins API token
        password=YOUR_BUILDFARM_TOKEN
        ```
      * Replace `YOUR_USERNAME` with your [build.ros2.org](http://build.ros2.org/) login and replace `YOUR_BUILDFARM_TOKEN` with the token generated in the previous step.
    - [ ] Clone the source repo:
    ```
    git clone https://github.com/ros-infrastructure/ros_buildfarm.git
    ```
    - [ ] Change to the directory and point `PYTHONPATH` to it:
        ```
        cd ros_buildfarm
        export PYTHONPATH=${PYTHONPATH}:$(pwd)
        ```
    - [ ] Run the script without committing to check for errors (replace `ROSDISTRO` as appropriate):
        ```
        ./scripts/generate_all_jobs.py https://raw.githubusercontent.com/ros2/ros_buildfarm_config/ros2/index.yaml --ros-distro-names ROSDISTRO
        ```
    - [ ] If there are no errors, re-run with the --commit option (replace `ROSDISTRO` as appropriate):
        ```
        ./scripts/generate_all_jobs.py https://raw.githubusercontent.com/ros2/ros_buildfarm_config/ros2/index.yaml --ros-distro-names ROSDISTRO --commit
        ```
    - [ ] Go to [build.ros2.org](http://build.ros2.org/) and run the `Xci_reconfigure-jobs` job for your distro (replace X with the first letter of the distro name, e.g. [Fci_reconfigure-jobs](http://build.ros2.org/job/Fci_reconfigure-jobs/))
- [ ] Create matrix for tracking manual testing
  Example from Foxy: https://docs.google.com/spreadsheets/d/1EzDKMGoYopKuGWmzceXTVJ1zWqq1Wxw2jIGsh3CawAQ/edit
    Suggest starting after beta and doing it in rounds (testing/bug fixes for a week, then make new releases, repeat)
- [ ] Update ROS 2 documentation
  - [ ] List install instructions
    Example from Foxy: ros2/ros2_documentation#746
  - [ ] List distribution
    Example from Foxy: ros2/ros2_documentation#747
- [ ] Start executing the first test matrix (TBD)
- [ ] Update API documentation: ros2/docs.ros2.org#56
   Follow instructions here: https://github.com/ros2/docs.ros2.org/tree/doc_gen
- [ ] Add distro to ROS Index: ros-infrastructure/rosindex/pull/264
  - [ ] Trigger the Jenkins job: http://build.ros.org/job/doc_rosindex/
- Create Humble patch 1 board

Pre-release day
---
- [ ] Announce art on Discourse
- [ ] Launch Teespring campaign
- [ ] Get documentation PR ready
- [ ] Get release ros2/ros2 repos file ready
  - [ ] Build all packaging jobs
- [ ] Draft PR to update release status on ros/rosdistro
- [ ] Prepare the Discourse post
  - [ ] Get team feedback

On Release day
---
- [ ] Merge PR and make tag [ros2/ros2](https://github.com/ros2/ros2): ros2/ros2#1147
  - [ ] Create a release and upload the packaging job artifacts
- [ ] Sync debian packages to main
  - [ ] Trigger `Xrel_sync-packages-to-main` on [build.ros2.org](http://build.ros2.org) (replace X with the first letter of the distro name, e.g. [Grel_sync-packages-to-main](http://build.ros2.org/job/Grel_sync-packages-to-main/1/))
  - [ ] Verify that the packages were uploaded to OSU
- [ ] Merge ros/rosdistro PR
  - [ ] Make tag
- [ ] Update release status on rosdistro: ros/rosdistro#29635
- [ ] Announce on Discourse: https://discourse.ros.org/t/ros-2-galactic-geochelone-released
  - [ ] Close metaticket: https://github.com/ros2/ros2/issues/954
  - [ ] Close this checklist
  - [ ] Close out the project board

pablogs9, 2022-05-19:
Hello, is there an approx date for ROS 2 Humble Docker release?

Oasis-Guo, 2022-05-20:
hi, when we can use ros2/humble docker image at docker/hub?

audrow, 2022-05-20:
> Hello, is there an approx date for ROS 2 Humble Docker release?

@pablogs9, this PR adds the Humble Docker images: https://github.com/osrf/docker_images/pull/620. We're hoping to get it in soon after the Humble release on May 23rd.

krsche, 2022-05-23:
So excited :smiley:  
Switching from the testing install to the released one would be as simple as removing `-testing` from the apt sources I suppose (on ubuntu)?  
```
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2-testing/ubuntu jammy main
```
to
```
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu jammy main
```
And re-running `apt update && apt upgrade` of course

audrow, 2022-05-23:
That's right, here's the PR making this change: https://github.com/ros2/ros2_documentation/pull/2580/files.

We'll merge this PR in when everything else is ready.

audrow, 2022-05-23:
> Hello, is there an approx date for ROS 2 Humble Docker release?

https://github.com/osrf/docker_images/pull/622 has been closed. I think this means the images should build tonight, if all goes well.

clalancette, 2022-05-24:
Humble is released!  https://discourse.ros.org/t/ros-2-humble-hawksbill-released/25729/4

closed 2022-05-24 by clalancette

christophebedard, 2022-06-29:
@clalancette @audrow (not sure who's in charge of the Iron release) could the release checklist template be updated to include updating the ros.org website? The landing page was updated in https://github.com/ros-infrastructure/www.ros.org/pull/35, but the "Getting started" page hasn't been updated, see: https://github.com/ros-infrastructure/www.ros.org/pull/37.