ros2/ros2 #1150: Overlaying packages using CMake export targets can fail with merge install underlay

jacobperron, 2021-06-08:
## Bug report

**Required Info:**

- Operating System:
  - Any
- Installation type:
  - Any installation that is a merge install (e.g. built with `colcon build --merge-install`)
- Version or commit hash:
  - Any version

#### Steps to reproduce issue

I've created some mock packages that reproduce the issue. These packages and instructions for reproducing can be found in this repository: https://github.com/jacobperron/overlay_bug_reproduction 

#### Expected behavior

We are able to build a package that exists in an underlay ROS 2 installation and have other packages in the overlay workspace properly link against it.

#### Actual behavior

In some circumstances, subject to the dependency tree, the compiler is given the underlay include directory before the overlay include directories. If a package appears in both the overlay and underlay, then the underlay include directory will be used instead.

hexonxons, 2021-06-11:
Seems like problem is due to [ament_target_dependencies.cmake](https://github.com/ament/ament_cmake/blob/master/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake).

While debugging same problem with ros2/rosbag2#687:
Having `ament_target_dependencies` called like
```cmake
ament_target_dependencies(${PROJECT_NAME}
  rcpputils
  rosbag2_compression
  zstd)
```
this code executed([link](https://github.com/ament/ament_cmake/blob/3f992adb8b35f86246bb6d0cd05cfafb0afd086c/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake#L127-L143)):
```cmake
if(NOT ARG_INTERFACE)
  target_compile_definitions(${target}
    ${required_keyword} ${definitions})
  # the interface include dirs must be ordered
  set(interface_include_dirs)
  foreach(interface ${interfaces})
    get_target_property(_include_dirs ${interface} INTERFACE_INCLUDE_DIRECTORIES)
    if(_include_dirs)
      list_append_unique(interface_include_dirs ${_include_dirs})
    endif()
  endforeach()
  ament_include_directories_order(ordered_interface_include_dirs ${interface_include_dirs})
  # the interface include dirs are used privately to ensure proper order
  # and the interfaces cover the public case
  target_include_directories(${target} ${system_keyword}
    PRIVATE ${ordered_interface_include_dirs})
endif()
```
with `interfaces` set to list `rosbag2_compression;zstd_vendor;rcpputils`.

`INTERFACE_INCLUDE_DIRECTORIES` of these targets are populated with values:
```
/home/hexonxons/ros/ws/install/rosbag2_compression/include
/home/hexonxons/ros/ws/install/zstd_vendor/include
/opt/ros/foxy/include
```
accordingly. `/home/hexonxons/ros/ws/` is my local workspace.

`rosbag2_compression` depends on `rosbag2_cpp`, but there is no `/home/hexonxons/ros/ws/install/rosbag2_cpp/include` path in that list, cuz `INTERFACE_INCLUDE_DIRECTORIES` property is populated by `target_include_directories` command according to [documentation](https://cmake.org/cmake/help/latest/prop_tgt/INTERFACE_INCLUDE_DIRECTORIES.html)

Also, according to same documentation:
```
The INSTALL_INTERFACE of the INTERFACE_INCLUDE_DIRECTORIES is only suitable for specifying the required include directories for headers provided with the target itself, not those provided by the transitive dependencies listed in its INTERFACE_LINK_LIBRARIES target property. Those dependencies should themselves be targets that specify their own header locations in INTERFACE_INCLUDE_DIRECTORIES.
```

So fast idea for this to work we need to recursively get dependencies from `INTERFACE_LINK_LIBRARIES` property and populate `interface_include_dirs` with `INTERFACE_INCLUDE_DIRECTORIES` property of each dependency. This needs to be investigated deeper to find out if this is right fix

jacobperron, 2021-06-13:
@hexonxons Thanks for posting your investigation!

The relevant ament_cmake code was added in https://github.com/ament/ament_cmake/pull/260 to fix a similar issue related to overlays (https://github.com/ros2/geometry2/issues/268). (I had forgotten about it even though I reviewed it :sweat_smile:).

I think your assessment is correct. The current code only orders include directories for the declared dependencies, it is not considering transitive dependencies. Though, with my current understanding of CMake, I agree with [this comment](https://github.com/ament/ament_cmake/pull/260#issuecomment-637137703), and CMake should already be able to handle the include directories of the transitive dependencies for us.

I tried the applying the following change:
```diff
--- a/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake
+++ b/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake
@@ -127,19 +127,7 @@ function(ament_target_dependencies target)
     if(NOT ARG_INTERFACE)
       target_compile_definitions(${target}
         ${required_keyword} ${definitions})
-      # the interface include dirs must be ordered
-      set(interface_include_dirs)
-      foreach(interface ${interfaces})
-        get_target_property(_include_dirs ${interface} INTERFACE_INCLUDE_DIRECTORIES)
-        if(_include_dirs)
-          list_append_unique(interface_include_dirs ${_include_dirs})
-        endif()
-      endforeach()
-      ament_include_directories_order(ordered_interface_include_dirs ${interface_include_dirs})
-      # the interface include dirs are used privately to ensure proper order
-      # and the interfaces cover the public case
-      target_include_directories(${target} ${system_keyword}
-        PRIVATE ${ordered_interface_include_dirs})
+      target_link_libraries(${target} ${interfaces})
```

This gives us control over the order that include directories are passed to the compiler. For example, if I make the following change to my [reproduction](https://github.com/jacobperron/overlay_bug_reproduction), then I can successfully compile the packages:

```diff
--- a/overlay/src/package_d/CMakeLists.txt
+++ b/overlay/src/package_d/CMakeLists.txt
@@ -13,10 +13,10 @@ add_library(${PROJECT_NAME}
   src/package_d.cpp
 )
 ament_target_dependencies(${PROJECT_NAME}
-  package_b
   # Targeting package_c causese underlay path to appear before overlay
   # The underlay contains an older version of package_a that package_b is not expecting
   package_c
+  package_b
```

I'm not sure if this is a desirable solution since it makes `ament_target_dependencies` sensitive to the order of it's arguments.

It seems like we need a way to detect if a dependency is in the overlay or not. Then we could order the target dependencies such that dependencies in the same overlay as the package being built appear first.

hexonxons, 2021-06-13:
Yep, cmake is handling include directories of the transitive dependencies.
In your case while compiling with `colcon build --ament-cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON` and having `ament_target_dependencies.cmake` changed:

Compile commands for `package_d`:

- `package_b` goes before `package_c`:

```
{
  "directory": "/home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/build/package_d",
  "command": "/usr/bin/c++   -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/underlay/install/include -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/install/package_c/include -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/install/package_a/include    -Wall -Wextra -Wpedantic -o CMakeFiles/package_d.dir/src/package_d.cpp.o -c /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/src/package_d/src/package_d.cpp",
  "file": "/home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/src/package_d/src/package_d.cpp"
}
```
- `package_b` goes after `package_c`:

```
{
  "directory": "/home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/build/package_d",
  "command": "/usr/bin/c++   -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/install/package_c/include -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/install/package_a/include -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/underlay/install/include    -Wall -Wextra -Wpedantic -o CMakeFiles/package_d.dir/src/package_d.cpp.o -c /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/src/package_d/src/package_d.cpp",
  "file": "/home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/src/package_d/src/package_d.cpp"
}
```

So in first case `-isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/underlay/install/include` goes first cuz it is in `package_b` `INTERFACE_INCLUDE_DIRECTORIES` property and you links it first.
Next goes `-isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/install/package_c/include -isystem /home/hexonxons/ros/ws/src/overlay_bug_reproduction/overlay/install/package_a/include`, cuz it is in `INTERFACE_INCLUDE_DIRECTORIES` property of `project_c`:
```cmake
ament_target_dependencies(${PROJECT_NAME}
  package_a
)
target_include_directories(${PROJECT_NAME}
  PUBLIC
  $<INSTALL_INTERFACE:include>
  $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
)
```

How do I suggested to fix:

```
if(NOT ARG_INTERFACE)
  target_compile_definitions(${target}
    ${required_keyword} ${definitions})
  # the interface include dirs must be ordered
  set(interface_include_dirs)
  set(interface_libraries)
  foreach(interface ${interfaces})
     set(current_interface_include_dirs)
     set(current_interface_libraries)
     ament_get_recursive_properties(current_interface_include_dirs current_interface_libraries ${interface})
     message(WARNING "interface_include_dirs for target ${interface}: ${current_interface_include_dirs}")
     message(WARNING "interface_libraries for target ${interface}: ${current_interface_libraries}")
     list_append_unique(interface_include_dirs "${current_interface_include_dirs}")
     list_append_unique(interface_libraries "${current_interface_libraries}")
  endforeach()
  message(WARNING "All interface_include_dirs: ${interface_include_dirs}")
  message(WARNING "All interface_libraries: ${interface_libraries}")
  ament_include_directories_order(ordered_interface_include_dirs ${interface_include_dirs})
  # the interface include dirs are used privately to ensure proper order
  # and the interfaces cover the public case
  target_include_directories(${target} ${system_keyword}
    PRIVATE ${ordered_interface_include_dirs})
endif()
```

You could try to build your packages and see debugging warning messages :)

hexonxons, 2021-06-13:
But still that is very strange way of doing cmake job. In non-ros cmake projects if you have `package_a` depending on `package_b` depending on `package_c` and want to
* Build `package_b`
* export CMAKE_PREFIX_PATH to `package_b`
* Build `package_a`
* export CMAKE_PREFIX_PATH to `package_a`
* Build `package_c`
So you will get `package_b` linked to system wide installed `package_a` and `package_d` linked to system wide installed `package_a`. If you also export `LD_LIBRARY_PATH` to local installed `package_a` you will get `package_d` build with system-wide installed `package_a` headers, but in runtime it will load shared lib of `package_a` :)

So I prefer to build full ROS repo in case of building overlay workspaces. In this case `target_link_libraries(${target} ${interfaces})
` might be sufficient

jacobperron, 2021-06-13:
I pushed an experimental patch that relies on `target_link_libraries` (https://github.com/ament/ament_cmake/commit/a284c007f375d59c71cb261c6ce7e6b253d0d4be), but it has a brittle heuristic for checking if targets are from the same overlay or not. We could instead do something based on the AMENT_PREFIX_PATH like `ament_include_directories_order` does, which might be a bit better. But I'm still not convinced it is much better than your proposed patch. In general, you are are right that building all packages is safer than attempting an overlay.

hexonxons, 2021-06-13:
@jacobperron Could we just use `target_link_libraries`? Overlay should work properly if you are building all packages in repo without skipping(e.g don't skip any package from rosbag2 repo if building rosbag2 repo in overlay) and maintainers declare package dependencies properly.

Otherwise your case is a quite "pro" case when you are trying to build not one, but several repos of ros packages while skipping dependencies in between (package_b).

I will try to check if `target_link_libraries` fixes problem for rosbag2 tomorrow

jacobperron, 2021-06-13:
> Could we just use `target_link_libraries`?

IIUC, this is what was happening before https://github.com/ament/ament_cmake/pull/260 (ie. just try reverting https://github.com/ament/ament_cmake/pull/260 and see what happens). I think we still need to worry about the order of targets passed to `target_link_libraries`. Currently `ament_target_dependencies` is trying to handle the ordering for the user (but failing in some cases). Alternatively, we could say that it's the users responsibility to order dependencies passed to `ament_target_dependencies` "correctly", but I don't think the order is obvious and depends workspace layout.

sloretz, 2021-09-29:
This ticket is about sorting include directories based on the order workspaces are overlayed, but that's not the only problem with overriding packages.
Any non-leaf package must only be overriden by an API and ABI compatible version or the runtime behavior is undefined.

Include directory search order matters anytime headers with the same name can be found in different directories.
Usually this is only a problem when overriding a package in a merged underlay workspace.
The underlay's include directory might be searched first depending on the order of include directories given in CMake.
Depending on the differences with the overlay's include, it could cause build failures or undefined behavior at runtime.

Here are few types of solutions that could address the include directory search order problem.

## Option 1: Reorder the include directories in CMake (ROS 1 does this)

Catkin solved this with `find_package(catkin REQUIRED COMPONENTS pkg_a pkg_b)`.
That calls `find_package()` `pkg_a` and `pkg_b`, then it appends their `*_INCLUDE_DIRS` into the combined variable `catkin_INCLUDE_DIRS`.
Finally it orders the directories in that variable based on the order of workspaces sourced.
It knows about workspaces using a combination of `CMAKE_PREFIX_PATH` and marker files.
This mostly works, except quite a few packages don't use it or use it incorrectly.

Something that makes this kind of solution harder in ROS 2 is the separation of the build tool (colcon) and the build system (CMake).
Instead of one kind of workspace (`catkin`), there are two kinds of workspaces a ROS 2 package would need to be made aware of: `ament` and `colcon`.
The debian packages for ROS 2 install into an `ament` workspace.
It is created by `ros_workspace` using scripts from `ament_package`.
From-source builds and packaging job archives are `colcon` workspaces.
ROS 2 packages need to have knowledge of both to solve this problem the same way.

Another reason this is hard is ROS 2 uses modern CMake targets instead of old-style CMake standard variables.
The old-style CMake standard variable `*_INCLUDE_DIRS` contains a list of directories, including the include directories from all transitive dependencies.
Modern CMake targets have an `INTERFACE_INCLUDE_DIRECTORIES` target property instead, and its dependencies are targets listed in `INTERFACE_LINK_LIBRARIES`.
Getting all include directories requires traversing a tree of targets and querying their parameters.

Something that makes this very hard is `INTERFACE_INCLUDE_DIRECTORIES` may contain generator expressions.
Even if an `ament_cmake` macro was made to traverse the tree of dependencies, read `INTERFACE_INCLUDE_DIRECTORIES` from each target, and overwrite the `INCLUDE_DIRECTORIES` property of the target with the sorted list, it cannot resolve and sort generator expressions.
The include directory search order could still be wrong depending on how any generator expressions got resolved.

Another thing that makes this kind of solution hard is it would almost certainly require all existing ROS packages to update their CMake code to use it.
Every package in the overlay workspace where at least one package is being overridden must implement this solution (whatever it looks like, `ament_target_dependencies()` or otherwise) or the include directory search order may be wrong.

I don't think this option is worth the effort in ROS 2.

## Option 2: Make the include directories be installed to another prefix?

Currently ROS 2 packages install headers `include`, meaning, in a merged workspace (`/opt/ros/foxy`) all packages have the same include directory (`/opt/ros/foxy/include`).
The include `#include <rclcpp/rclcpp.hpp>` would be found at `/opt/ros/foxy/include/rclcpp/rclcpp.hpp`.
If they add another package specific directory, like `include/${PROJECT_NAME}`, then the include directory will become `/opt/ros/foxy/include/rclcpp/`.
That would prevent includes from other packages in a merged workspace to be found.

```CMake
# Instead of this
# install(DIRECTORY include/ DESTINATION include)
# Do this
install(DIRECTORY include/ DESTINATION include/${PROJECT_NAME})
```

This would require all packages in the underlay workspace to implement it.
It would also mean updating [REP 122](https://www.ros.org/reps/rep-0122.html#include).

This alone would bring overriding in ROS 2 to a similar state as ROS 1, but without requiring packages to know about workspaces.
This option seems like much less work for a similar result as option 1.

## Option 3: Can the build tool (colcon) or workspace hide duplicate includes with sandboxing?

What if colcon detected when overriding packages, and made only the non-overridden packages from the underlay available?
Colcon would have to be able to know what files in an underlay belonged to which package.
When the underlay workspace is an isolated one, that's pretty much what happens already.
For merged underlay workspaces that's very hard.

Maybe when overriding a package from an `ament` workspace `/opt/ros/...` colcon could use `dpkg -L ros-...-...` to list the files for a package in a ROS distribution, and symlink them to a hidden underlay that then gets sourced.
For any underlay that is a merged colcon workspace (like the ROS 2 packaging job archives), colcon would need a different mechanisim to determine what files belong to a package.
[... cut ...]
Maybe a user could silence warnings or allow building to proceed with a CLI flag like `--allow-overriding pkg_a`.

At run time (when sourcing an overlay workspace)

  * if any non-leaf package is overridden, warn that it must be API/ABI compatible

Maybe there's no way to silence this warning, but it wouldn't block the user.

Personally I like this option the most because it gives an opportunity to warn about API/ABI issues.
It seems easy to me to miss those and face undefined behavior at runtime.

jacobperron, 2021-09-29:
@sloretz Thanks for laying out those options!

IIUC, https://github.com/ament/ament_cmake/pull/343 proposes something close to option 1 (though I guess it's missing a recursive traversal of target dependencies).

IMO, requiring calls to `ament_target_dependencies` (option 1) or installing headers into a different prefix (option 2) are not terrible options from a user standpoint. Inevitably there will be cases where packages don't follow suite, but that's their prerogative.

Option 3 is intriguing but sounds like the most challenging to implement IMO. 

+1 for option 4, regardless of whether another option is implemented since I don't see how we can do anything else reasonable to guard against ABI incompatibilities in the non-leaf package case.

clalancette, 2021-09-30:
> +1 for option 4, regardless of whether another option is implemented since I don't see how we can do anything else reasonable to guard against ABI incompatibilities in the non-leaf package case.

Yeah, totally agreed; regardless of what else we do, I think we should implement option 4 as a warning (maybe with a flag to disable).

sloretz, 2021-10-07:
See colcon/colcon-core#449 and colcon/colcon-ros#129 for an implementation of option 4. It warns when `colcon build` is called and adds a flag to disable it.

gbiggs, 2021-11-09:
I agree with the concern about non-leaf packages leading to undefined behaviour, but once a warning is in place for those I think that the leaf package use case needs to be fixed urgently. This bug breaks a common and intended workflow, and I regularly lose time to the confusing compiler errors it can cause or having to work around it.

sloretz, 2021-11-22:
>  I think that the leaf package use case needs to be fixed urgently

I'm not sure it needs to be fixed urgently as it's the easiest to work around. If the leaf package is uninstalled then the issue goes away.

gbiggs, 2021-11-22:
> I'm not sure it needs to be fixed urgently as it's the easiest to work around. If the leaf package is uninstalled then the issue goes away.

If you're installing from binary debs, which is when this problem occurs most often due to the merged install space, then uninstalling the leaf package is often not possible. In my case the leaf packages are all rosbag2 packages, which can't be uninstalled without removing the `ros-X-desktop` package. The problem is that they are leaf packages at compile time but not in the debs.

sloretz, 2021-11-29:
> In my case the leaf packages are all rosbag2 packages, which can't be uninstalled without removing the ros-X-desktop package.

Uninstalling is still possible, but I see how that makes it inconvenient. I tried uninstalling one of the rosbag2 debs and saw the other `ros-rolling-*` packages automatically installed with `ros-rolling-desktop` were still installed, but in a state where `apt autoremove` would uninstall them. A workaround for that is to mark the dependencies of `ros-rolling-desktop` as manually installed.

```console 
$ apt-cache depends ros-rolling-desktop | awk '/Depends/ {print $2}' | xargs apt-mark auto
```

sloretz, 2021-11-29:
These are the packages that install headers and are depended on by `ros-rolling-desktop`. Along with colcon warning (colcon/colcon-core#449), implementing option 2 in these packages seems good enough to resolve this ticket.

<details><summary>After installing ros-rolling-desktop, packages that install files into /opt/ros/rolling/include/</summary>

```console
$ dpkg -S /opt/ros/rolling/include/ | sed 's/, /\n/g' | sed 's/:.*$//'
ros-rolling-turtlesim
ros-rolling-map-msgs
ros-rolling-rviz-default-plugins
ros-rolling-interactive-markers
ros-rolling-rosbag2-interfaces
ros-rolling-rviz-rendering
ros-rolling-laser-geometry
ros-rolling-rosbag2-cpp
ros-rolling-rosbag2-compression-zstd
ros-rolling-pcl-msgs
ros-rolling-robot-state-publisher
ros-rolling-tf2-msgs
ros-rolling-message-filters
ros-rolling-kdl-parser
ros-rolling-tf2-eigen
ros-rolling-tf2-ros
ros-rolling-tf2
ros-rolling-pluginlib
ros-rolling-rosbag2-transport
ros-rolling-rcl-lifecycle
ros-rolling-keyboard-handler
ros-rolling-lifecycle-msgs
ros-rolling-pybind11-vendor
ros-rolling-example-interfaces
ros-rolling-trajectory-msgs
ros-rolling-tf2-eigen-kdl
ros-rolling-stereo-msgs
ros-rolling-nav-msgs
ros-rolling-rclcpp-lifecycle
ros-rolling-joy
ros-rolling-diagnostic-msgs
ros-rolling-std-srvs
ros-rolling-geometry-msgs
ros-rolling-tf2-kdl
ros-rolling-pendulum-msgs
ros-rolling-angles
ros-rolling-composition-interfaces
ros-rolling-actionlib-msgs
ros-rolling-class-loader
ros-rolling-rcl-action
ros-rolling-rclcpp
ros-rolling-rosgraph-msgs
ros-rolling-statistics-msgs
ros-rolling-image-transport
ros-rolling-urdf-parser-plugin
ros-rolling-libstatistics-collector
ros-rolling-rosidl-typesupport-fastrtps-c
ros-rolling-fastrtps
ros-rolling-rcl
ros-rolling-rmw-dds-common
ros-rolling-rqt-gui-cpp
ros-rolling-cyclonedds
ros-rolling-rosidl-typesupport-fastrtps-cpp
ros-rolling-iceoryx-utils
ros-rolling-tf2-geometry-msgs
ros-rolling-urdfdom-headers
ros-rolling-libyaml-vendor
ros-rolling-rosidl-typesupport-c
ros-rolling-action-msgs
ros-rolling-iceoryx-binding-c
ros-rolling-rcl-yaml-param-parser
ros-rolling-action-tutorials-interfaces
ros-rolling-unique-identifier-msgs
ros-rolling-rosidl-typesupport-introspection-c
ros-rolling-rosidl-runtime-cpp
ros-rolling-rmw
ros-rolling-rosidl-typesupport-interface
ros-rolling-rosbag2-compression
ros-rolling-shared-queues-vendor
ros-rolling-rosbag2-storage-default-plugins
ros-rolling-rosbag2-storage
ros-rolling-visualization-msgs
ros-rolling-logging-demo
ros-rolling-image-tools
ros-rolling-std-msgs
ros-rolling-shape-msgs
ros-rolling-rcl-interfaces
ros-rolling-urdfdom
ros-rolling-rclcpp-components
ros-rolling-rcpputils
ros-rolling-rosidl-runtime-c
ros-rolling-rcutils
ros-rolling-iceoryx-posh
ros-rolling-builtin-interfaces
ros-rolling-ament-index-cpp
ros-rolling-console-bridge-vendor
ros-rolling-tlsf-cpp
ros-rolling-rcl-logging-interface
ros-rolling-tlsf
ros-rolling-rosidl-typesupport-introspection-cpp
ros-rolling-rosidl-typesupport-cpp
ros-rolling-qt-gui-cpp
ros-rolling-rqt-image-view
ros-rolling-urdf
ros-rolling-image-geometry
ros-rolling-depthimage-to-laserscan
ros-rolling-rmw-fastrtps-cpp
ros-rolling-foonathan-memory-vendor
ros-rolling-tracetools
ros-rolling-rviz-common
ros-rolling-rttest
ros-rolling-fastcdr
ros-rolling-intra-process-demo
ros-rolling-resource-retriever
ros-rolling-tf2-sensor-msgs
ros-rolling-teleop-twist-joy
ros-rolling-zstd-vendor
ros-rolling-tf2-bullet
ros-rolling-sensor-msgs
ros-rolling-orocos-kdl
ros-rolling-cv-bridge
ros-rolling-rclcpp-action
ros-rolling-pcl-conversions
ros-rolling-rmw-fastrtps-shared-cpp
```

</details>

SteveMacenski, 2022-01-21:
Follow up from TSC discussion, I see the following warning when building a workspace containing only navigation2 based on rolling binaries (the most simple imaginable setup). This is when I have a cache in the workspace already from another build and I'm simply rebuilding again when making minor changes (or in the case of this example, no change what so ever). 

Steps
- Source rolling binaries
- Build workspace
- Rebuild workspace

```
[0.460s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'nav2_common' is in: /home/steve/Documents/nav2_ws/install/nav2_common
	'nav_2d_msgs' is in: /home/steve/Documents/nav2_ws/install/nav_2d_msgs
	'dwb_msgs' is in: /home/steve/Documents/nav2_ws/install/dwb_msgs
	'nav2_msgs' is in: /home/steve/Documents/nav2_ws/install/nav2_msgs
	'nav2_voxel_grid' is in: /home/steve/Documents/nav2_ws/install/nav2_voxel_grid
	'nav2_simple_commander' is in: /home/steve/Documents/nav2_ws/install/nav2_simple_commander
	'nav2_util' is in: /home/steve/Documents/nav2_ws/install/nav2_util
	'nav2_amcl' is in: /home/steve/Documents/nav2_ws/install/nav2_amcl
	'nav2_behavior_tree' is in: /home/steve/Documents/nav2_ws/install/nav2_behavior_tree
	'nav2_lifecycle_manager' is in: /home/steve/Documents/nav2_ws/install/nav2_lifecycle_manager
	'nav2_map_server' is in: /home/steve/Documents/nav2_ws/install/nav2_map_server
	'nav_2d_utils' is in: /home/steve/Documents/nav2_ws/install/nav_2d_utils
	'nav2_costmap_2d' is in: /home/steve/Documents/nav2_ws/install/nav2_costmap_2d
	'nav2_rviz_plugins' is in: /home/steve/Documents/nav2_ws/install/nav2_rviz_plugins
	'costmap_queue' is in: /home/steve/Documents/nav2_ws/install/costmap_queue
	'nav2_core' is in: /home/steve/Documents/nav2_ws/install/nav2_core
	'dwb_core' is in: /home/steve/Documents/nav2_ws/install/dwb_core
	'nav2_bt_navigator' is in: /home/steve/Documents/nav2_ws/install/nav2_bt_navigator
	'nav2_controller' is in: /home/steve/Documents/nav2_ws/install/nav2_controller
	'nav2_navfn_planner' is in: /home/steve/Documents/nav2_ws/install/nav2_navfn_planner
	'nav2_planner' is in: /home/steve/Documents/nav2_ws/install/nav2_planner
	'nav2_recoveries' is in: /home/steve/Documents/nav2_ws/install/nav2_recoveries
	'nav2_regulated_pure_pursuit_controller' is in: /home/steve/Documents/nav2_ws/install/nav2_regulated_pure_pursuit_controller
	'nav2_smac_planner' is in: /home/steve/Documents/nav2_ws/install/nav2_smac_planner
	'nav2_smoother' is in: /home/steve/Documents/nav2_ws/install/nav2_smoother
	'nav2_theta_star_planner' is in: /home/steve/Documents/nav2_ws/install/nav2_theta_star_planner
	'nav2_waypoint_follower' is in: /home/steve/Documents/nav2_ws/install/nav2_waypoint_follower
	'dwb_critics' is in: /home/steve/Documents/nav2_ws/install/dwb_critics
	'dwb_plugins' is in: /home/steve/Documents/nav2_ws/install/dwb_plugins
	'nav2_dwb_controller' is in: /home/steve/Documents/nav2_ws/install/nav2_dwb_controller
	'navigation2' is in: /home/steve/Documents/nav2_ws/install/navigation2
	'nav2_bringup' is in: /home/steve/Documents/nav2_ws/install/nav2_bringup
	'nav2_system_tests' is in: /home/steve/Documents/nav2_ws/install/nav2_system_tests
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding costmap_queue dwb_core dwb_critics dwb_msgs dwb_plugins nav2_amcl nav2_behavior_tree nav2_bringup nav2_bt_navigator nav2_common nav2_controller nav2_core nav2_costmap_2d nav2_dwb_controller nav2_lifecycle_manager nav2_map_server nav2_msgs nav2_navfn_planner nav2_planner nav2_recoveries nav2_regulated_pure_pursuit_controller nav2_rviz_plugins nav2_simple_commander nav2_smac_planner nav2_smoother nav2_system_tests nav2_theta_star_planner nav2_util nav2_voxel_grid nav2_waypoint_follower nav_2d_msgs nav_2d_utils navigation2

```

EduPonz, 2022-01-21:
Hi,

I'm seeing this warning when re-building ROS 2 from sources on the same workspace, even when the build was not performed with `merge-install`. It threw me off a little bit at first, but my not merged installed re-build works without problems. What I'm worried is that, apart from @SteveMacenski warning, I see a final line saying:

```
This may be promoted to an error in a future release of colcon-core.
```

I'm concerned that promoting the warning to error may cause my workspace to fail building.

Steps:

* Build ROS 2 from sources with `colcon build`
* Source the installation with `source <path_to_ws>/install/setup.bash` (I'm not sure this step is necessary to reproduce).
* Rebuild workspace with `colcon build`

gbiggs, 2022-01-21:
It is recommended that you _do not_ source and build a workspace in the same terminal.

> Before sourcing the overlay, it is very important that you open a new terminal, separate from the one where you built the workspace. Sourcing an overlay in the same terminal where you built, or likewise building where an overlay is sourced, may create complex issues.

https://docs.ros.org/en/foxy/Tutorials/Workspace/Creating-A-Workspace.html

SteveMacenski, 2022-01-21:
> It is recommended that you do not source and build a workspace in the same terminal.

I'm obviously very late to this conversation but that seems like an unreasonable state of affairs for anyone doing serious development on top of ROS 2. This is not an "overlay" in the sense I understand the phrase being used. This is just rebuilding the code in my workspace which does not exist anywhere else on my system (e.g. no binaries) that would be analogous to any company/researcher's codebase. 

If that is causing some sort of issue, that feels like the kind of thing that either needs to be fixed before hitting a user level or this warning needs to be suppressed. 

Asking users to not use terminals they use to build to also run code or rebuild again seems unreasonable to me

EduPonz, 2022-01-21:
I can confirm that re-building on a clean shell takes my warning away.

I've been building on a sourced terminal for quite some time already and never run into any problems, so I'm curious as to what issues may arise there. I agree with @SteveMacenski that it breaks my workflow a bit to have to switch terminals all the time. I normally have one terminal where I run tests, modify code, re-compile and run tests again. Is this some colcon related issue? I haven't found any reference in its documentation (does not mean is not written somewhere of course)

SteveMacenski, 2022-01-21:
What would resolve most of my concerns would be 
- Whitelisting the install space of the workspace you're currently building so rebuilding workspaces that have been sourced don't cause an error message
- Whitelisting leaf packages that can be trivially overrided successfully so those don't cause an error (since there would be no actual issue present)

sloretz, 2022-01-21:
> Steps
>
>    Source rolling binaries
>    Build workspace (standard colcon build without additional flags)
>    Rebuild workspace

@SteveMacenski Regarding warning when rebuilding a workspace that's sourced, there's an issue about that case here https://github.com/colcon/colcon-core/issues/465 and PR to not warn about overriding in that case here: https://github.com/colcon/colcon-core/pull/473 . Would you be willing to try it out and report if it works for you?

> I've been building on a sourced terminal for quite some time already and never run into any problems, so I'm curious as to what issues may arise there

@EduPonz AFAIK, the issue is it breaks the isolation between packages when building in an isolated workspace. See just after the first quote in this comment https://github.com/colcon/colcon-core/issues/465#issuecomment-1009454932 for more info.

> Whitelisting leaf packages - [...] that can be trivially overrided successfully

@SteveMacenski I'm not sure this is possible. This issue is about the include directory search order problem that arises when overriding a package from a merged underlay. This problem exists with all packages that install headers, even if they're leaf packages in the underlay. My goal to resolve this issue is to install each package's includes to another folder with the package's name for all packages in `ros-*-desktop` (option 2 above: https://github.com/ros2/ros2/issues/1150#issuecomment-930541886).  I don't think colcon can know if a package implements a solution like that or not.

SteveMacenski, 2022-01-21:
> Would you be willing to try it out and report if it works for you?

Will do Monday :+1: 

Couldn't you use the dependendy topography to see if a given package is depended on by any other package? If not (or if so but only depending on packages in the current workspace) pass on the error message.

EduPonz, 2022-01-24:
> @EduPonz AFAIK, the issue is it breaks the isolation between packages when building in an isolated workspace. See just after the first quote in this comment [colcon/colcon-core#465 (comment)](https://github.com/colcon/colcon-core/issues/465#issuecomment-1009454932) for more info.

This is great info, thanks! Then, I understand that once that work is merged, then the warning will disappear when re-building all ROS 2 from sources on a terminal that has that workspace sourced already. Is that correct?

DLu, 2022-02-09:
I am here to second @SteveMacenski 's frustration at this warning.

SteveMacenski, 2022-02-09:
I tried out the PR, helps a great deal! 

I still think the size of this error should be reduced with a flag to show more detail, but that helps at least with the install space related problems which is 95% of my current direct concern.

MiguelBarro, 2022-03-03:
After some testing on the [example repository](https://github.com/jacobperron/overlay_bug_reproduction) that @jacobperron made available to reproduce the issue, I think there is another option available besides the ones in @sloretz [comment](https://github.com/ros2/ros2/issues/1150#issuecomment-930541886).

As **Option 4** it involves a change in **colcon** behaviour (note option 4 is already implemented in **colcon**). 
Currently **colcon** refrains from building a package (unless `--allow-overriding` is used) if it detects a duplicate. It should also refrain from creating the **CMake** config files for the package that is not built. This *ghost* config files created in `/overlay_bug_reproduction/overlay/install/share/package_a/cmake` are the ones that prevent the build from succeed.

When **colcon** directs **CMake** to build package **c** that depends on package **a**, **CMake** `find_package(package_a REQUIRED)` call loads the *ghost* config instead on the real one available in the *underlay* and breaks the include structure.
This can be verified by peering into the `/overlay_bug_reproduction/overlay/install/share/package_a/cmake/package_aExport.cmake` generated by **CMake**:

```CMake
# Compute the installation prefix relative to this file.
get_filename_component(_IMPORT_PREFIX "${CMAKE_CURRENT_LIST_FILE}" PATH)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
if(_IMPORT_PREFIX STREQUAL "/")
  set(_IMPORT_PREFIX "")
endif()

# Create imported target package_a::package_a
add_library(package_a::package_a STATIC IMPORTED)

set_target_properties(package_a::package_a PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"
)
```

the header's position is calculated relative to the **CMake** config file located by `find_package(..)`. Removing these deceitful config prevents the problem.

Both **CMake** and **Colcon** support versioning. So, it's possible to handle two **a** packages with different API/ABI in the overlay/underlay by specifying in the [find_package](https://cmake.org/cmake/help/latest/command/find_package.html#config-mode-version-selection) call the version desired (matching the **colcon** package.xml file value and **CMake** `project()` call value). This way only the **CMake** config file that matches the right version is linked to the **c** target.

sloretz, 2022-03-03:
> As Option 4 it involves a change in colcon behaviour (note option 4 is already implemented in colcon).
Currently colcon refrains from building a package (unless --allow-overriding is used) if it detects a duplicate

Colcon briefly refrained from building the package, but that was demoted to a warning in the latest release.

> It should also refrain from creating the CMake config files for the package that is not built. This ghost config files created in `/overlay_bug_reproduction/overlay/install/share/package_a/cmake` are the ones that prevent the build from succeed.

I think those files should be there. They are created by CMake when `package_a` in the overlay is built.

> Removing these deceitful config prevents the problem.

May I ask for commands I can use to reproduce? Was the overlay a merged workspace too? If the overlay is an isolated workspace, I don't see how the build of `package_d` could find overlay `package_a`'s headers if the `overlay/.../package_a/cmake` files were removed.

> Both CMake and Colcon support versioning. So, it's possible to handle two a packages with different API/ABI in the overlay/underlay by specifying in the [...]

The `find_package()` calls find the correct version when building the overlay, that's not the problem. It's the order targets are depended on that matters.

Say a user installs `ros-rolling-fastdds` which also installs `ros-rolling-fastcdr` because Fast-DDS depends on Fast-CDR. Those packages are installed to `/opt/ros/rolling` which is a merged workspace. Say they want to try a new version of Fast-CDR. For simplicity lets say the new version of Fast-CDR is ABI compatible with the version in `ros-rolling-fastcdr`, but adds a new method `super_fast_cdr()`. The user has their own colcon workspace `~/ws` in which they add the newer version of Fast-CDR to `~/ws/src/Fast-CDR`. The user creates a CMake package `~/ws/src/coolstuff` which depends on `Fast-DDS` and `Fast-CDR` and creates an executable target `coolexe` that tries to use `super_fast_cdr()`. It has this in its `CMakeLists.txt`.

```CMake
find_package(fastdds)
find_package(fastcdr)

target_link_libraries(coolexe fastdds fastcdr)
```

Calling `find_package(fastdds)` first isn't a problem. Even though the `fastddsConfig.cmake` file in `/opt/ros/rolling/...` [does not specify the version of Fast-CDR to find](https://github.com/eProsima/Fast-DDS/blob/ba01ca6a87975c3daccce74c9af90d7621f6fdf5/cmake/packaging/Config.cmake.in#L25), the `overlay` comes first in `CMAKE_PREFIX_PATH`, and so the overlay version of Fast-CDR will be found. However, the build of `coolstuff` does not succeed.

The problem is the order passed to `target_link_libraries()`. The `fastdds` target says to add `/opt/ros/rolling/include` to the list of directories to search for headers. The `fastcdr` target says to add `~/ws/install/fastcdr/include` to the list of directories to search for headers. When `coolexe` is compiled, the compiler sees that it is supposed to search for  `#include <fastcdr/FastCdr.h>`, but it finds it in `/opt/ros/rolling/include/fastcdr/FastCdr.h` because the `fastdds` target told it to look for headers in `/opt/ros/rolling/include` first. The user can solve this by reordering the `target_link_libraries()` call to.

```CMake
# In this example, this order would cause the build to succeed.
target_link_libraries(coolexe fastcdr fastdds)
```

However, that gets impractical with multiple packages depending on each other and the overridden package. The changes are subject to what packages are being overridden so they can't really be committed because coworkers may need the opposite order if they're overriding different packages. Lastly, it's hard to confirm the order is correct because depending on the differences between the headers of any overridden packages, the build may succeed but undefined behavior at runtime.

sloretz, 2022-03-03:
Option 2 solves the case above because the include directory the `fastdds` target says to use is `/opt/ros/rolling/include/fastdds`. That directory doesn't include Fast-CDRs headers so the compiler won't accidentally find the older ones first.

MiguelBarro, 2022-03-04:
Thanks @sloretz for the promptly answer and for pointing out the current **colcon** behavior on package duplication. It seems I was misled by the **colcon** warning message while I reviewed the **CMake** traces. **Colcon** installs the duplicate packages too, there are no dangling configure files.

The whole point of the [repo example](https://github.com/jacobperron/overlay_bug_reproduction) is that having two versions (different API/ABI) of a package in the underlay/overlay may lead to a C++ header issue. That's because several include paths pass to the compiler contain different versions of the headers.
This point has been clarified by @sloretz [last comment](https://github.com/ros2/ros2/issues/1150#issuecomment-1058432516).

Unfortunately **CMake** is not able to deal with this problem because the install dirs (*install prefixes* on **CMake** terminology) are setup externally.
**Colcon** solves the issue by using different **install prefixes** per-CMake package.
For example, a plain  `colcon build` will install `package_a` in `install\package_a` and its headers in
`install\package_a\include`. This effectively prevents the C++ header issue because each package include path only
constains its own headers.

The actual issue is using **colcon** with the flag `--merge-install`. This flag disables the default behavior of assigning a different **install prefix** to each **CMake** package, though effectively merging all include headers in a single include path. The [colcon docs](https://colcon.readthedocs.io/en/released/reference/verb/build.html?highlight=--merge-install#command-line-arguments) make this point clear:

> The disadvantage of using this option is that it doesn’t provide proper
> isolation between packages. For example declaring a dependency on one package
> also allows access to resources from other packages installed in the same
> install prefix (without requiring a declared dependency).

The reason to use a **merge install** is that it provides a unix-like (or gnu-like) deployment of the binaries and
headers resembling those in `/usr`.

ROS2 apt and docker deployments use `--merge-install` and consequently disable the possibility of safely replace core
packages using an overlay approach.

The [Option 2](https://github.com/ros2/ros2/issues/1150#issuecomment-930541886) fixes this issue in the underlay
(in our case the ROS2 deployment) by effectively disabling the `--merge-install` effect. That is, applying **Option 2** to
the underlay is like using current **colcon** default behavior.

Applying **Option 2** only to the overlay doesn't prevent the issue at all, in fact is like using **colcon** without the
`--merge-install` flag.

There is an easy **colcon** workaround for the use case of replacing an underlay package based on the `--install-base`
flag. Basically we can make the overlay use the underlay install dir. This way the headers and binaries are effectively
replaced and the issue is corrected. For the [repo example](https://github.com/jacobperron/overlay_bug_reproduction)
it would be:

```bash
#! /bin/bash

git clone https://github.com/jacobperron/overlay_bug_reproduction.git
. /ros_entrypoint.sh
cd overlay_bug_reproduction/underlay

colcon build --merge-install
# colcon build

. ./install/local_setup.bash
cd ../overlay

# colcon build --merge-install
touch $(find ./src/package_a)
colcon build --merge-install --install-base ../underlay/install
```

note we must *touch* the package_a files in order to assure CMake will replace them.

sloretz, 2022-03-04:
> Unfortunately CMake is not able to deal with this problem because the install dirs (install prefixes on CMake terminology) are setup externally.

The way I like to think about this is a workspace is a folder with software and shell scripts. Workspace shell scripts can only modify environment variables. Environment variables can't control the order include directories are searched within a project, so a different mechanism is needed. Installing packages to different folders is one option. Installing just the headers to unique folders is a hybrid option.

> The https://github.com/ros2/ros2/issues/1150#issuecomment-930541886 fixes this issue in the underlay (in our case the ROS2 deployment) by effectively disabling the --merge-install action. That is, applying Option 2 to the underlay is like using current colcon default behavior.

It separates the directories the headers are installed, but nothing else is isolated.

> ROS2 apt and docker deployments use --merge-install and consequently disable the possibility of safely replace core packages using an overlay approach.

Small correction, I'm pretty sure the `apt` packages don't use `colcon`. 

> Applying Option 2 only to the overlay doesn't prevent the issue at all, in fact is like using colcon without the
--merge-install flag.

Correct, option 2 requires the packages in the underlay to implement it.

> There is an easy colcon workaround for the use case of replacing an underlay package based on the --install-base
flag. Basically we can make the overlay use the underlay install dir.

That's an interesting idea. I'm not sure would be desirable for common use cases. For `apt` packages it would require building as root to specify `--install-base /opt/ros/rolling`. I'm not sure how the system package manager would respond if the package being overridden would respond. If the overriding package has more files than the overridden package, would those get left behind if the user uninstalled them?

MiguelBarro, 2022-03-04:
> It separates the directories the headers are installed, but nothing else is isolated.

That's the key issue. **colcon** only can modify the **install prefixes** but fine tunning where the headers are installed is only possible from the **CMake** packages itself. Thus, neither **colcon**'s `--merge-install` nor another flag can solve the header issue. This justifies ROS2 package updates like https://github.com/eProsima/Fast-CDR/pull/120 or https://github.com/eProsima/Fast-DDS/pull/2535.

> I'm not sure how the system package manager would respond if the package being overridden would respond. If the
> overriding package has more files than the overridden package, would those get left behind if the user uninstalled them?

I'm afraid I've only follow this approach on docker containers for debugging purposes. I won't advise to use this kind of approach on production environments 😅.

sloretz, 2022-04-06:
All packages in `ros-rolling-desktop` have been updated to install their headers to a unique include directory.

I'm looking for feedback on this PR to colcon's documentation about this and other known issues with overriding packages. https://github.com/colcon/colcon.readthedocs.org/pull/88/files

If overriding packages is important to you, I would appreciate your review.

doganulus, 2022-08-16:
I don't care what problems overriding can cause. Installing headers to `include\${PROJECT_NAME}` is a pretty bad decision for `C++` convention. 

I may be late to the discussion but it is not hard to see that this decision isolates the ROS community from the best `C++` practices. Now we have to include as follows for a proper structure. Does it look good? Nope...

```
#include rclcpp/rclcpp/rclcpp.hpp
```

clalancette, 2022-08-16:
> ```c++
> #include "rclcpp/rclcpp/rclcpp.hpp"
> ```

As far as we know, this should never be the case.  In all packages currently released into Humble, a simple:

```
#include "rclcpp/rclcpp.hpp
```

works just fine.

doganulus, 2022-08-16:
In `C++` you conventionally include `include` directory as the name implies. The paths must be defined relative to `include` directory. 

Of course we can add multiple `include\${PROJECT_NAME}` paths but this is an abuse. Just works fine is not the point here. Spaghetti codes can work too when you setup correctly. But it is a bad practice that will hurt you later.

sloretz, 2022-08-16:
> #include "rclcpp/rclcpp/rclcpp.hpp"

That `#include` directive should never be used. It only works if something else (not rclcpp) has added `/opt/ros/<distro>/include` as an include path, meaning it could stop working if your dependencies change.

> Of course we can add multiple `-Iinclude\${PROJECT_NAME}` paths to the compiler but this is an abuse. 

It sounds like you'd like to provide only one include path to the compiler; however, I don't think that's generally feasible. Packages having unique include paths are very common. Looking at only system packages (not ROS packages), I see 98 unique include paths on my Ubuntu Bionic system - and that's just the packages which provide a pkg-config file.

``` console
$ for i in $(pkg-config --list-all | awk '{print $1}'); do pkg-config --cflags $i 2>/dev/null | grep --only-matching -e "-I[^ ]*" ; done | sort | uniq | wc -l
98
```

doganulus, 2022-08-16:
I would be happy too if you follow `/usr/include` conventions as other `pkg-config` without polluting my compiler commands with 256 or something different include directives. 

Two options here:
1. Install all packages on `/usr/install/ros/humble` or `/usr/local/include/ros/humble`
2. Install all packages on `/opt/ros/humble/include` (as in `galactic`)

But make it single. Currently you just seem to patch a small fringe problem by changing established conventions in my opinion. (You did it anyway for an LTS...argh)

clalancette, 2022-08-16:
Since the original problem this ticket was addressing is now solved in Humble and Rolling, I'm going to close this one out.

If you'd like to open a separate issue about changing back, you are free to open one up.  But I will say that we'd have to hear very strong arguments on why this issue is not important enough to fix (it is not a fringe problem; many users ran into it).

closed 2022-08-16 by clalancette

doganulus, 2022-08-16:
You have closed it but don't you give depreciation warnings for the feature described here? I wouldn't make significant changes for features to be depreciated...  (And actually I didn't encounter the problem mentioned here, I do not want to rely on ROS tooling as much as possible. Just wanted to switch from `galactic` to `humble`.)