ros2/ros2 #1151: Python wrapper performance terrible, especially when customized msg includes big array

crazyhank, 2021-06-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS2 Foxy
- DDS implementation:
  - default

#### Steps to reproduce issue

My customized message struct is as follow:
```
int32 width
int32 height
int32 seq
uint64 time
byte[2097152] data
```
As you see, I try to simulate depth camera data frame. Then, I wrote a simple test program using python API to see how much latency cost for sending data.

For subscriber:
```
import rclpy
from rclpy.node import Node

from my_struct.msg import ImageFixed
import time

class MinimalSubscriber(Node):
    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            ImageFixed,
            'my_topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning
    def listener_callback(self, msg):
        now_us = (int)(time.time() * 1000000)
        latency = now_us - msg.time
        self.get_logger().info('I heard: (%d, %d, %d), size = %d, latency = %d us' % (msg.width, msg.height, msg.seq, len(msg.data), latency))

def main(args=None):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()
    rclpy.spin(minimal_subscriber)

    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

For publisher:
```
import rclpy
import time
from rclpy.node import Node

from my_struct.msg import ImageFixed

class MinimalPublisher(Node):
    def __init__(self):
        super().__init__('minimal_publisher')
        self.publisher_ = self.create_publisher(ImageFixed, 'my_topic', 10)
        timer_period = 2  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.count_ = 0
        self.msg = ImageFixed()  
    def timer_callback(self):
        self.msg.width = 128
        self.msg.height = 64
        self.msg.seq = self.count_
        self.msg.time = (int)(time.time() * 1000000)
        self.publisher_.publish(self.msg)
        self.get_logger().info('Publishing: (%d %d %d) ' % (self.msg.width, self.msg.height, self.msg.seq))
        self.count_ += 1

def main(args=None):
    rclpy.init(args=args)
    minimal_publisher = MinimalPublisher()

    rclpy.spin(minimal_publisher)
    minimal_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

I did my test on a Xeon(R) CPU E5-2680 Server, from the output message, you can see the performance is very bad.
> $ ros2 run py_listen_talk listener
[INFO] [1623206317.455130818] [minimal_subscriber]: I heard: (128, 64, 18), size = 2097152, latency = 1035255 us
[INFO] [1623206319.418601105] [minimal_subscriber]: I heard: (128, 64, 19), size = 2097152, latency = 1018215 us
[INFO] [1623206321.409383927] [minimal_subscriber]: I heard: (128, 64, 20), size = 2097152, latency = 1008987 us
[INFO] [1623206323.420597885] [minimal_subscriber]: I heard: (128, 64, 21), size = 2097152, latency = 1019648 us
[INFO] [1623206325.418819171] [minimal_subscriber]: I heard: (128, 64, 22), size = 2097152, latency = 1018413 us
[INFO] [1623206327.408069254] [minimal_subscriber]: I heard: (128, 64, 23), size = 2097152, latency = 1007643 us
[INFO] [1623206329.408010756] [minimal_subscriber]: I heard: (128, 64, 24), size = 2097152, latency = 1007611 us
[INFO] [1623206331.414896283] [minimal_subscriber]: I heard: (128, 64, 25), size = 2097152, latency = 1014110 us
[INFO] [1623206333.418220902] [minimal_subscriber]: I heard: (128, 64, 26), size = 2097152, latency = 1017825 us
[INFO] [1623206335.407955137] [minimal_subscriber]: I heard: (128, 64, 27), size = 2097152, latency = 1007563 us
[INFO] [1623206337.408134504] [minimal_subscriber]: I heard: (128, 64, 28), size = 2097152, latency = 1007736 us
[INFO] [1623206339.407842308] [minimal_subscriber]: I heard: (128, 64, 29), size = 2097152, latency = 1007455 us
[INFO] [1623206341.407910282] [minimal_subscriber]: I heard: (128, 64, 30), size = 2097152, latency = 1007512 us
[INFO] [1623206343.407709552] [minimal_subscriber]: I heard: (128, 64, 31), size = 2097152, latency = 1007316 us

If I use C++ API to implement the same behaviour, the performance is normal. I think the problem is strongly related with python wrapper.

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

gavanderhoorn, 2021-06-09:
Is this https://github.com/ros2/rclpy/issues/763?

If so: does https://github.com/ros2/rosidl_python/pull/129 change anything for you @crazyhank ?

crazyhank, 2021-06-09:
> Is this [ros2/rclpy#763](https://github.com/ros2/rclpy/issues/763)?
> 
> If so: does [ros2/rosidl_python#129](https://github.com/ros2/rosidl_python/pull/129) change anything for you @crazyhank ?

Even if I use latest ROS2(Galactic) version, the problem still exists. Is there some soultion? I can't find something in the thread you provide.

clalancette, 2021-06-09:
> Even if I use latest ROS2(Galactic) version, the problem still exists. Is there some soultion? I can't find something in the thread you provide.

Right, we have not yet integrated the patch that should improve it.  If you'd like to test out ros2/rosidl_python#129, you'll have to compile ROS 2 [from source](https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html).

But this definitely seems like the same problem as ros2/rclpy#763.  So I'm going to close this.

crazyhank, 2021-06-10:
> Right, we have not yet integrated the patch that should improve it. If you'd like to test out [ros2/rosidl_python#129](https://github.com/ros2/rosidl_python/pull/129), you'll have to compile ROS 2 [...]

I tried your branch, but the problem still exists. Maybe you can test it using my code I provide.

clalancette, 2021-06-11:
OK, so a couple of things are going on here.

Using your code exactly as-is above, I'm seeing the same slowness that you are:

```
[INFO] [1623440630.815730800] [minimal_subscriber]: I heard: (128, 64, 2), size = 2097152, latency = 924453 us
[INFO] [1623440632.714061476] [minimal_subscriber]: I heard: (128, 64, 3), size = 2097152, latency = 834692 us
[INFO] [1623440634.710980982] [minimal_subscriber]: I heard: (128, 64, 4), size = 2097152, latency = 831651 us
[INFO] [1623440636.701111073] [minimal_subscriber]: I heard: (128, 64, 5), size = 2097152, latency = 821717 us
[INFO] [1623440638.713725811] [minimal_subscriber]: I heard: (128, 64, 6), size = 2097152, latency = 834298 us
[INFO] [1623440640.712988727] [minimal_subscriber]: I heard: (128, 64, 7), size = 2097152, latency = 833569 us
```

We've now integrated ros2/rosidl_python#129, and I'm testing with that, so that can't be the cause.

However, if I change the `byte[2097152] data` in the message to `uint8[2097152] data`, then all of a sudden I get much better performance:

```
[INFO] [1623440929.875981749] [minimal_subscriber]: I heard: (128, 64, 47), size = 2097152, latency = 37596 us
[INFO] [1623440931.876028432] [minimal_subscriber]: I heard: (128, 64, 48), size = 2097152, latency = 37612 us
[INFO] [1623440933.876379817] [minimal_subscriber]: I heard: (128, 64, 49), size = 2097152, latency = 37900 us
[INFO] [1623440935.876499468] [minimal_subscriber]: I heard: (128, 64, 50), size = 2097152, latency = 37920 us
[INFO] [1623440937.876581707] [minimal_subscriber]: I heard: (128, 64, 51), size = 2097152, latency = 38051 us
[INFO] [1623440939.875991618] [minimal_subscriber]: I heard: (128, 64, 52), size = 2097152, latency = 37348 us
```

So there is clearly some difference on how we handle bytes vs. uint8.  So that looks like a workaround for now.

That said, I'm going to reopen this since it is a bug.  I'm also going to rename the title to more clearly reflect what's going on here.

sloretz, 2021-06-23:
I'm pretty sure this is a duplicate of https://github.com/ros2/rosidl_python/issues/134 . The field `byte[2097152] data` is generated as a list of 2097152 `bytes()` instances of size 1 instead of a single `bytes()` instance of size 2097152.

clalancette, 2021-06-24:
OK, that is good to know, thanks @sloretz .  I'm going to close this one as a duplicate of that, then.

closed 2021-06-24 by clalancette