ros2/ros2 #1153: Gallactic compilation from source leading to lib-linking failure with OGRE library; package failed: rviz_ogre_vendor

nitin5, 2021-06-22:
## Bug report

**Required Info:**
- Operating System:
Ubuntu 20.04, ROS Gallactic
- Installation type: From Source
- Version or commit hash:
  - 
  Since the code was pulled by the build script as provided on the official web-page, Code shall be Gallactic-release. The code was downloaded yesterday.
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
I have followed the steps for marked in page [https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html](https://docs.ros.org/en/galactic/Installation/Ubuntu-Development-Setup.html) 
#### Expected behavior
Error free compliation
#### Actual behavior
                                                                                                                                                                       
Starting >>> trajectory_msgs                                                                                                                                                                       
--- stderr: rviz_ogre_vendor                                                                                                                                                                              
/usr/bin/ld: ../../lib/libOgreMain.so.1.12.1: undefined reference to `Ogre::GpuProgramManager::getByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/usr/bin/ld: ../../lib/libOgreMain.so.1.12.1: undefined reference to `Ogre::GpuProgramManager::getSingleton()'
/usr/bin/ld: ../../lib/libOgreMain.so.1.12.1: undefined reference to `Ogre::ArchiveManager::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/usr/bin/ld: ../../lib/libOgreMain.so.1.12.1: undefined reference to `Ogre::ArchiveManager::ArchiveManager()'
/usr/bin/ld: ../../lib/libOgreMain.so.1.12.1: undefined reference to `Ogre::ArchiveManager::getSingleton()'
/usr/bin/ld: ../../lib/libOgreMain.so.1.12.1: undefined reference to `Ogre::ArchiveManager::addArchiveFactory(Ogre::ArchiveFactory*)'
collect2: error: ld returned 1 exit status
make[5]: *** [Tools/VRMLConverter/CMakeFiles/VRMLConverter.dir/build.make:175: bin/VRMLConverter] Error 1
make[4]: *** [CMakeFiles/Makefile2:1464: Tools/VRMLConverter/CMakeFiles/VRMLConverter.dir/all] Error 2
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:152: all] Error 2
make[2]: *** [CMakeFiles/ogre-v1.12.1.dir/build.make:116: ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_ogre_vendor [14min 6s, exited with code 2]
Aborted  <<< diagnostic_msgs [13.5s]                                                                                                                                                                    
Aborted  <<< shape_msgs [13.3s]
Aborted  <<< trajectory_msgs [13.2s]
Aborted  <<< visualization_msgs [13.7s]
Aborted  <<< nav_msgs [13.9s]
Aborted  <<< tf2_msgs [14.5s]                                                                                                                                              
Aborted  <<< sensor_msgs [14.9s]                                                                                                                  
Aborted  <<< rmw_implementation [23.5s]                                                                                
Aborted  <<< tf2 [17.5s]                                            
                                           
Summary: 175 packages finished [14min 52s]
  1 package failed: rviz_ogre_vendor
  9 packages aborted: diagnostic_msgs nav_msgs rmw_implementation sensor_msgs shape_msgs tf2 tf2_msgs trajectory_msgs visualization_msgs
  1 package had stderr output: rviz_ogre_vendor
  145 packages not processed

#### Additional information

----

emersonknapp, 2021-06-22:
Hi, please make sure to put your console output in triple-backtick code blocks - it makes it a lot easier to read.

Additionally, this question seems best suited for https://answers.ros.org, where the community will be more likely to help you with a troubleshooting problem like this.

flynneva, 2021-06-22:
@nitin5 i think this is a case of future development interfering with old releases.

it looks like [the ros2.repos](https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos) file you downloaded is checking out the ROS2 development branch....that branch already has commits for the next ROS2 release (Humble). A [separate `galactic` branch](https://github.com/ros2/rviz/tree/galactic) was created to "freeze" what was released for galactic already.

a simple solution would be to checkout the `galactic` branch and recompile I think.

clalancette, 2021-06-23:
Yes, that is almost certainly the case.  We have an [open PR](https://github.com/ros2/ros2_documentation/pull/1693) to update the instructions explicitly for Galactic, so that should solve this particular issue.

As @emersonknapp says, the first place to ask questions of this kind is on https://answers.ros.org.  I'm going to close this one out.

closed 2021-06-23 by clalancette