ros2/ros2 #1154: geometry_msgs is not included in tf2/impl/utils

HosseinSheikhi, 2021-06-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Both
  
#### Steps to reproduce issue
I was trying to build [slam_toolbox](https://github.com/SteveMacenski/slam_toolbox) 
```
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

#### Expected behavior
Builds normally
#### Actual behavior
     |  ^~~~~~~
In file included from /opt/ros/rolling/include/tf2/utils.h:20,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/loop_closure_assistant.hpp:29,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/loop_closure_assistant.cpp:22:
/opt/ros/rolling/include/tf2/impl/utils.h: In function ‘tf2::Quaternion tf2::impl::toQuaternion(const Quaternion&)’:
/opt/ros/rolling/include/tf2/impl/utils.h:44:3: error: ‘fromMsg’ was not declared in this scope
   44 |   fromMsg(q, res);
      |   ^~~~~~~
/opt/ros/rolling/include/tf2/impl/utils.h: At global scope:
/opt/ros/rolling/include/tf2/impl/utils.h:53:56: error: ‘QuaternionStamped’ in namespace ‘geometry_msgs::msg’ does not name a type
   53 | tf2::Quaternion toQuaternion(const geometry_msgs::msg::QuaternionStamped & q)
      |                                                        ^~~~~~~~~~~~~~~~~
/opt/ros/rolling/include/tf2/impl/utils.h: In function ‘tf2::Quaternion tf2::impl::toQuaternion(const int&)’:
/opt/ros/rolling/include/tf2/impl/utils.h:56:13: error: request for member ‘quaternion’ in ‘q’, which is of non-class type ‘const int’
   56 |   fromMsg(q.quaternion, res);
      |             ^~~~~~~~~~
/opt/ros/rolling/include/tf2/impl/utils.h:56:3: error: ‘fromMsg’ was not declared in this scope
   56 |   fromMsg(q.quaternion, res);
      |   ^~~~~~~
/opt/ros/rolling/include/tf2/impl/utils.h: In function ‘tf2::Quaternion tf2::impl::toQuaternion(const tf2::Stamped<T>&)’:
/opt/ros/rolling/include/tf2/impl/utils.h:67:23: error: ‘QuaternionStamped’ is not a member of ‘geometry_msgs::msg’
   67 |   geometry_msgs::msg::QuaternionStamped q = toMsg<tf2::Stamped<T>,
      |                       ^~~~~~~~~~~~~~~~~
/opt/ros/rolling/include/tf2/impl/utils.h:69:23: error: ‘q’ was not declared in this scope
   69 |   return toQuaternion(q);
      |                       ^
/opt/ros/rolling/include/tf2/impl/utils.h: In function ‘tf2::Quaternion tf2::impl::toQuaternion(const T&)’:
/opt/ros/rolling/include/tf2/impl/utils.h:80:38: error: ‘toMsg’ was not declared in this scope
   80 |   geometry_msgs::msg::Quaternion q = toMsg<T, geometry_msgs::msg::QuaternionStamped>(t);
      |                                      ^~~~~
/opt/ros/rolling/include/tf2/impl/utils.h:80:45: error: expected primary-expression before ‘,’ token
   80 |   geometry_msgs::msg::Quaternion q = toMsg<T, geometry_msgs::msg::QuaternionStamped>(t);
      |                                             ^
In file included from /opt/ros/rolling/include/tf2/utils.h:20,
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_mapper.hpp:24,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/slam_mapper.cpp:22:
/opt/ros/rolling/include/tf2/impl/utils.h:41:36: error: ‘geometry_msgs’ does not name a type
   41 | tf2::Quaternion toQuaternion(const geometry_msgs::msg::Quaternion & q)
      |                                    ^~~~~~~~~~~~~
/opt/ros/rolling/include/tf2/impl/utils.h:41:54: error: expected unqualified-id before ‘::’ token
   41 | tf2::Quaternion toQuaternion(const geometry_msgs::msg::Quaternion & q)
      |                                                      ^~
/opt/ros/rolling/include/tf2/impl/utils.h:41:54: error: expected ‘)’ before ‘::’ token
   41 | tf2::Quaternion toQuaternion(const geometry_msgs::msg::Quaternion & q)
      |                             ~                        ^~
      |                                                      )
/opt/ros/rolling/include/tf2/impl/utils.h:41:54: error: expected initializer before ‘::’ token
   41 | tf2::Quaternion toQuaternion(const geometry_msgs::msg::Quaternion & q)
      |                                                      ^~
/opt/ros/rolling/include/tf2/impl/utils.h:53:36: error: ‘geometry_msgs’ does not name a type
   53 | tf2::Quaternion toQuaternion(const geometry_msgs::msg::QuaternionStamped & q)
      |                                    ^~~~~~~~~~~~~
[... 54 lines cut ...]
                 from /home/hossein/slam_ws/src/slam_toolbox/include/slam_toolbox/slam_toolbox_common.hpp:32,
                 from /home/hossein/slam_ws/src/slam_toolbox/src/slam_toolbox_common.cpp:23:
/opt/ros/rolling/include/rclcpp/node.hpp:406:3: note: declared here
  406 |   declare_parameter(const std::string & name);
      |   ^~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/toolbox_common.dir/build.make:115: CMakeFiles/toolbox_common.dir/src/slam_mapper.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/toolbox_common.dir/build.make:89: CMakeFiles/toolbox_common.dir/src/loop_closure_assistant.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:227: CMakeFiles/toolbox_common.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< slam_toolbox [42.4s, exited with code 2]

Summary: 0 packages finished [42.5s]
  1 package failed: slam_toolbox
  1 package had stderr output: slam_toolbox

#### Additional information
It appears for both methods of installing ros2. It can be solved by including the geometry_msgs to the header file, so slam_toolbox can be built successfully.
----
## Feature request

#### Feature description

#### Implementation considerations

ahcorde, 2021-06-23:
Did you check this PR https://github.com/ros2/geometry2/pull/432 ?

HosseinSheikhi, 2021-06-23:
I have checked this PR, and build the ROS2 again, now the following error appears:
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/async_slam_toolbox_node.dir/build.make:253: async_slam_toolbox_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:765: CMakeFiles/async_slam_toolbox_node.dir/all] Error 2
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/lifelong_slam_toolbox_node.dir/build.make:253: lifelong_slam_toolbox_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:979: CMakeFiles/lifelong_slam_toolbox_node.dir/all] Error 2
/usr/bin/ld: libtoolbox_common.so: undefined reference to `void tf2::fromMsg<geometry_msgs::msg::Quaternion_<std::allocator<void> >, tf2::Quaternion>(geometry_msgs::msg::Quaternion_<std::allocator<void> > const&, tf2::Quaternion&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sync_slam_toolbox_node.dir/build.make:253: sync_slam_toolbox_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:424: CMakeFiles/sync_slam_toolbox_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< slam_toolbox [1min 7s, exited with code 2]

HosseinSheikhi, 2021-06-24:
It seems to be an issue with slam_toolbox.

HosseinSheikhi, 2021-06-24:
not sure, let it be open :)

ShivamPR21, 2021-06-25:
@HosseinSheikhi which branch/installation of ros2 you are using?

The SLAM toolbox has some problems with `galactic/rolling` in its parameter definitions. The [error](https://github.com/ros2/ros2/issues/1154#issuecomment-867171780) is more likely to be caused by some linking errors, try to rebuild from scratch(Sometimes solved the problems for me with `no-reference found` errors).

HosseinSheikhi, 2021-07-20:
@clalancette would you please take a look at the [corresponding issue in slam_toolbox.](https://github.com/SteveMacenski/slam_toolbox/issues/423#issue-945167015)

clalancette, 2021-10-01:
As far as I can tell, this was fixed in slam_toolbox, so I'm going to close this.  @HosseinSheikhi if that is not the case, feel free to reopen.

closed 2021-10-01 by clalancette