ros2/ros2 #1155: File ros2-dashing-20210610-windows-debug-amd64.zip incomplete

mce16, 2021-07-03:
## Bug report

**Required Info:**

- Operating System:
  - 
  - Windows 10
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
Download file and unzip then error 

! D:\ros2-dashing-20210610-windows-debug-amd64.zip: Unexpected end of archive

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

christophebedard, 2021-07-03:
I'm also getting an error when trying to unzip the file. The non-debug file (`ros2-dashing-20210610-windows-amd64.zip`) works fine.

mce16, 2021-07-04:
Release build has build issue if rti connext not installed, similar like this: the build simply failed when you try to build your own ros2 overlay worksapce. Same issue as follow, apparently was fixed by reintroduced here.

https://github.com/ros2/rmw_connext/issues/388

christophebedard, 2021-07-04:
> I'm also getting an error when trying to unzip the file. The non-debug file (`ros2-dashing-20210610-windows-amd64.zip`) works fine.

Looks like you can get the "original" Windows debug archive from the packaging job: https://ci.ros2.org/job/ci_packaging_windows/173/artifact/ws/ros2-package-windows-AMD64.zip. This one is 1.05 GB while the one from the release page is 973 MB, so something probably went wrong when uploading it to GitHub for the release.

@nuclearsandwich could the Windows debug archive for the Dashing patch release 9 be updated?

> Release build has build issue if rti connext not installed, similar like this: the build simply failed when you try to build your own ros2 overlay worksapce. Same issue as follow, apparently was fixed by reintroduced here.

Unfortunately Dashing is now end-of-life, so you're going to have to move to a non-EOL distro or apply the fix yourself (or just install Connext).

mce16, 2021-07-06:
Thank you @christophebedard , but i have a new problem with this;

i am wondering is there a way for ros2 build to include dependencies like python such a way that it doesn't matter what version of python is running on my system, and all the binaries and executables downloaded should be able to work flawlessly as soon as i sourced the environment?

I have problem to run tools like rqt now even with this full debug download at the moment, and i believe it was due
to python version incompatibility, and i couldn't solve it. I have followed this guide previously try to build the debug version of ros2. 

[https://docs.ros.org/en/dashing/Installation/Windows-Development-Setup.html#extra-stuff-for-debug-mode]

```

PySide2/__init__.py: Unable to import shiboken2 from D:\MyGit\ros2-prebuilt\ros2-windows\Scripts, D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages, c:\python37\python37_d.zip, c:\python37\DLLs, c:\python37\lib, c:\python37, c:\python37\lib\site-packages, c:\python37\lib\site-packages\win32, c:\python37\lib\site-packages\win32\lib, c:\python37\lib\site-packages\Pythonwin
Traceback (most recent call last):
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Scripts\rqt-script.py", line 33, in <module>
    sys.exit(load_entry_point('rqt-gui==1.0.7', 'console_scripts', 'rqt')())
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\rqt_gui\main.py", line 87, in main
    sys.exit(Main().main())
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\rqt_gui\main.py", line 59, in main
    plugin_argument_provider=plugin_argument_provider)
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\qt_gui\main.py", line 409, in main
    from python_qt_binding import QT_BINDING
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\__init__.py", line 56, in <module>
    from python_qt_binding.binding_helper import loadUi  # noqa: F401
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 279, in <module>
    getattr(sys, 'SELECT_QT_BINDING_ORDER', None),
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 120, in _select_qt_binding
    (', '.join(["'%s'" % b for b in binding_order]), '\n'.join(error_msgs)))
ImportError: Could not find Qt binding (looked for: 'pyqt', 'pyside'):
  ImportError for 'pyqt': No module named 'PyQt5.QtCore'
Traceback (most recent call last):
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 153, in _load_pyqt
    _named_import('PyQt5.%s' % module_name)
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
ModuleNotFoundError: No module named 'PyQt5.QtCore'

  ImportError for 'pyside': No module named 'shiboken2.shiboken2'
Traceback (most recent call last):
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 185, in _load_pyside
    _named_import('PySide2.%s' % module_name)
  File "D:\MyGit\ros2-prebuilt\ros2-windows\Lib\site-packages\python_qt_binding\binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
  File "c:\python37\lib\site-packages\PySide2\__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "c:\python37\lib\site-packages\PySide2\__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "c:\python37\lib\site-packages\shiboken2\__init__.py", line 27, in <module>
    from .shiboken2 import *
ModuleNotFoundError: No module named 'shiboken2.shiboken2'

```

`

christophebedard, 2021-07-06:
> i am wondering is there a way for ros2 build to include dependencies like python such a way that it doesn't matter what version of python is running on my system, and all the binaries and executables downloaded should be able to work flawlessly as soon as i sourced the environment?
>
> I have problem to run tools like rqt now even with this full debug download at the moment, and i am not sure if it was due
to python version incompatibility, and i couldn't solve it. I have followed this guide previously try to build the debug version of ros2. I want to know if you get the same error or not.

For "isolating" Python packages, you should look into virtual environments (venv). However, this question should probably be posted on https://answers.ros.org/ instead. I can try to help out if you post your question over there.

nuclearsandwich, 2021-07-06:
The windows debug binary has been re-uploaded and is now complete.

As mentioned by @christophebedard Dashing is no longer supported so other issues will not be addressed.
The pre-built binaries for ROS 2 require the exact system setup described in the installation instructions.
Other configurations may work from source but are unlikely to be successful when using binaries.

closed 2021-07-06 by nuclearsandwich

mce16, 2021-07-07:
@christophebedard , i have posted it to 
[https://answers.ros.org/question/381786/python-and-rqt-problems/](url)

Could you take a look, i have run out of ideas what to do to get the rqt to work in my machine.