ros2/ros2 #1157: curl: (35)  OpenSSL SSL_connect: SSL_ERROR_SYSCAL

dongpan90, 2021-07-18:
## Bug report

**Required Info:**

- Operating System:
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
  - 
- Installation type:  i want to get source code and build it
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
```

#### Expected behavior
I can get the ros.key
#### Actual behavior
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to raw.githubusercontent.com:443
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2021-07-18:
ASAIK, Raspbian is not officially supported. (see https://docs.ros.org/en/galactic/Installation.html) And this seems to be a problem for curl setting, could be related to https://github.com/openssl/openssl/issues/9566.

clalancette, 2021-07-19:
Yes, this seems to be a problem on your end with being able to access GitHub, as it seems to work fine here.  Probably you are either using a proxy, and have something set up incorrectly, or you need to use a proxy.  Either way, this isn't likely to be a bug in ROS 2, so I'm closing this.  If you still think that this is a problem on our side, feel free to reopen.

closed 2021-07-19 by clalancette