ros2/ros2 #1159: Message definitions at runtime ("on-the-wire")

emersonknapp, 2021-07-21:
## Feature request

#### Feature description

This is a request for some way to discover the definition of message types, dynamically at runtime. Additionally, some equivalent of the ROS 1 message hashsum would be valuable to detect message definition mismatches.

This feature is desired for a few use cases:
* Store definitions in `rosbag2` for inspection of bags on systems that may not have all message packages installed
* Inspect messages in off-host visualization tools - specifically Foxglove/Webviz web-tech tools, for plotting or printing of message contents
  * This use case could equally apply to rqt/rviz 

#### Implementation considerations

Not sure yet, needs design discussion. Bringing up at next Middleware WG meeting.

Perhaps this could be  a similar API to `rmw_get_publishers_info_by_topic` from `rmw/get_topic_endpoint_info.h` - different publishers on the same topic could potentially be using different versions of a message definition (or even messages with the same name but unrelated definitions), so getting this info per-publisher is probably necessary for completeness.

defunctzombie, 2021-07-22:
Worth noting that in this functionality exists in ROS1. When a subscriber connects to a publisher, the publisher sends the message definition for the topic.

emersonknapp, 2021-07-28:
Notes from Middleware WG discussion:

* (Gerardo) DDS supports this via DDS XTypes and the TypeObject
  * DynamicType API lets you introspect types at runtime
  * (Emerson) is this on top of what we’re already using?
    * (Gerardo) Data type is part of the “TypeCode” field in the discovery of DataWriter.
* (Gerardo) When using XTypes if the types are not convertible you will get an error/warning and the type name is basically ignored.
* (Emerson) is there a new API for dynamic data?
  * (Gerardo) not really, it should just be turned on to get some implicit features, but if you want to use the dynamic type introspection then that is a new API.
* (Emerson) is it being used already?
  * (Gerardo/Andrea) it is on by default in rmw_connextdds
  * (Miguel) It is not on by default, but could be turned on.
  * Not sure about cyclonedds
    * (Joe) On going work
* (Gerardo) There is a type hash similar to ROS 1’s md5sum
  * (Andrea) DDS has a limit on typecode sizes, rmw_connextdds sets it to about ~65kb
* (Gerardo) There is a “type service” where you can request a type code with a “type id”
  * Type service is offered by each participant, but could be offered at a network level too (separate centralized type service host)
* (Emerson) can you get the aggregate message or recursively all of the types
  * (Miguel) you can get either the first level or all of the recursive message definitions
* (William) Could pass message hashsum on UserData if we didn’t want to lean on XTypes API for it. 
  * Could offer a ROS 2 Service by default on nodes that provides the .msg definition - xtypes would get us the .IDL definition which isn’t exactly the same.

jhurliman, 2021-08-01:
From my reading of documentation and testing, it looks like RTPS type introspection is only available with the "RTI Connext DDS Professional" (https://github.com/ros2/rmw_connextdds) and possibly "eProsima Fast RTPS"  (https://github.com/ros2/rmw_fastrtps) ROS2 middlewares. The default Eclipse CycloneDDS rmw does not send a PID_TYPE_OBJECT parameter during discovery, so type checking is done using the datatype string identifier only.

It seems like the two options are to implement TypeObject support in CycloneDDS, or switch the default connector and enable sending TypeObjects by default. Is one of these paths viable for the Humble release?

emersonknapp, 2021-08-03:
From discussing with Cyclone representative, my understanding is that TypeObject support is on the immediate roadmap for the next CycloneDDS release. https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/ROADMAP.md shows that 0.9 has "XTypes API". 0.8 release has already gone out. I believe the intention is that Humble will be using CycloneDDS >= 0.9

clalancette, 2021-08-19:
@emersonknapp We've added this to the Humble project board as something that we expect to get into Humble.  Is it OK if I also assign this ticket to you in order to track this and push it across the line?

emersonknapp, 2021-08-24:
I will accept being assigned, but as a fairly core feature I will need collaboration from OpenRobotics team members to make it happen - at least on the design and review levels, but ideally also some help with implementation when we get to that point.

sloretz, 2021-09-02:
> Could offer a ROS 2 Service by default on nodes that provides the .msg definition - xtypes would get us the .IDL definition which isn’t exactly the same.

The `rosidl` pipeline first converts `.msg/.srv/.action` files to `.idl` files and then generates code based off of the `.idl` files. It's also possible to create a topic/service/action using only an `.idl` file. I think getting the `.idl` definition would be more appropriate than getting the `.msg/.srv/.action` (if one exists) that was used to generate the `.idl`.

A per-node service that returns an `.idl` for a requested type could be implemented at the `rcl` level so that it is also available to non-DDS RMW implementations. It could be implemented by having the typesupport generators in the rosidl pipeline making the idl available as a string that could be fetched with some API.

emersonknapp, 2021-09-02:
>  think getting the .idl definition would be more appropriate than getting the .msg/.srv/.action (if one exists) that was used to generate the .idl.

I think this makes sense - and it makes me think it makes sense to go down the `UserData` route for DDS-based RMWs. But, that can be an implementation detail, if we craft the RMW API in such a way that it doesn't expose this information.

jhurliman, 2021-09-02:
> think getting the .idl definition would be more appropriate than getting the .msg/.srv/.action (if one exists) that was used to generate the .idl.

This would create a tight coupling between ROS2 and DDS that currently doesn't exist. Right now, it's possible to implement a compatible ROS2 node using only RTPS by using ROS2 msgdefs to serialize to/from CDR without any intermediate .idl translation. Here's an example ROS2 implementation that does exactly that: <https://github.com/foxglove/ros2>

If the RMW API was extended to now support querying IDLs by datatype (or by publisher/subscriber, however the final API looks) it would leak the DDS implementation detail out, and now any RMW or ROS2 implementation has to become aware of the `.idl` file format and how to do a lossless translation from idl back to msg. I'd like to avoid writing more parsers and increasing the surface area of the ROS2 API if possible.

emersonknapp, 2021-09-02:
I may be wrong - but I think there might be an overload of the term "IDL". I think that ROS2 considers there to be an "Interface Definition Language" that is specific to ROS2 - and it is not equivalent to the RTPS/DDS formats. This format is used to do code generation for language-specific structs - as per the design diagram at https://docs.ros.org/en/foxy/Concepts/About-Internal-Interfaces.html#type-specific-interfaces . The packages `rosidl_typesupport_${RMW_IMPLEMENTATION}` are the actual DDS-specific interfaces, so this should more precisely be called "RMW Typesupport formats", whereas the "ROS2 IDL" is universal to all possible RMW implementations.

IIUC, the "ROS2 IDL" is used for specific payloads, e.g. `MyServiceRequest` and `MyServiceResponse` - whereas the `.srv` contains definitions for both of these things. On the other hand, a `.msg` would produce only a single "ROS2 IDL" definition.

Somebody pretty please correct me if that's wrong.

jhurliman, 2021-09-02:
@emersonknapp that doesn't align with my reading of the code. https://github.com/ros2/rosidl/blob/master/rosidl_adapter/rosidl_adapter/msg/__init__.py is converting from `.msg/.srv/.action` files, ROS 2 message definitions as (loosely) defined at <https://docs.ros.org/en/foxy/Concepts/About-ROS-Interfaces.html> to IDL, an OMG specification defined at <https://www.omg.org/spec/IDL/4.2/PDF> that is tightly coupled to DDS.

sloretz, 2021-09-02:
> I may be wrong - but I think there might be an overload of the term "IDL" causing confusion.

By IDL I mean [OMG IDL 4.2](https://www.omg.org/spec/IDL/4.2/About-IDL/). ROS 2 message generation first converts from `.msg/.srv/.action` to OMG IDL `.idl` files (done by the `rosidl_adapter` package), and then passes the `.idl` files to code generators. https://design.ros2.org/articles/idl_interface_definition.html . Those files  get installed, have a look at `share/builtin_interfaces/msg/Duration.idl` for example.

> This would create a tight coupling between ROS2 and DDS that currently doesn't exist. Right now, it's possible to implement a compatible ROS2 node using only RTPS by using ROS2 msgdefs to serialize to/from CDR without any intermediate .idl translation.

Is this being done outside of the RMW API? There's no requirement for ROS 2 RMW implementations to use RTPS or CDR. I think [rmw_ecal](https://github.com/continental/rmw_ecal) and [rmw_iceoryx](https://github.com/ros2/rmw_iceoryx/) are a couple examples of non DDS and therefore non RTPS/CDR middlewares.

> on reading the linked design doc even more closely - it seems that .idl is specifically referring to OMG IDL format for DDS

I think part of the decision to use OMG IDL is an assumption that OMG IDL is not specifically for DDS, even if it happens to be used by DDS.

emersonknapp, 2021-09-02:
> I think part of the decision to use OMG IDL is an assumption that OMG IDL is not specifically for DDS, even if it happens to be used by DDS.

OK, this was not immediately obvious but makes sense.
* ROS 2 uses the OMG IDL as its language for defining messages/services/actions
* Some RMW implementations (including the default) are DDS, and that happens to also use the OMG IDL spec
* non-DDS RMW implementations will also receive the ROS 2 OMG IDL files as input for generating their own internal wire formats, because that's what ROS 2 uses to define its interfaces

jhurliman, 2021-09-02:
> ROS 2 uses the OMG IDL as its language for defining messages/services/actions

It does not. ROS 2 uses the "ROS2 Message description specification", available at <https://docs.ros.org/en/galactic/Concepts/About-ROS-Interfaces.html>. This is a similar definition format to OMG IDL but it is not the same thing. That's why conversion code exists.

> non-DDS RMW implementations will also receive the ROS 2 OMG IDL files as input for generating their own internal wire formats, because that's what ROS 2 uses to define its interfaces

@foxglove/ros2 deals exclusively with .msg/.srv/.action files. If it received a .idl file it would not know what to do with it.

emersonknapp, 2021-09-02:
> It does not. ROS 2 uses the "ROS2 Message description specification", 

I think the intention is - "ROS 2 uses OMG IDL as the common basis for implementation of the message/service/action descriptions and therefore it is inherently a part of ROS 2's interface description format" ?

jhurliman, 2021-09-02:
@emersonknapp is that decision documented anywhere? Why have the `.msg` format if OMG IDL is the official interface description format for ROS 2? There's no current need at all for OMG IDL to create a usable ROS 2 client right now which is why this seems like a major change.

jhurliman, 2021-09-02:
Circling back to the use cases for this feature request, part of runtime support for message definitions in visualization tools is the ability to show the definition to an end user. Users write `.msg` files, they are not familiar with the implementation detail of OMG IDL and would not understand why this alternative syntax is being shown back to them when they wrote a `.msg` file that looks different.

emersonknapp, 2021-09-02:
I'm just trying to interpret what I can see in the designs and the implementation of clients based on `rcl`. Maybe @sloretz or perhaps @wjwwood or @clalancette can shed more light than I could.

Enumerating the research hole:

Starting at https://design.ros2.org/articles/interface_definition.html 
> With the transition to use IDL for specifying interfaces in ROS 2 Dashing this article has been superseded by the Interface definition using .msg / .srv / .action files article.

Which is https://design.ros2.org/articles/legacy_interface_definition.html, where there is a "Conversion to IDL" section. The way it's worded seems to be saying that "ROS1 format is a legacy format, here's how we use it in ROS2"
> This article specifies the file format coming from ROS 1 describing the data structures exchanged by ROS components to interact with each other.

> Code is generated for defined interfaces to be usable by different client libraries. Interfaces described using the above format are first converted to IDL. Code generation uses the generated file.

Which points to the article https://design.ros2.org/articles/idl_interface_definition.html, which says
> This article identifies a subset of the Interface Definition Language (IDL) that can be used to describe interfaces between components.

Which to me means "ROS 2 interface language is a subset of OMG IDL"

A question I haven't been able to answer is, "is .msg/.srv/.action -> IDL a reversible operation?". I would agree that we should display definitions to users in the way that they define them, though maybe that doesn't mean that's exactly how they have to be represented in code, as long as we have all the necessary information

wjwwood, 2021-09-02:
I haven't had time to catch up on all of this, but...

> A question I haven't been able to answer is, "is .msg/.srv/.action -> IDL a reversible operation?".

I'd say probably not. There are things that can be done in IDL that cannot be done in msg/srv/action. If the original source was msg/srv/action, then presumably the resulting IDL could be simple enough to do that reverse transition, but I'm not 100% sure.

wjwwood, 2021-09-02:
My recollection is that we decided to stop extending the "old rosidl" formats (`.msg`/`.srv`/`.action`), but never deprecate them. If there's a feature that requires new IDL logic that we want to expose, e.g. annotating key fields or unions or something like that, then we'd recommend people use `.idl` files directly rather than try to extend the older formats .

The idea being that `.idl`, especially OMG IDL 4.2+, has all the features we'd ever need and we can parse it and use it as input for other code generation tasks, may that be directly generating C++/Python code or generating another intermediate file that is then used by another tool to do code generation, e.g. we could in theory generate protobuf files based on the `.idl` input.

Soooo

> Which to me reads as "ROS 2 interface language is a subset of OMG IDL"

I think this is right. The `rosidl_adapter` package is responsible for converting to the `.idl` format from the older `.msg`/`.srv`/`.action` interfaces, but after that our whole pipeline expects `.idl` and our parser `rosidl_parser` which parses these `.idl` files only supports a subset of the standard.

@sloretz or @clalancette can correct me if I'm wrong...

emersonknapp, 2021-09-02:
> then we'd recommend people use `.idl` files directly 

I'm not finding a tutorial that does this - but I do see that `rosidl_generate_interfaces` does take `.idl` files directly https://github.com/ros2/rosidl/blob/master/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake#L24 . 

This is a key bit of information that's been missing from this thread - ROS 2 applications may define all their interfaces without ever using a `msg/srv/action` file, which means there would be no "original" in that sense - and these would be perfectly valid ROS 2 applications. Conversely, all `msg/srv/action` files are converted to IDL before being used "as ROS 2"

wjwwood, 2021-09-02:
We probably don't have tutorials for this. It's not something we necessarily recommend unless you need it just because there's so many messages already in the `.msg` format and because for like ~90% of use cases the `.msg` format is perfectly fine and easier to read.

> This is a key bit of information that's been missing from this thread - ROS 2 applications may define all their interfaces without ever using a msg/srv/action file

Yeah, I'm not sure of even an example of this, but it is possible for sure.

> which means there would be no "original" in that sense - and these would be perfectly valid ROS 2 applications.

There is an original, it's just `.idl` instead of `.msg`...

> Conversely, all msg/srv/action files are converted to IDL before being used "as ROS 2"

Right, but that's kind of an implementation detail. From the user's perspective it's just that `.idl` or `.msg` can be used, it doesn't matter what our pipeline does with it after. For this interface, if I ask a node to give me the definition for example `sensor_msgs/msg/Image`, I want the original format, whether that be `.idl` or `.msg`. Always returning `.idl` might be ok, but isn't preferable to getting the original imo. Ideally you'd get both maybe if `.msg` was used originally.

emersonknapp, 2021-09-02:
>  I want the original format

This gets us to the crux of the issue. Is there any way for us to tell, at runtime, the exact text of the original file? i mean, maybe it's ok if comments are stripped, but it probably better if it's presented verbatim as originally written.

Not an expert here - but to do this maybe we need something like:
* `rosidl_adapter` includes the original text as metadata in the output `.idl` file
* Typesupport generation includes that information from the `.idl` file in a place where the RMW implementation can get it, perhaps as a static string in generated code. This probably doesn't need to live in the language-specific generators e.g. `rosidl_generator_cpp`/`rosidl_generator_py` - because it only needs to be parsed at the RMW implementation level so it can be passed along whichever mechanism is used to transmit the definitions. Whatever language is used on the receiving end would then get this definition.

wjwwood, 2021-09-02:
> This gets us to the crux of the issue. Is there any way for us to tell, at runtime, the exact text of the original file?

Sure the original files are installed. So when you're compiling anything that uses the message it could read that file and embed it in the application or it could look it up at runtime.

For example:

```
% ll /opt/ros/rolling/share/sensor_msgs/msg/Image.*
-rw-r--r-- 1 root root 2.0K Aug 11 08:58 /opt/ros/rolling/share/sensor_msgs/msg/Image.idl
-rw-r--r-- 1 root root 1.5K Aug 11 08:06 /opt/ros/rolling/share/sensor_msgs/msg/Image.msg
```

> rosidl_adapter includes the original text as metadata in the output .idl file

I don't think that's necessary, though it might be convenient.

> Does it make sense to pass along a raw string of the input - since this can be passed through rosidl_adapter/rosidl_parser simply enough to produce usable structures on the receiving end?

I think so. The only thing I could imagine is that this could cause the binaries to get pretty big, but that would only be an issue on embedded targets maybe, where library size is actual a concern.

amacneil, 2021-09-03:
So based on my read of this thread, are the following statements both true?

- ROS 2 supports using _both_ `.msg` or `.idl` (CDR) as the "source of truth" message definition format stored in codebases
- ROS 2 supports pluggable RMWs which may not depend on CDR (e.g. [rmw_ecal](https://github.com/continental/rmw_ecal))

If so, that seems like a bad leaky abstraction. It seems like ROS 2 should either (a) proclaim `.msg` as the source of truth, and allow RMWs to implement translation from `.msg` to their preferred wire format (i.e. CDR for DDS), or (b) proclaim `.idl` as the source of truth, all RMWs are required to support `.idl` only, with `.msg` only exists as a backwards compatibility layer (ideally phased out over time).

It seems strange to me that there is no clear answer on whether `.idl` or `.msg` is the source of truth for serialization at the RMW layer, and statements above claiming either or both.

wjwwood, 2021-09-03:
Well, OMG IDL does not necessarily == CDR. It's true they are used together but IDL does not assume the serialization format. I know that's confusing because counter examples aren't like this, e.g. protobuf's idl is somewhat tightly coupled to the serialization format. OMG IDL tries to avoid this coupling by having annotations which can contain "extra" info that might be used by serialization engines or even the middleware (like DDS specifics) but isn't required and the core syntax is just interface declaration.

In this case ROS 2 uses IDL to define the interface only. How that translates to serialization if at all is an implementation detail. 

The only leak is if you use annotations in a DDS/RTPS/CDR specific way and in this case tools that don't understand those annotations can mostly ignore them. 

Does that make more sense?

amacneil, 2021-09-03:
I see, I missed that distinction.

So essentially my proclamation (b) is accurate - ROS 2 architecturally (at least at rmw layer) speaks full IDL, and `.msg` is only a backwards compatible sugar?

If so, has it been considered to remove the additional layer of abstraction by marking `.msg` as a legacy ROS 1 format and move all new code over to `.idl`? It might be a slight learning curve for people migrating to ROS 2, but it would probably save people time in the long run since it is no longer a safe assumption that user space only needs to deal with `.msg` files.

wjwwood, 2021-09-03:
So I'd reformat your assertions as:

In ROS 2, a message, service, or action may be defined as either a `.idl` file or a `.msg`, `.srv`, or `.action` file, respectively. 

---

We have a python API which may ingest either `.msg` or `.idl` (limiting it to messages for example) and gives you an object that lets you introspect the message definition. 

Now the problem with this is that you probably want to interpret this message definition in C or C++ or TypeScript, etc. 

For this problem, I always imagined that we would store the description in a more commonly parsed format like XML or even a binary format. 

If you look at the XTypes standard that's exactly what they do. They support IDL, XML, XSD, and the binary format they call the TypeObject. 

See section 7.3: https://www.omg.org/spec/DDS-XTypes/1.3/PDF

We could do something similar where we provide either the original text (either msg or idl depending on the original source) or a JSON representation (for example) or XML copying XTypes or a binary representation (I would recommend our own so as not to leak XTypes as a dependency).

sloretz, 2021-09-03:
> @emersonknapp is that decision documented anywhere? Why have the .msg format if OMG IDL is the official interface description format for ROS 2?

The move to [OMG IDL was part of the Dashing release](https://docs.ros.org/en/crystal/Releases/Release-Dashing-Diademata.html#rosidl). I'm not sure where that decision is documented, but it's probably in one of the links from this ticket: https://github.com/ros2/rosidl/issues/346

> It seems like ROS 2 should either (a) proclaim .msg as the source of truth, and allow RMWs to implement translation from .msg to their preferred wire format (i.e. CDR for DDS), or (b) proclaim .idl [...]

I think (b) is mostly the case now. The handling of `.msg/.srv/.action` is a layer on top of the whole process, but it will likely never be phased out all the way. RMWs are required to support `.idl` as that's all they are given when asked to generate the code that converts the C/C++/Python/etc ROS 2 message to/from the RMW's wire format. See [rosidl_typesupport_fastrtps as an example](https://github.com/ros2/rosidl_typesupport_fastrtps/blob/152c360d6f241da75d5acb731d738c08bd419631/rosidl_typesupport_fastrtps_c/cmake/rosidl_typesupport_fastrtps_c_generate_interfaces.cmake#L21); it's given only the `.idl` files.

emersonknapp, 2021-09-03:
Agreeing with @wjwwood  that supporting both is nice. From looking at the idl files - I'd rather not write them by hand if I don't have to - the `msg/srv/action` syntax is certainly a lot more user friendly for humans both for reading and writing. Much bigger learning curve - and since all the functionality exists already I see no reason why it would need to go away.

For tool-builders however it's a different story and it's more complete to speak IDL - to the example of non-DDS impls,  `rmw_ecal` already understands and does conversions from IDL->protobuf, not from msg->protobuf. https://github.com/continental/rosidl_typesupport_protobuf/blob/master/rosidl_adapter_proto/rosidl_adapter_proto/__init__.py - so that provides a pretty good example that we don't need a CDR dependency to use non-DDS RMW

wjwwood, 2021-09-03:
I don't think we'll remove the support for msg files. Even if that incurs tooling overhead. Simply because it's a simple format that works well for simple things and it's so ingrained. Perhaps in the far future we could remove it, but in some sense the "original" text is only useful for humans. And really we should provide the already parsed description in a neutral format as I described above. That way the tool only has to parse that representation.

wjwwood, 2021-09-03:
> to the example of non-DDS impls, rmw_ecal already understands and does conversions from IDL->protobuf, not from msg->protobuf. 

Oh that's a good example, I didn't even know about it.

So, not sure how complete that is, but at least in theory it's something that can be done.

---

Also, if you look at Appendix B from the DDS XTypes standard they define the binary object as an IDL file, which is attractive to me at least, supposing that it uses a the same subset of IDL that we do, perhaps we could just use that more or less as-is. It's more that we need but it also is complete in theory and all of our end points can already parse it and receive it over the wire. At the very least it is a good starting point for our own version of this that doesn't leak XTypes directly.

amacneil, 2021-09-03:
Makes sense, thanks for the extra context. I not at least understand the argument that this new "runtime message definitions" should return IDL.

It's still unclear to me whether it's better for this API to support "always return IDL" or "return IDL or msg depending on what the original source used". For developer tools (like ours), it might be surprising to the user if they authored a msg definition and we showed them IDL. However, if IDL is what RMW is using to communicate then it's probably the most accurate thing for this API to return.

Another option would be if the msg->IDL conversion was made strictly reversible (only for message definitions which started as msg, assuming this information existed in metadata - if the user hand-authored IDL then obviously we would not support creating msg from it).

MiguelCompany, 2021-09-03:
@amacneil @wjwwood @emersonknapp 

After reading all your conversations about this, I think we could implement the service call with a parameter saying which definition to return. The service will be something like `TypeDefinitions get_type_definition(string topic_name, TypeDefinitionKindMask requested_definitions)` where `TypeDefinitionKindMask` would be a bit composable mask of an enumeration with values `MSG_DEFINITION, IDL_DEFINITION`, and `TypeDefinitions` would be a `sequence<pair<TypeDefinitionKind, string>>`.

The service will only return the requested definitions (if present). This way we could have tools requesting msg only, idl only, or both. Now that I think about it, having a tool that shows the msg format along with the IDL one would help the users in learning about the IDL format.

danthony06, 2021-09-22:
Is there any consensus on how to solve this, or a path to getting consensus? This is a pretty important issue for us, so if there's something we can do to move this along, please let us know.

emersonknapp, 2021-09-24:
Consensus will only be gained via an actual design review, probably with accompanying proof-of-concept prototype. I have currently committed as AWS Robotics to this feature on the Humble Roadmap https://docs.ros.org/en/rolling/Roadmap.html (to deliver Q1 2022). I expect to start work in October. Please reach out if you would like to collaborate.

My current distillation of this conversation is the following requirements:
* Always make IDL definition available
* If IDL was generated from a msg/srv/action, have that original source available as well
* The above should be a new RMW API against specific participants (to get their definition as-they-know-it)
* Ideally provide a hashsum to avoid passing full definition when it's not needed
* We might be able to use DDS XTypes to implement this API for DDS RMWs (maybe also DDS UserData field)
* As a fallback, or alternate implementation that is RMW-independent, this could be implemented via Service calls

jhurliman, 2021-09-25:
XTypes will not be able to speak msg/srv/action definitions so it may be a false path, but I haven’t looked closely at DDS UserData or how that is transmitted at the RTPS level.

clalancette, 2021-09-25:
I'd also like to make sure this functionality is RMW-independent.  Adding too many barriers to a non-DDS RMW doesn't seem like a good long-term strategy.

jhurliman, 2022-03-11:
Since it looks like there wasn't any movement on this in time for Humble, what is the recommended way of retrieving message definitions (the string contents of a msg/srv/action file) from the local filesystem at runtime? Is there an API to do this, or to at least get the filesystem paths given a datatype? Or do I need to write something from scratch that traverses AMENT_PREFIX_PATH looking for rosidl_interfaces index files?

Aposhian, 2022-04-14:
Why can Foxglove not simply load message definitions from the filesystem like every other ROS2 tool? Why is a new API needed? Yes, having the capability to inspect bags on systems without the messages is kind of interesting, but in my mind, having standard support for ROS2 custom messages (via the filesystem) is better than no support at all (the current situation). For example, Plotjuggler handles ROS2 custom messages just fine.

wjwwood, 2022-04-14:
I think tools like plotjuggler or rosbag would love to have this feature too, and not have to load things from the filesystem.

It's a common use case (in my experience) to want to install rosbag (or a similar tool) on a machine and record topics coming from a machine on the robot without having to install the packages that provide the messages onto your local machine too. 

This becomes even more important if you want to record messages from a topic that are out of date, i.e. you have the message package on your filesystem but it's a different version than the remote machine is using.

amacneil, 2022-10-12:
Cross-linking with REP-2011 which will hopefully provide a solution for this:

- https://github.com/ros-infrastructure/rep/pull/358

emersonknapp, 2023-02-23:
Just a note for anyone following this issue - implementation PRs for type hash generation and distribution are now ready for review. Distribution portion will need some unit/integration testing but they are working end-to-end by manual tests.

emersonknapp, 2023-03-13:
Doing a full pass for Type Hash Distribution PRs, making sure there's nothing e.g. glaringly wrong on Windows.

This build includes: ros2/rosidl#722, ros2/rmw#348, ros2/rmw_dds_common#70, ros2/rosidl_typesupport_fastrtps#95, ros2/rmw_fastrtps#671, ros2/rmw_cyclonedds#437, ros2/rmw_connextdds#104

* Linux [image]
* Linux-aarch64 [image]
* Windows [image]

Falimonda, 2023-07-20:
Do either the `iron` or `rolling` rosbag2_py APIs already support deserialization of messages without the message type's generated modules being installed on the system as is demonstrated in [the sequential reader test file](https://github.com/ros2/rosbag2/blob/iron/rosbag2_py/test/test_sequential_reader.py#L55) now that the encoded_message_definition can be pulled from the bag file using the reader's get_all_message_definitions method?

If not, is there an open issue or other reference to work being done to perform deserialization using just the encoded_message_definition and serialized data? Doing that seems core to addressing related issues such as [rosbag2/issues/782](https://github.com/ros2/rosbag2/issues/782), but that's been closed and I haven't found examples of it being done.

emersonknapp, 2023-07-21:
No, that is not implemented. It's not covered by this particular issue, which just covers generating and distributing TypeDescriptions. I think https://github.com/ros2/ros2/issues/1374 , and more specifically "Runtime interface reflection" language. @methylDragon would be able to add more insight to exactly what is implemented or not so far - but I believe we do have the capability, given a `TypeDescription`, to use `rcl` APIs to create a deserializer of some type. There would still be a few steps to being able to use the `rosbag2` stored message definitions, which are still original source text rather than extracted `TypeDescription` - but I think most if not all of the core capabilities do exist.

Feel free to make a new feature request issue on https://github.com/ros2/rosbag2, if one does not exist already!

Falimonda, 2023-07-24:
Thanks for the feedback. I'll take a look at the existing issues to understand where things stand today.

emersonknapp, 2023-07-28:
I am going to close this issue as successfully completed. There will be improvements, extensions, tooling etc but the core functionality has been implemented, it will be more useful to track ongoing work in targeted issues for those specific improvements.

closed 2023-07-28 by emersonknapp