ros2/ros2 #1161: ROS2 compiled fail due to foonathan_memory_vendor

Bug-terminator, 2021-07-25:
## Bug report

- Operating System:
  -- MacOS 11.5 Big Sur
- Installation type:
  --from source 
- Version or commit hash:
  -- Rolling 
- DDS implementation:
  -- I'm not sure,  my laptop is macbook pro 2019 16` Intel
- Client library (if applicable):
  -- N/A 

#### Steps to reproduce issue
I followed steps on the official developer website https://docs.ros.org/en/rolling/Installation/macOS-Development-Setup.html, errors occured while compiling, here is the log: 
```shell
--- stderr: iceoryx_utils
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: no LC_RPATH load command with path: /Users/bytedance/ros2_rolling/build/iceoryx_utils found in: /Users/bytedance/ros2_rolling/install/iceoryx_utils/lib/libiceoryx_utils.dylib (for architecture x86_64), required for specified option "-delete_rpath /Users/bytedance/ros2_rolling/build/iceoryx_utils"
```
```shell
--- stderr: foonathan_memory_vendor
In file included from /Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/src/detail/align.cpp:5:
In file included from /Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/include/foonathan/memory/detail/align.hpp:14:
In file included from /Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/include/foonathan/memory/detail/../config.hpp:21:
/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext-build/foonathan/hosted_implementation.hpp:30:60: error: expected ';' at end of declaration
    typename std::remove_reference<T>::type&& move(T&& arg) FOONATHAN_NOEXCEPT
                                                           ^
                                                           ;
/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext-build/foonathan/hosted_implementation.hpp:30:61: error: C++ requires a type specifier for all declarations
    typename std::remove_reference<T>::type&& move(T&& arg) FOONATHAN_NOEXCEPT
                                                            ^
/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext-build/foonathan/hosted_implementation.hpp:32:9: error: expected expression
        return static_cast<typename std::remove_reference<T>::type&&>(arg);
        ^
/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext-build/foonathan/hosted_implementation.hpp:33:6: error: expected ';' after top level declarator
    }
     ^
     ;
In file included from /Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/src/detail/align.cpp:5:
In file included from /Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/include/foonathan/memory/detail/align.hpp:15:
/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/include/foonathan/memory/detail/assert.hpp:20:56: error: expected function body after function declarator
                                      const char* fnc) FOONATHAN_NOEXCEPT;
                                                       ^
/Users/bytedance/ros2_rolling/build/foonathan_memory_vendor/foo_mem-ext-prefix/src/foo_mem-ext/include/foonathan/memory/detail/assert.hpp:23:50: error: expected function body after function declarator
                                const char* fnc) FOONATHAN_NOEXCEPT;
...
```

#### Expected behavior
Compile successfully.
#### Actual behavior
Complie fail.
#### Additional information
My questions are:
1. What is LC_RPATH, will this error affect the compilation result?
2. How to fix errors from foonathan_memory_vendor?
3. Is it a MacOS Big Sur problem?

clalancette, 2021-08-02:
>     3\. Is it a MacOS Big Sur problem?

It very well may be related to that.  We have nightly CI on macOS Mojave that does work: https://ci.ros2.org/view/nightly/job/nightly_osx_release/2143/ .

If you find out what it is and how to fix it, please feel free to open a PR with the fixes.  We're happy to review.

clalancette, 2023-03-30:
Given the age of this, and the fact that we phased out macOS, I'm going to close this.  Feel free to reopen if you feel differently.

closed 2023-03-30 by clalancette