ros2/ros2 #1164: Fresh install of ROS2 Galactic fails to load _rclpy_pybind11.cp38-win_amd64.pyd

wl2776, 2021-08-03:
## Bug report

**Required Info:**

- Operating System: 
    - Windows 10, x64, all recent updates
- Installation type:
  - binaries
- Version or commit hash:
  - Galactic
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
1. launch cmd.exe 
2. `set PATH=c:\Python38\Scripts;c:\Python38;%PATH% && call c:\opt\ros\galactic\local_setup.bat`
3. `ros2 doctor --report`

#### Expected behavior

Report without errors.

#### Actual behavior
```
c:\>  ros2 doctor --report
Failed to load entry point 'hello': DLL load failed while importing _rclpy_pybind11: The specified procedure could not be found.
The C extension 'c:\opt\ros\galactic\Lib\site-packages\rclpy\_rclpy_pybind11.cp38-win_amd64.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
c:\opt\ros\galactic\Lib\site-packages\ros2doctor\api\__init__.py: 143: UserWarning: Fail to call RMWReport class functions.
c:\opt\ros\galactic\Lib\site-packages\ros2doctor\api\__init__.py: 127: UserWarning: Report entry point TopicReport fails to load: DLL load failed while importing _rclpy_pybind11: The specified procedure could not be found.
The C extension 'c:\opt\ros\galactic\Lib\site-packages\rclpy\_rclpy_pybind11.cp38-win_amd64.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions

   NETWORK CONFIGURATION
inet        : 106.xxx.xxx.xxx
inet4       : ['106.xxx.xxx.xxx']
device      : ppp adapter _g_vpn - eussl2.example.com
hostname    : desktop-ejl17ns
ether       : 30:xx
device      : ethernet adapter ethernet
ether       : e0:xx
device      : wireless lan adapter local area connection* 1
ether       : e2:xx
device      : wireless lan adapter local area connection* 2
inet        : 192.168.2.43
inet4       : ['192.168.2.43']
ether       : e0:xx
device      : wireless lan adapter wi-fi
hostname    : desktop-ejl17ns

   PACKAGE VERSIONS
actionlib_msgs                            : latest=2.2.3, local=2.2.3
action_msgs                               : latest=1.0.3, local=1.0.3
action_tutorials_cpp                      : latest=0.14.3, local=0.14.3
action_tutorials_interfaces               : latest=0.14.3, local=0.14.3
action_tutorials_py                       : latest=0.14.3, local=0.14.3
ament_clang_format                        : latest=0.10.6, local=0.10.6
ament_clang_tidy                          : latest=0.10.6, local=0.10.6
ament_cmake                               : latest=1.1.4, local=1.1.4
ament_cmake_auto                          : latest=1.1.4, local=1.1.4
ament_cmake_clang_format                  : latest=0.10.6, local=0.10.6
ament_cmake_clang_tidy                    : latest=0.10.6, local=0.10.6
ament_cmake_copyright                     : latest=0.10.6, local=0.10.6
ament_cmake_core                          : latest=1.1.4, local=1.1.4
ament_cmake_cppcheck                      : latest=0.10.6, local=0.10.6
ament_cmake_cpplint                       : latest=0.10.6, local=0.10.6
ament_cmake_export_definitions            : latest=1.1.4, local=1.1.4
ament_cmake_export_dependencies           : latest=1.1.4, local=1.1.4
ament_cmake_export_include_directories    : latest=1.1.4, local=1.1.4
ament_cmake_export_interfaces             : latest=1.1.4, local=1.1.4
[... 286 lines cut ...]
urdf                                      : latest=2.5.2, local=2.5.2
urdf_parser_plugin                        : latest=2.5.2, local=2.5.2
visualization_msgs                        : latest=2.2.3, local=2.2.3
yaml_cpp_vendor                           : latest=7.1.0, local=7.1.0
zstd_vendor                               : latest=0.9.1, local=0.9.0

   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD

   ROS 2 INFORMATION
distribution name      : galactic
distribution type      : ros2
distribution status    : active
release platforms      : {'rhel': ['8'], 'ubuntu': ['focal']}
```

#### Additional information

DependenciesGui shows missing imports from WS2_32.dll
[image]

I've tried both recommended Python version 3.8.3 and more recent version 3.8.10

fujitatomoya, 2021-08-03:
Just FYI, this does not happen Linux platform.

<details>
  <summary>ros2 doctor --report on Linux / Galactic Binary Release </summary>
  
```
root@263f0153ac57:~# ros2 doctor --report

   NETWORK CONFIGURATION
inet         : 172.17.0.4
inet4        : ['172.17.0.4']
ether        : 02:42:ac:11:00:04
netmask      : 255.255.0.0
device       : eth0
flags        : 4163<UP,BROADCAST,RUNNING,MULTICAST>
mtu          : 1500
broadcast    : 172.17.255.255
inet         : 127.0.0.1
inet4        : ['127.0.0.1']
netmask      : 255.0.0.0
device       : lo
flags        : 73<UP,LOOPBACK,RUNNING>
mtu          : 65536

   PACKAGE VERSIONS
launch_testing_ros                        : latest=0.14.2, local=0.14.2
ament_cmake_export_libraries              : latest=1.1.4, local=1.1.4
ament_cmake_copyright                     : latest=0.10.6, local=0.10.6
console_bridge_vendor                     : latest=1.3.2, local=1.3.2
ament_cmake_export_include_directories    : latest=1.1.4, local=1.1.4
ros2doctor                                : latest=0.13.2, local=0.13.2
rclpy                                     : latest=1.9.0, local=1.9.0
diagnostic_msgs                           : latest=2.2.3, local=2.2.3
rcutils                                   : latest=4.0.2, local=4.0.2
rclcpp                                    : latest=9.1.0, local=9.1.0
launch_ros                                : latest=0.14.2, local=0.14.2
ament_cmake_version                       : latest=1.1.4, local=1.1.4
rosidl_default_runtime                    : latest=1.1.1, local=1.1.1
ros2cli_common_extensions                 : latest=0.1.1, local=0.1.1
composition_interfaces                    : latest=1.0.3, local=1.0.3
ament_cmake_cppcheck                      : latest=0.10.6, local=0.10.6
rcl_action                                : latest=3.1.2, local=3.1.2
ament_cmake_flake8                        : latest=0.10.6, local=0.10.6
uncrustify_vendor                         : latest=1.5.3, local=1.5.3
ament_cmake_export_interfaces             : latest=1.1.4, local=1.1.4
ros2lifecycle                             : latest=0.13.2, local=0.13.2
rosidl_generator_c                        : latest=2.2.1, local=2.2.1
ament_cmake_target_dependencies           : latest=1.1.4, local=1.1.4
ament_cppcheck                            : latest=0.10.6, local=0.10.6
rmw                                       : latest=3.3.1, local=3.3.1
tinyxml2_vendor                           : latest=0.7.4, local=0.7.4
ament_cmake_python                        : latest=1.1.4, local=1.1.4
sensor_msgs                               : latest=2.2.3, local=2.2.3
rosidl_adapter                            : latest=2.2.1, local=2.2.1
libstatistics_collector                   : latest=1.1.0, local=1.1.0
ros2service                               : latest=0.13.2, local=0.13.2
ament_cmake_export_definitions            : latest=1.1.4, local=1.1.4
rcl_interfaces                            : latest=1.0.3, local=1.0.3
ament_cmake_export_targets                : latest=1.1.4, local=1.1.4
ros2param                                 : latest=0.13.2, local=0.13.2
osrf_pycommon                             : latest=0.2.1, local=0.2.1
domain_coordinator                        : latest=0.9.2, local=0.9.2
ros2node                                  : latest=0.13.2, local=0.13.2
rosidl_typesupport_introspection_cpp      : latest=2.2.1, local=2.2.1
rosidl_cli                                : latest=2.2.1, local=2.2.1
ament_index_cpp                           : latest=1.2.0, local=1.2.0
launch_xml                                : latest=0.17.0, local=0.17.0
rosidl_default_generators                 : latest=1.1.1, local=1.1.1
rosidl_cmake                              : latest=2.2.1, local=2.2.1
ros_workspace                             : latest=1.0.2, local=1.0.2
[... 127 lines cut ...]
yaml_cpp_vendor                           : latest=7.1.0, local=7.1.0
eigen3_cmake_module                       : latest=0.1.1, local=0.1.1
sqlite3_vendor                            : latest=0.9.1, local=0.9.0

   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.8.0-59-generic-x86_64-with-glibc2.29
release          : 5.8.0-59-generic
processor        : x86_64

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : galactic
distribution type      : ros2
distribution status    : active
release platforms      : {'rhel': ['8'], 'ubuntu': ['focal']}

   TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```
</details>

wl2776, 2021-08-04:
I've built ROS2 from sources, binding to Python 3.9.

`ros2 doctor --report` gives similar error, and dependencies show the same missing imports
[image]

wl2776, 2021-08-04:
Well, it looks like missing imports from WS2_32.dll is not the reason of the issue.
I've got one more installation of ROS2, rolling build from sources, it doesn't show import errors, but dependensices shows the same missing imports

wl2776, 2021-08-04:
I've also noticed that there are no mentions about RMW in the report.

j-rivero, 2021-09-02:
:farmer: might be the same problem that is present in ci.ros2.org, https://ci.ros2.org/view/nightly/job/nightly_win_deb/lastCompletedBuild/testReport/(root)/laser_geometry/___/

wl2776, 2022-03-22:
The problem is still here. Any updates?

tianrking, 2023-01-16:
How to solve ?

clalancette, 2024-09-11:
Since this was filed against Galactic, which is out of support, I'm going to close this out.  If you are still having issues with Humble or newer, please file a new bug and we can debug it there.

closed 2024-09-11 by clalancette