ros2/ros2 #1170: How to pause/resume ros2 bag

liuxw7, 2021-08-24:
Hi all, In ros1 bag , we use rosbag play xxxx.bag, and we will key-down space key to pause playing, but in ros2 foxy or galastic, I can't find the way to pause the bag playing .  please give me some support to solve the problem . thanks a lot.

Kaju-Bubanja, 2021-08-24:
Check: https://github.com/ros2/rosbag2/issues/696

emersonknapp, 2021-08-24:
This feature is in progress and will be released into Humble. [Feature Request Ticket](https://github.com/ros2/rosbag2/issues/55) - see https://github.com/ros2/rosbag2/pull/847 for the in-progress implementation work. 

In Galactic, this feature was exposed via a Service on the playback node, so a simple custom "remote-control" client could be created. See a sample implementation of such a client here https://github.com/ros2/rosbag2/pull/729#issuecomment-819747652

I don't have Triage access on this repository to close this issue, but I recommend the maintainers do so as this is not a global ROS2 issue and is tracked locally on rosbag2

liuxw7, 2021-08-25:
I got it, thanks a lot!

closed 2021-08-25 by liuxw7

GPrathap, 2024-10-11:
In Humble: Pause the playback: Press the Spacebar to pause the playback