ros2/ros2 #1173: ROS2 Type Inference from YAML-Parameters

mmatthe, 2021-08-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - tested in Docker with binary install according to installation instructions
- Version or commit hash:
  ```
   apt-cache show ros-galactic-desktop
   Package: ros-galactic-desktop
   Version: 0.9.3-2focal.20210820.074533
   ```

- DDS implementation:
  - the default

#### Steps to reproduce issue
I use this rclpy node source:

```python
# Refer to the following API:
import rclpy
from rclpy.node import Node
from rcl_interfaces.msg import SetParametersResult

class ParametersNode(Node):
    def __init__(self):
        super().__init__('ParametersNode', automatically_declare_parameters_from_overrides=True)
        self.get_logger().info("Initialized Parameters Node.")
        self._timer = self.create_timer(1, self.printParam)
        self.declare_parameter("self_declared_param", 45)
        # self.set_parameters_callback(self.onParameterChange)
        self.add_on_set_parameters_callback(self.onParameterChange)

    def printParam(self):
        params = ["test", "group.list_param", "group.string_param", "group2.complex_param",
                  "group.group2.float_param", "group.group2.bool_param"]
        for p in params:
            try:
                obj = self.get_parameter(p)
                value = obj._value
                self.get_logger().info(f"Parameter {p}: {value} ({type(value)}), ({obj.type_})")
            except rclpy.exceptions.ParameterNotDeclaredException:
                self.get_logger().info(f"Parameter {p} not defined!")

    def onParameterChange(self, params):
        txt = "\n".join(f"   {p.name}={p.value}" for p in params)
        self.get_logger().info("Parameter changed callback called! Changed parameters:\n" + txt)
        return SetParametersResult(successful=True)

def main(args=None):
    rclpy.init(args=args)

    node = ParametersNode()
    rclpy.spin(node)

    node.destroy_node()
    rclpy.shutdown()
```

with this parameter file:

```yaml
ParametersNode:
  ros__parameters:
    test: 42
    group:
      list_param: [1, 2, 3]
      string_param: "this is a string"
      group2:
        float_param: 3.1415
        bool_param: false
    group2:
      complex_param: +1+2j
```

After building, I run in one console

```console
# ros2 run parameters_pkg parameters_node --ros-args --params-file parameters/parameters.yaml 
[INFO] [1630307729.066721800] [ParametersNode]: Initialized Parameters Node.
[INFO] [1630307730.068282942] [ParametersNode]: Parameter test: 42 (<class 'int'>), (Type.INTEGER)
[...]
[INFO] [1630307731.068062784] [ParametersNode]: Parameter test: 42 (<class 'int'>), (Type.INTEGER)
[...]
```

In another console I do
```console
root@7d36bcfd5724:/tmp/ros2-examples-workspace# ros2 param get ParametersNode self_declared_param
Integer value is: 45
root@7d36bcfd5724:/tmp/ros2-examples-workspace# ros2 param get ParametersNode test
Integer value is: 42
root@7d36bcfd5724:/tmp/ros2-examples-workspace# ros2 param set ParametersNode test 874
Setting parameter failed: Wrong parameter type, expected 'Type.NOT_SET' got 'Type.INTEGER'
root@7d36bcfd5724:/tmp/ros2-examples-workspace# ros2 param set ParametersNode self_declared_param 874
Set parameter successful
root@7d36bcfd5724:/tmp/ros2-examples-workspace# ros2 param get ParametersNode test
Integer value is: 42
root@7d36bcfd5724:/tmp/ros2-examples-workspace# ros2 param get ParametersNode self_declared_param
Integer value is: 874
root@7d36bcfd5724:/tmp/ros2-examples-workspace# 
```

#### Expected behavior
I expect that I can set both the `test` as well as the `self_declared_parameter` from the console using `ros2 param set`.

#### Actual behavior
I can only set the self_declared_parameter. The type of the `test` parameter is expected to be `NOT_SET` even though the console output shows, that the parameter is actually parsed as type `TYPE.INTEGER`

Is it my fault or is this a bug with the present version? (The above code was working with Eloquent as is)

clalancette, 2021-08-30:
Hm, that does seem like a bug to me.  @ivanpauno Can you take a look and comment?

mmatthe, 2021-09-13:
@ivanpauno can you confirm this behaviour?

mmatthebi, 2021-09-28:
Can somebody confirm this or forward me to the correct place where this should be handled? I can also reproduce with the current master branch of ros2

ivanpauno, 2021-11-23:
Sorry for the delay in replying.

This looks like a bug, I will take a look locally to see what's the issue

ivanpauno, 2021-11-24:
https://github.com/ros2/rclpy/pull/853 should fix the issue.
@mmatthe could you test it?

closed 2021-11-29 by ivanpauno