ros2/ros2 #1175: get "EE.EEEE.EE..." when "vcs import src < ros2.repos"

mumu1408, 2021-09-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - from source

#### Steps to reproduce issue

I follow the installation instructions at "https://docs.ros.org/en/eloquent/Installation/Linux-Development-Setup.html".
And when I clone all repos with ```vcs import src < ros2.repos```, I get a string of ```EE.EEEE.EE....```, and it takes a lot of time.

I want to know if the output of `EE.EEEE.EE...` is normal.

Thanks for your help.

aprotyas, 2021-09-01:
You get the `E` outputs when `vcstool` cannot access the remote repositories.

mumu1408, 2021-09-01:
> You get the `E` outputs when `vcstool` cannot access the remote repositories. This could indicate some internet problems on your end.

Thanks for your reply. It has taken more than 2 hours. Do I need to interrupt the instruction `vcs import src < ros2.repos` with `Ctrl+C`? Or wait for it to end itself?

aprotyas, 2021-09-01:
It depends on what you want to do. If you want to terminate the process without pulling all of the source code, you can interrupt with <kbd>Ctrl+C</kbd>.

aprotyas, 2021-09-01:
I would suggest closing this issue and redirecting this question to https://answers.ros.org.

closed 2021-09-01 by mumu1408