ros2/ros2 #1177: Declare a dependency on the (Linux-only) LTTng userspace tracer

christophebedard, 2021-09-01:
## Feature request

#### Feature description

This is a follow-up to #957 and relates to https://gitlab.com/ros-tracing/ros2_tracing/-/issues/71

I would like to discuss declaring a dependency on the LTTng userspace tracer (LTTng-UST) on Linux. I'd like to make sure that this change would be welcomed. My goal would be for this to be implemented a fair amount of time before the next release (Humble).

Core ROS 2 packages are instrumented with calls to functions provided by the `tracetools` package. These functions contain:

1. LTTng tracepoints if LTTng is installed/detected and `tracetools` is re-compiled/overlayed
2. nothing when building from source if LTTng isn't installed/detected
3. nothing in debians/Linux binaries
4. nothing in any code built for/on non-Linux platforms (macOS, Windows)

and

5. the function calls themselves can be completely removed by building instrumented core packages with `-DTRACETOOLS_DISABLED=ON`

This LTTng dependency isn't declared. This was initially for simplicity (especially since this is a platform-specific dependency) and because of some concerns about overhead.

This means that this feature currently isn't available out-of-the-box; users need to manually install LTTng and build from source (or at least just `tracetools`), since it's not installed when installing debians or running `rosdep install`. This has led to some questions and issues:

* https://gitlab.com/ros-tracing/ros2_tracing/-/issues/94
* https://gitlab.com/ros-tracing/ros2_tracing/-/merge_requests/139

As mentioned in https://github.com/ros2/ros2/issues/957#issuecomment-658920580, the LTTng userspace tracer is less "invasive" than the kernel tracer. I don't think we need to declare a dependency on the kernel tracer. Users could install it manually if they want to record kernel data for their analyses.

Also, in practice, the overhead is really negligible if tracepoints are not explicitly enabled: https://github.com/lttng/lttng-ust/blob/1f8a8ec9581af89d98aec47de9ad9e25087cd54a/include/lttng/tracepoint.h#L53-L62. Users/companies that do not want to ship a ROS 2-based product with LTTng can completely disable/remove it by setting the right CMake option (see item 5 above) and building from source, which I assume is something they are already doing.

#### Implementation considerations

New dependencies:

* `liblttng-ust-dev`: LTTng-UST
* `lttng-tools`: LTTng CLI (not strictly necessary, but recommended)

These would of course need to be installed on the Linux ci.ros2.org executors.

If the [tracing tests](https://gitlab.com/ros-tracing/ros2_tracing/-/tree/master/test_tracetools/test) turn out to create problems on ci.ros2.org, skipping them could be an option, but I think it would probably be fine.

Also, the `ros2_tracing` README currently recommends installing both kernel and userspace tracers: https://gitlab.com/ros-tracing/ros2_tracing#building. Some logic would need to change in the package supporting the `ros2 trace` command and the `Trace` action so that it doesn't assume that it's either all (kernel+userspace tracers) or nothing.

Any other platform-related considerations?

christophebedard, 2022-01-18:
@clalancette friendly ping :smiley: I see that you self-assigned this. Is there any way I can help push this forward?

christophebedard, 2022-03-04:
@clalancette friendly ping

mjcarroll, 2023-01-25:
Also of note, the users who are deeply concerned with performance (on the level of lttng serspace tracepoints being an issue) are likely _not_ using our pre-compiled binaries and are instead compiling themselves.

cottsay, 2023-01-25:
> cottsay I may need your input on RHEL, it appears to be 2.12, but I'm not sure I was using the correct package listing.

It is indeed 2.12.0:
```
$ dnf list -q --available --showduplicates lttng-ust
Available Packages
lttng-ust.x86_64          2.12.0-6.el9          rhel-9-for-x86_64-appstream-rpms
```

christophebedard, 2023-01-25:
I can work on the main `ros2_tracing` PR to enable the instrumentation by default.

> The only concern I can think of would be the availability of lttng-ust on our supported platforms which is currently:

This kind of is related to https://github.com/ros2/ros2_tracing/issues/16. If all current Linux platforms have LTTng-UST >= 2.12, we should be fine, but things could still be improved.

mjcarroll, 2023-01-25:
Yes, 2.12 is going to be the least common denominator for rolling.  I think we would have no intention of backporting this into released distributions.

christophebedard, 2023-01-25:
> Yes, 2.12 is going to be the least common denominator for rolling. I think we would have no intention of backporting this into released distributions.

Agreed

clalancette, 2023-01-25:
One thing I will note is that RHEL-9 is aspirational for Rolling at this time.  We may or may not get it in before Iron, so Iron may end up supporting RHEL-8 still.  I'm fine if we end up disabling tracing by default on RHEL-8, but we should keep that in mind.

christophebedard, 2023-04-04:
The PRs have been merged, so I'm closing this.

closed 2023-04-04 by christophebedard