ros2/ros2 #1179: About #1073 solution

dzywater, 2021-09-05:
I am always curious about why rviz2 can run on one computer and failed on another computer. Both computers have same environment. Based on ros2-release5, I test it again. I found that :
(1) rviz2 cannot run and print #1073 error if I unzip ros2-foxy-20210902-windows-release-amd64.tar.bz2 and rename ros2-windows as DIYName(call this setup.bat)
(2) rviz2 can run if I unzip ros2-foxy-20210902-windows-release-amd64.tar.bz2 and copy ros2-windows(delete ) as DIYName(call this setup.bat)

The difference is folder's modified date(Or authority)
"rename" way does not update folder's modified date.
"copy" way updates folder's date as today.

Besides,  #1072 still exists.

fujitatomoya, 2021-09-08:
is this the another issue aside from https://github.com/ros2/ros2/issues/1073? if not, probably better not to have duplicated issue to track.

clalancette, 2021-09-23:
No response to the question, so I'm going to close this out.  If you are still having problems with RViz, please open an issue on https://github.com/ros2/rviz

closed 2021-09-23 by clalancette

flynneva, 2021-09-23:
@dzywater rviz does a lot of things behind the scenes and tries to use the right graphics library depending on the hardware available at hand. this means it might run just fine on an Intel x86 machine with an NVIDIA graphics card but then have some weird bugs on one with a lower-powered GPU. you can see an example of this in [this issue where it was specific to Intel laptops](https://github.com/ros-visualization/rviz/issues/1508) and wasnt an issue with RVIZ but the underlying graphics library `lib-mesa`.