ros2/ros2 #1180: Galatic rclpy demo_nodes_py listener unresponsive to cpp talker

desmith17, 2021-09-05:
## Bug report
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11' for Galatic demo_nodes_py listener unresponsive to cpp talker but cpp talker works with cpp listener.
**Required Info:**

- Operating System:
Ubuntu 20.04
- Installation type:
  -I followed the ROS installation guide "Installing ROS 2 on Ubuntu Linux"
- Version or commit hash:
  - ros2-galactic-20210616-linux-focal-amd64.tar.bz2
- DDS implementation:
  - RTI Connext
- Client library (if applicable):
  - 

#### Steps to reproduce issue
In one terminal, 

`davidsros@DavidsROS-VirtualBox:~/ros2_galactic$ ros2 run demo_nodes_cpp talker
[INFO] [1630874128.215680031] [talker]: Publishing: 'Hello World: 1'
[INFO] [1630874129.215548018] [talker]: Publishing: 'Hello World: 2'
`
In another  terminal,
`davidsros@DavidsROS-VirtualBox:~/ros2_galactic$ ros2 run demo_nodes_py listener
`

#### Expected behavior
`[INFO] [1630874165.215848180] [listener]: I heard: [Hello World: 38]`

#### Actual behavior
`Traceback (most recent call last):
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/demo_nodes_py/listener", line 33, in <module>
    sys.exit(load_entry_point('demo-nodes-py==0.14.3', 'console_scripts', 'listener')())
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/demo_nodes_py/listener", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 855, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/python3.8/site-packages/demo_nodes_py/topics/listener.py", line 16, in <module>
    from rclpy.node import Node
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/python3.8/site-packages/rclpy/node.py", line 43, in <module>
    from rclpy.client import Client
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/python3.8/site-packages/rclpy/client.py", line 22, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/python3.8/site-packages/rclpy/impl/implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "/home/davidsros/ros2_galactic/ros2-linux/lib/python3.8/site-packages/rpyutils/import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension '/home/davidsros/ros2_galactic/ros2-linux/lib/python3.8/site-packages/_rclpy_pybind11.cpython-39-x86_64-linux-gnu.so' isn't present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system' for possible solutions
`
#### Additional information
I ran listener in cpp and was able to listen to the talker,
`davidsros@DavidsROS-VirtualBox:~/ros2_galactic$ ros2 run demo_nodes_cpp listener
[INFO] [1630874165.215848180] [listener]: I heard: [Hello World: 38]
[INFO] [1630874166.215945083] [listener]: I heard: [Hello World: 39]
[INFO] [1630874167.216308192] [listener]: I heard: [Hello World: 40]
`

fujitatomoya, 2021-09-08:
I cannot reproduce this issue.

```
root@263f0153ac57:~# dpkg -l | grep demo-nodes-py
ii  ros-galactic-demo-nodes-py                          0.14.3-1focal.20210819.183850        amd64        Python nodes which were previously in the ros2/examples repository but are now just used for demo purposes.

root@263f0153ac57:~# ros2 run demo_nodes_cpp talker
[INFO] [1631122413.656561001] [talker]: Publishing: 'Hello World: 1'
[INFO] [1631122414.656431342] [talker]: Publishing: 'Hello World: 2'
[INFO] [1631122415.656463209] [talker]: Publishing: 'Hello World: 3'
[INFO] [1631122416.656400408] [talker]: Publishing: 'Hello World: 4'
[INFO] [1631122417.656428172] [talker]: Publishing: 'Hello World: 5'
[INFO] [1631122418.656361490] [talker]: Publishing: 'Hello World: 6'
[INFO] [1631122419.656408759] [talker]: Publishing: 'Hello World: 7'
[INFO] [1631122420.656345711] [talker]: Publishing: 'Hello World: 8'
[INFO] [1631122421.656393457] [talker]: Publishing: 'Hello World: 9'
[INFO] [1631122422.656308843] [talker]: Publishing: 'Hello World: 10'

root@263f0153ac57:~# ros2 run demo_nodes_py listener
[INFO] [1631122419.679276603] [listener]: I heard: [Hello World: 7]
[INFO] [1631122420.658358729] [listener]: I heard: [Hello World: 8]
[INFO] [1631122421.658501246] [listener]: I heard: [Hello World: 9]
```

aprotyas, 2021-09-08:
Did any of the suggestions in the [linked troubleshooting page](https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system) help?

clalancette, 2021-09-23:
No response in 15 days, so I'm going to close this out.  Please feel free to reopen if you are still having problems.

closed 2021-09-23 by clalancette