ros2/ros2 #1183: Connect the ROS2 logger to console_bridge

martiniil, 2021-09-11:
## Feature request

#### Feature description

Given a library using `console_bridge` for logging, it is possible to register an output-handler that uses the `ros2` logger.
This would allow utilizing the `ros2` logging mechanism via `console_bridge`. See http://wiki.ros.org/rosconsole_bridge.

As far as I can see, this feature is still missing.

#### Implementation considerations

It should be possible to start from http://wiki.ros.org/rosconsole_bridge and replace the `ros::console` usage by the `ros2` logging.

martiniil, 2021-10-01:
@clalancette Can you shed some light on the state of `console_bridge` in ros2 for us? I saw that the `class_loader` uses it (since it is ros-independent), whereas in pluginlib the ros2 logging is used. I don't see a connection there.

Also we wondered what is the purpose of https://github.com/ros2/console_bridge_vendor?

We would be glad to help.

clalancette, 2021-10-01:
> Also we wondered what is the purpose of https://github.com/ros2/console_bridge_vendor?

That repository is so that we can build `console_bridge` on platforms that don't have it packaged, i.e. Windows.

In general, we don't currently have a cohesive strategy around `console_bridge` in ros2.  The places that use it have been generally ported from ROS 1, but anything new basically is using the ros2 logging infrastructure.  Clarifying this is part of a larger effort of clarifying the logging story in ROS 2 in general, like https://github.com/ros2/design/pull/315 is trying to do.